espressif/esp-adf #1106: white noise  at playing 24-bit audio

vwwwvwww, 2023-11-08:
esp-adf v2.6
esp-idf v5.1.1
board:  lyrat_v4_3

 ..\esp-adf\examples\player\pipeline_sdcard_mp3_control

    // mp3_decoder_cfg_t mp3_cfg = DEFAULT_MP3_DECODER_CONFIG();
    // mp3_decoder = mp3_decoder_init(&mp3_cfg);
    wav_decoder_cfg_t wav_cfg=DEFAULT_WAV_DECODER_CONFIG();
    wav_decoder = wav_decoder_init(&wav_cfg);

When playing 16-bit 44.1k audio,the  sound is generally normal  (with some interruptions).
When playing 24-bit 48k audio, noise (white noise) is mixed with the music sound.
When playing 24-bit 96k audio, it is all noise (white noise).

Please let me know what needs to be adjusted to obtain the correct sound ?

TempoTian, 2023-11-10:
when play 24bits audio need adjust MCK div to be 384, for default 256 can not devided by 3
```
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_WRITER;
    i2s_cfg.i2s_config.mclk_multiple = I2S_MCLK_MULTIPLE_384;
    i2s_stream_writer = i2s_stream_init(&i2s_cfg);
```
If you are using codec chip, you need also align the div settings:
Such as change MCLK_DIV_FRE 256 in es8311.c to be 384

vwwwvwww, 2023-11-11:
Thank you . I set the i2s_cfg.i2s_config.mclk_multiple to I2S_MCLK_MULTIPLE_384 and observed some improvement, but the issue was not completely resolved. 

At 48 kHz, there is still some noise present, although significantly reduced compared to before. 
At 96 kHz, the noise is no longer pure white noise; instead, the music is mixed with more white noise  (and the playback speed is slightly slower).

 

I tried to change the values of .dma_buf_count and .dma_buf_len, but it had no effect. I also attempted to set I2S_STREAM_BUF_SIZE = 3600*3, but it only resulted in more stuttering without any improvement in the issue.
I only changed the sampling rate to 96000 in the i2s_stream.h.

I noticed that my situation is similar to the one described here: 
https://github.com/espressif/esp-adf/issues/941. 
Should I consider installing this patch?

play_sdcard_music_example.c

     // Example of linking elements into an audio pipeline -- START
        audio_pipeline_handle_t pipeline;
        audio_element_handle_t fatfs_stream_reader, i2s_stream_writer, music_decoder;
    
        esp_log_level_set("*", ESP_LOG_WARN);
        esp_log_level_set(TAG, ESP_LOG_INFO);
    
        ESP_LOGI(TAG, "[ 1 ] Mount sdcard");
        // Initialize peripherals management
        esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
        esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
    
        // Initialize SD Card peripheral
        audio_board_sdcard_init(set, SD_MODE_1_LINE);
    
        ESP_LOGI(TAG, "[ 2 ] Start codec chip");
        audio_board_handle_t board_handle = audio_board_init();
        audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_DECODE, AUDIO_HAL_CTRL_START);
    
        ESP_LOGI(TAG, "[3.0] Create audio pipeline for playback");
        audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
        pipeline = audio_pipeline_init(&pipeline_cfg);
        mem_assert(pipeline);
    
        ESP_LOGI(TAG, "[3.1] Create fatfs stream to read data from sdcard");
        fatfs_stream_cfg_t fatfs_cfg = FATFS_STREAM_CFG_DEFAULT();
        fatfs_cfg.type = AUDIO_STREAM_READER;
        fatfs_stream_reader = fatfs_stream_init(&fatfs_cfg);
    
        ESP_LOGI(TAG, "[3.2] Create i2s stream to write data to codec chip");
        i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
        i2s_cfg.type = AUDIO_STREAM_WRITER;
        i2s_cfg.i2s_config.mclk_multiple = I2S_MCLK_MULTIPLE_384 ;
        i2s_stream_writer = i2s_stream_init(&i2s_cfg);
    
        ESP_LOGI(TAG, "[3.3] Create wav decoder");
        wav_decoder_cfg_t  wav_dec_cfg  = DEFAULT_WAV_DECODER_CONFIG();
        music_decoder = wav_decoder_init(&wav_dec_cfg);
    
        ESP_LOGI(TAG, "[3.4] Register all elements to audio pipeline");
        audio_pipeline_register(pipeline, fatfs_stream_reader, "file");
        audio_pipeline_register(pipeline, music_decoder, "dec");
        audio_pipeline_register(pipeline, i2s_stream_writer, "i2s");
    
        ESP_LOGI(TAG, "[3.5] Link it together [sdcard]-->fatfs_stream-->music_decoder-->i2s_stream-->[codec_chip]");
        const char *link_tag[3] = {"file", "dec", "i2s"};
        audio_pipeline_link(pipeline, &link_tag[0], 3);
     
        ESP_LOGI(TAG, "[3.6] Set up uri: /33.wav");  
        audio_element_set_uri(fatfs_stream_reader, "/sdcard/34.wav"); // 96k wav
     
        ESP_LOGI(TAG, "[ 4 ] Set up  event listener");
        audio_event_iface_cfg_t evt_cfg = AUDIO_EVENT_IFACE_DEFAULT_CFG();
        audio_event_iface_handle_t evt = audio_event_iface_init(&evt_cfg);
    
        ESP_LOGI(TAG, "[4.1] Listening event from all elements of pipeline");
        audio_pipeline_set_listener(pipeline, evt);
    
        ESP_LOGI(TAG, "[4.2] Listening event from peripherals");
        audio_event_iface_set_listener(esp_periph_set_get_event_iface(set), evt);
    
        ESP_LOGI(TAG, "[ 5 ] Start audio_pipeline");
        audio_pipeline_run(pipeline);
        // Example of linking elements into an audio pipeline -- END

jason-mao, 2023-11-20:
@vwwwvwww  https://github.com/espressif/esp-adf/issues/941. is already in v2.6.   Could you provide your source music file and your sdkconfig file?