espressif/esp-adf #1107: Add ESP32-S3-Korvo-1 from ESP-Skainet

abmantis, 2023-11-11:
**Is your feature request related to a problem? Please describe.**

I was trying to use ESP ADF for ESP32-S3-Korvo-1, but the configuration is missing.

**Describe the solution you'd like**

Please add esp32s3-korvo-1 to `components/audio_board` from `esp-skainet/components/hardware_driver/boards`, so we can use `CONFIG_ESP32_S3_KORVO_1_V4_0_BOARD`.

jason-mao, 2023-11-20:
@abmantis Thanks for your mention.I need to consider ways to prevent doing the same task repeatedly for the `esp32s3-korvo-1` board.

abmantis, 2023-11-20:
@jason-mao  I was trying to follow what is done for other boards like Esp32-s3-box: https://github.com/espressif/esp-adf/compare/master...abmantis:esp-adf:korvo1_2.5

But still not able to get proper audio.

jason-mao, 2023-11-21:
@abmantis 

> But still not able to get proper audio.

Could you provide more details?, such as is it not playing music? or Can't receive the voice ?

abmantis, 2023-11-21:
Hi @jason-mao . I was having high pitched audio, but I was finally able to get it working! Both mic and speaker output working: https://github.com/espressif/esp-adf/pull/1113

abmantis, 2023-12-16:
@jason-mao have you had a change to look at the PR?

jason-mao, 2023-12-20:
@abmantis Sure, I reviewed that PR, and it is good.  As far as we are concerned how can we avoid some duplication of work and mutual inclusion of such boards in the future. I'll handle it after we discuss it internally.

abmantis, 2024-02-26:
Hi @jason-mao . Any update regarding getting the S3 Korvo1 officially supported on esp-adf?

abmantis, 2024-04-03:
Hi again @jason-mao . No plans to officially supporto the S3 Korvo 1?