espressif/esp-adf #1109: ESP-ADF 2.6 plays audio on different channel on mono mp3 file with play_mp3_control example as compared to ESP-ADF 2.4's

chewhs00, 2023-11-14:
## Environment

- Audio development kit:      ESP32-LyraT 4.3
- Audio kit version v4
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it): v5.1.1
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it): v2.6
- Build system:         idf.py
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
- Operating system:     Linux
- Power supply:         USB

## Problem Description
When running the example "play_mp3_control" from adf-2.4 and adf-2.6 on the same ESP32-LyraT 4.3 dev board where the channel format is 'I2S_CHANNEL_FMT_RIGHT_LEFT' as default, and play some mono mp3 files by converting the existing stereo mp3 files to mono, I noticed the following: On adf-2.4, only the right channel is played, while on adf-2.6, only the left channel is played. 
When running the above example with customized "my_board" running on ESP32S3, on adf-2.4, both channels are played regardless of channel format.

By changing to different channel formats while playing on ESP32-LyraT 4.3:
adf-2.4:
  * Left channel:    
      I2S_CHANNEL_FMT_ONLY_LEFT
  * Right channel:  
    I2S_CHANNEL_FMT_RIGHT_LEFT
    I2S_CHANNEL_FMT_ALL_LEFT
    I2S_CHANNEL_FMT_ALL_RIGHT
    I2S_CHANNEL_FMT_ONLY_RIGHT

adf-2.6:
   * Left channel:
       I2S_CHANNEL_FMT_RIGHT_LEFT
       I2S_CHANNEL_FMT_ALL_LEFT
       I2S_CHANNEL_FMT_ONLY_LEFT
   * Right channel:
       I2S_CHANNEL_FMT_ALL_RIGHT
       I2S_CHANNEL_FMT_ONLY_RIGHT

### Expected Behavior
I am not sure what the expected behavior should be given that the mp3 files used is mono, and with I2S_CHANNEL_FMT_RIGHT_LEFT, I thought it should play mono source on both channels, which it does on my custom ESP32S3 board. However, after updating to idf 5.1.1 and adf 2.6, only the left channel is played.

### Actual Behavior
After updating to idf 5.1.1 and adf 2.6, only the left channel is played.

### Goal
How do I setup I2S such that both channels get played if the mp3 source is mono?

jason-mao, 2023-11-20:
@chewhs00  ADF v2.4 use IDF v4.4.0 and v2.6 use IDF v4.4.4. I remember some I2S related bugs had been fixed on v4.4.2. I think your problem is where the differences come from.  Recommend adding a filter element after decoder [like](https://github.com/espressif/esp-adf/blob/ef058dac1fb53d73b354ed8e7b2ae873e09fd937/examples/player/pipeline_sdcard_mp3_control/main/play_sdcard_mp3_control_example.c#L185C14-L185C14) to suitable different channel music source.

chewhs00, 2023-11-22:
> @chewhs00 ADF v2.4 use IDF v4.4.0 and v2.6 use IDF v4.4.4. I remember some I2S related bugs had been fixed on v4.4.2. I think your problem is where the differences come from. Recommend adding a [...]

Is ADF v2.6 compatible with IDF 5.1? The readme file from ADF 2.6 gave me the impression that ESP-ADF Release/v2.6 is compatible with ESP-IDF Release/v5.1. 
By default, does the mono source from the mp3 get played at which channel? Both adf versions seems to play at different channels.