espressif/esp-adf #1110: example\algorithm_examples 录音问题

Jackyliszsh, 2023-11-17:
-df 5.1.1, adf 2.6, hardware esp32-s3-box-lite

运行examples\advanced_examples\algorithm 下的例子， 把录音后的文件通过WIFI传出来。 测试发现不管编码为amrnb,armwb或者opus, 录音后的文件播放时都是快速前进。附件有新闻联播的一段录音。而运行examples\recorder\pipeline_recording_to_sdcard，录音文件正常。

换了lyrat_mini_v1_1的ESP32 
[audio-test3-3A.zip](https://github.com/espressif/esp-adf/files/13389244/audio-test3-3A.zip)
DEMO板测试，运行这个algorithm录音文件为正常。

请问一下esp32s3运行这个alrorithm录音变快可能是什么原因呢？

jason-mao, 2023-11-20:
@Jackyliszsh 先排除WIFI传输带来的影响，你把录音通过fatfs stream 写入到SDCARD，看看声音是否正常。另外，你试一下wav格式，是否有同样的问题。听你的文件感觉是采样率对应不上。

Jackyliszsh, 2023-11-20:
不用WIFI，改入写SD卡。 当完整运行（不改任何代码），声音正常。 当停止pipeline_play（注释掉audio_pipeline_run（pipeline_play）,改为录制20秒钟），录出来的声音就会出现以前那个现象（感觉像快进）。 
实际案列应用中，不需要播放，只是想用到AEC/AGC/NS/VAD这几个算法来处理录音。

Jackyliszsh, 2023-11-24:
请问一下，这个问题，您们能复现到不？ 或者还需要提供一下更多的一些信息吗？ @jason-mao