espressif/esp-adf #1111: ESP not responding to OPTIONS during registration process

davidedigesualdo, 2023-11-21:
## Environment

- Audio development kit: ESP32-LyraT
- Audio kit version: v4.3
- Module or chip used:  ESP32-WROVER-E
- IDF version: both v4.4.4-278-g3c8bc2213c and v5.0.1
- ADF version: v2.6-23-gef058dac
- Build system: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0
- Operating system: Linux
- Using an IDE?: No
- Power supply: USB

## Problem Description

I tried to build and run the example code under "examples/protocols/voip". Codec and streams initializations are ok. 
When the SIP client tries to register to the SIP PBX (Asterisk), the PBX replies with "Auth requested". Then the SIP client sends authenticated registration message, and the SIP PBX replies with an OPTIONS message. The client doesn't reply, but restarts the loop of registration.

### Expected Behavior

Into the "Allow" string the SIP client sends "OPTIONS", so it should implement "OPTIONS" methods.
The expecteed behaviour is the SIP client replies with "200 OK" to the options message, so the PBX sends "200 OK" to the REGISTER message and the registration process is complete.

### Actual Behavior

The SIP client doesn't reply to the PBX OPTIONS message and restarts the registration process again and again.

## Debug Logs

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7296
load:0x40078000,len:16484
ho 0 tail 12 room 4
load:0x40080400,len:4260
entry 0x40080668
I (29) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (29) boot: compile time 15:18:03
I (29) boot: chip revision: v3.1
I (33) qio_mode: Enabling default flash chip QIO
I (38) boot.esp32: SPI Speed      : 80MHz
I (43) boot.esp32: SPI Mode       : QIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00200000
I (90) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=34038h (213048) map
I (169) esp_image: segment 1: paddr=00044060 vaddr=3ffb0000 size=03360h ( 13152) load
I (173) esp_image: segment 2: paddr=000473c8 vaddr=40080000 size=08c50h ( 35920) load
I (187) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=ee8a4h (977060) map
I (455) esp_image: segment 4: paddr=0013e8cc vaddr=40088c50 size=0d334h ( 54068) load
I (484) boot: Loaded app from partition at offset 0x10000
I (484) boot: Disabling RNG early entropy source...
I (496) psram: This chip is ESP32-D0WD
I (496) spiram: Found 64MBit SPI RAM device
I (497) spiram: SPI RAM mode: flash 80m sram 80m
I (499) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (506) cpu_start: Pro cpu up.
I (510) cpu_start: Starting app cpu, entry point is 0x40081530
I (0) cpu_start: App cpu up.
[... 186 lines cut ...]
I (4757) SIP:

OPTIONS sip:104245410@192.168.10.100:13573;transport=UDP SIP/2.0
Via: SIP/2.0/UDP SIPSERVER:5060;branch=z9hG4bK70ada0ed;rport
Max-Forwards: 70
From: "asterisk" <sip:asterisk@SIPSERVER>;tag=as01fd19c1
To: <sip:104245410@192.168.10.100:13573;transport=UDP>
Contact: <sip:asterisk@SIPSERVER:5060>
Call-ID: 2628799d441a0e9d0ed931863cad185e@SIPSERVER:5060
CSeq: 102 OPTIONS
User-Agent: OpenVoip PBX
Date: Tue, 21 Nov 2023 06:25:35 GMT
Allow: INVITE, ACK, CANCEL, OPTIONS, BYE, REFER, SUBSCRIBE, NOTIFY, INFO, PUBLISH, MESSAGE
Supported: replaces, timer
Content-Length: 0

I (4807) SIP: [1970-01-01/00:00:01]<<======================
E (4815) SIP: Error register device
W (4818) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1850
W (5825) SIP: CHANGE STATE FROM 0, TO 0, :func: sip_reconnect:383
```

## Suggestions

It would be great if you can add the possibility to customize the allowed methods in the SIP client configuration, so that a developer can enable/disable the supported methods.

ahhfzhang, 2023-11-23:
Hi @davidedigesualdo ,

Our SIP client usually replies with "200 OK" to the 'OPTIONS' message, but not during registration process.

Maybe you could turn off the 'OPTIONS' message on the PBX ？ In addition, we also support configuring the 'OPTIONS' message on client side [here](https://github.com/espressif/esp-adf-libs/blob/619b3bac9a34617f288c6e68f3c587618b1c68bb/esp_media_protocols/include/esp_rtc.h#L111C40-L111C40).

The other thing is why the 'OPTIONS' message is sent before the registration is replied. Maybe you can try to use TCP/TLS to registration to the server.

br.

davidedigesualdo, 2023-11-23:
Hi @ahhfzhang,
thank you for your kind reply.

Unfortunately the PBX is a public service which is not directly managed by me. I've already notified the PBX customer service, but they cannot switch off the OPTIONS during the registration process.

Moreover, the RFC 3261 (SIP: Session Initiation Protocol) doesn't prevent the UAS to send an OPTIONS message to the UA during the registration process, so the correct behaviour should be the UA to answer to the OPTIONS message.

I've tried to use TCP connection, and the behaviour is the same.

davidedigesualdo, 2024-01-19:
Hi,
is there any update about this issue?

kamcio2603, 2024-06-08:
+1, the same problem with public voip provider, it works great on fritzbox voip device, but not with public providers :(

carlosrodridel, 2025-03-31:
@ahhfzhang having the same problem here.

[image]

the server's default behaviour is sending `OPTIONS` message jsut after `REGISTER` and right before replying `REGISTER`with `200 OK`

jimforr, 2025-08-27:
@kamcio2603 @davidedigesualdo @carlosrodridel 
hello everyone, sorry for wait such a long time.
You can try replacing this library file and try again.

[lib.zip](https://github.com/user-attachments/files/22003548/lib.zip)