espressif/esp-adf #1116: ESP-RTSP client play issue

soloex1991eric, 2023-12-07:
- Audio development kit:    ESP32-S3-Korvo-2
- Audio kit version: v3.1
- Module or chip used:  ESP32-S3-WROOM-1
- IDF version: v4.4.5
- ADF version: v2.6
- Build system: idf.py
- Running log:
```cpp
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x178c
load:0x403c9700,len:0xe78
load:0x403cc700,len:0x3184
SHA-256 comparison failed:
Calculated: e119777bb061c3121d710f70e08f6133a25f4941cbaa38f1f7152f91925042e5
Expected: 34285a0f8ad089f0025f7eba06b53e0cb9ba4401bbcd049102d37dc161b683ae
Attempting to boot anyway...
entry 0x403c9980
I (43) boot: ESP-IDF v4.4.5-dirty 2nd stage bootloader
I (43) boot: compile time 12:30:01
I (43) boot: chip revision: v0.1
I (45) qio_mode: Enabling default flash chip QIO
I (51) boot.esp32s3: Boot SPI Speed : 80MHz
I (56) boot.esp32s3: SPI Mode       : QIO
I (60) boot.esp32s3: SPI Flash Size : 16MB
I (65) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (89) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  2 factory          factory app      00 00 00010000 00400000
I (104) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=3c130020 size=4a400h (304128) map
I (163) esp_image: segment 1: paddr=0005a428 vaddr=3fc9d730 size=05bf0h ( 23536) load
I (167) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=128bb4h (1215412) map
I (353) esp_image: segment 3: paddr=00188bdc vaddr=3fca3320 size=02760h ( 10080) load
I (355) esp_image: segment 4: paddr=0018b344 vaddr=40378000 size=15724h ( 87844) load
I (384) boot: Loaded app from partition at offset 0x10000
I (385) boot: Disabling RNG early entropy source...
I (396) opi psram: vendor id : 0x0d (AP)
I (396) opi psram: dev id    : 0x02 (generation 3)
I (397) opi psram: density   : 0x03 (64 Mbit)
I (400) opi psram: good-die  : 0x01 (Pass)
I (405) opi psram: Latency   : 0x01 (Fixed)
I (410) opi psram: VCC       : 0x01 (3V)
I (414) opi psram: SRF       : 0x01 (Fast Refresh)
I (420) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (425) opi psram: BurstLen  : 0x01 (32 Byte)
I (430) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (436) opi psram: DriveStrength: 0x00 (1/1)
I (442) spiram: Found 64MBit SPI RAM device
I (446) spiram: SPI RAM mode: sram 80m
I (450) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (458) cpu_start: Pro cpu up.
I (461) cpu_start: Starting app cpu, entry point is 0x403795f4
0x403795f4: call_start_cpu1 at /home/soloex/esp/esp-idf/components/esp_system/port/cpu_start.c:147

I (0) cpu_start: App cpu up.
I (756) spiram: SPI SRAM memory test OK
I (756) spiram: Instructions copied and mapped to SPIRAM
I (839) spiram: Read only data copied and mapped to SPIRAM
I (867) cpu_start: Pro cpu start user code
I (867) cpu_start: cpu freq: 240000000
I (867) cpu_start: Application information:
I (870) cpu_start: Project name:     esp-rtsp
I (875) cpu_start: App version:      1
I (879) cpu_start: Compile time:     Dec  7 2023 12:29:59
[... 186 lines cut ...]
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379c7c
0x40379c7c: esp_restart_noos at /home/soloex/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

```
## Problem Description
ESP-RTSP client example does not work, crashes when using happytime-rtsp-server-x64,
[https://www.happytimesoft.com/products/rtsp-server/index.html](url), even though application works with VLC player.
 
``
E (37137) TRANS_TCP: tcp_poll_write select error 104, errno = Connection reset by peer, fd = 54
E (37142) ESP_RTSP_CLIENT: Write failed, error: Socket is not connected
``
### Expected Behavior
Audio streams to board

### Actual Behavior
Audio board reboots with error

### Steps to Reproduce
Upload code to board start rtsp in rtsp client play mode

### Code to Reproduce This Issue
Rtsp example code with
``.vcodec_type = AV_VCODEC_NULL,``

ahhfzhang, 2023-12-19:
Hi @soloex1991eric 

We do not support this rtsp-server yet, but we will support it soon.

Thanks.

soloex1991eric, 2023-12-22:
Noted one other thing is it possible to implement two way communication using rtsp? if so how do I proceed?

matz-9, 2024-02-14:
> Noted one other thing is it possible to implement two way communication using rtsp? if so how do I proceed?

Hi, by any chance, did you manage to get two-way audio communication working?

ahhfzhang, 2024-02-29:
Hi ,

Yes, You can create two rtsp clients, one for puser to push local video and audio to rtsp server, another for player to pull rtsp server audio for playback. So that is two-way audio communication and one-way video preview.

br.

matz-9, 2024-02-29:
I'll try to implement this solution.

Thanks!

jason-mao, 2024-07-31:
@matz-9 Any update from your side?

closed 2025-02-17 by jason-mao