espressif/esp-adf #1128: VoIP应用，通话中设备网络波动导致esp_rtc状态切换出错

shangpeng01, 2023-12-27:
esp-idf-v4.4.5+esp-adf-master
VoIP实际应用中，如果在通话过程中网络波动导致WiFi中断，然后又重新连接上。
SIP协议栈的状态会从ESP_RTC_EVENT_AUDIO_SESSION_BEGIN跳转到ESP_RTC_EVENT_REGISTERED
ESP_RTC_EVENT_AUDIO_SESSION_END和ESP_RTC_EVENT_HANGUP没有被调用到，导致音频资源一直被占用
然后如果手机端重新呼叫设备，第一次呼叫设备无法正常接通，需要挂断后再一次呼叫

复现步骤：esp-adf-master中的voip例程，WiFi连接改为手动调用esp_wifi_set_config()，和esp_wifi_connect()去连接。（用例程里的wifi_service的话，网络波动导致WiFi断开后不能自动重连）
然后用手机端SIP应用呼叫设备，接通后，将设备连接的热点关闭几秒再重新打开。

[image]

ahhfzhang, 2024-01-16:
Hi @shangpeng01 

网络断开需要销毁voip服务，因为重新连接ip都可能发生变化的，这时应该重新去注册服务。

br.

shangpeng01, 2024-01-16:
这种网络短时间波动断开又恢复的情况，有办法可以恢复原来的通话吗？
类似手机端的Linphone，如果通话中网络短时间断开又恢复，是可以继续正常通话的