espressif/esp-adf #1133: SDP misformat in VoIP example

matz-9, 2024-01-17:
## Environment

- Audio development kit: ESP32-LyraT
- Audio kit version v3
- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.4 or v5.1.2-602-gdb1e54a0c5
- ADF version: v2.6-52-gdd204c42
- Build system: idf.py
- Compiler version (crosstool-NG esp-2021r2) 8.4.0 or (crosstool-NG esp-12.2.0_20230208) 12.2.0
- Operating system:  Linux
- Using an IDE?: No
- Power supply: USB

## Problem Description

When I encountered the problem, I was running some voip call tests (using the VoIP example project) between 2 sip clients (ESP32 SIP and Linphone) connected to a Kamailio proxy server. The call from LyraT to client works fine, the real problem occurs when from the sip client one tries to make a call to LyraT, as it sends a badly formatted SDP packet: specifically the owner username and session name are set to "Z" instead of the acutal values (100 in my case). 

### Expected Behavior

Filling in SDP fields with the correct values

### Actual Behavior

Compilation of some SDF fields with "Z"

### Steps to Reproduce

Clone and Run VoIP example 

### Code to Reproduce This Issue

VoIP example in ADF 

## Debug Logs

Some logs to clarify the issue:

linphonec> Call 3 to <sip:100@192.168.1.159;realFrom=65001> ringing.
2024-01-16 17:55:02:743 ortp-error-session_description parser error for [v=0
o=Z 0 3 IN IP4 192.168.1.93
s=Z
t=0 0
c=IN IP4 192.168.1.93
m=audio 31362 RTP/AVP 8 101
a=rtpmap:8 PCMA/8000
a=rtpmap:101 telephone-event/8000
]

## Other Items

[image]

Thank you for your attention. I'm doing a thesis project with this, please let me know if your are planning to work on it.

ahhfzhang, 2024-02-29:
Hi @matz-9 ,

We don't support linphone yet but I don't think that may be the reason here as other servers or clients or even protocol standards don't require this, but you can try using twilio which is a fully supported service by us .

br.

matz-9, 2024-02-29:
Hi @ahhfzhang,

you are right. That wasn't the problem here, but we got to the bottom of it! And if it helps, we found out that the problem with linphone can be solved by reversing the (t) and (c) fields, so (t) under (c), in the SDP offer, when the esp receives an incoming call.

[image]

Like that.