espressif/esp-adf #1137: pwm_stream在播放完成退出时未等待pwm缓冲区中的数据被实际刷新完成

yyjdelete, 2024-01-23:
测试: idf5.1+adf master

当前版本的pwm_stream, 疑似在来自上游管道的数据全部被写入完成后就立即由于AEL_IO_DONE触发了close, 但由于pwm本身不是同步写入的, 而是通过timer_group_isr在异步刷新, close时又会立即关闭这个timer, 会造成最后一段时间仍在缓冲区中的音频数据丢失
(放tts的音频可以试出来, 默认speed下, 大概会截断最后1~2个字的语音)

TempoTian, 2024-02-02:
可以测试下附件的patch OK吗，加了一个等播完的流程
[pwm_stream.patch](https://github.com/espressif/esp-adf/files/14134834/pwm_stream.patch)

yyjdelete, 2024-02-02:
@TempoTian 
试了下, 有个小问题, 改了之后能正常结束
1. 完成时，这里count应该是判断等于0

2. 另外有个之前就存在的用法错误，虽然好像没啥直接的影响。栈上分配的xHigherPriorityTaskWoken没有设置初始值(xSemaphoreGiveFromISR仅会设置pdTRUE，pdFALSE时是不会设置的)
```diff
     }
 
     if ((0 == handle->data->is_give && pwm_data_list_get_free(handle->data) > BUFFER_MIN_SIZE) ||
-        (2 == handle->data->is_give && pwm_data_list_get_count(handle->data) > 0)) {
+        (2 == handle->data->is_give && pwm_data_list_get_count(handle->data) == 0)) {
         handle->data->is_give = 1;
-        BaseType_t xHigherPriorityTaskWoken;
+        BaseType_t xHigherPriorityTaskWoken = pdFALSE;
         xSemaphoreGiveFromISR(handle->data->semaphore, &xHigherPriorityTaskWoken);
         if (pdFALSE != xHigherPriorityTaskWoken) {
             portYIELD_FROM_ISR();
```

TempoTian, 2024-02-02:
Thanks，我local测试注意到了,目前测试delay在100ms左右(6868-6777)
```
6777 run on line  216
6868 run on line 326 left 0
6868 run on line 218
```

jason-mao, 2024-07-31:
It's fixed on 09987f45ce465c374ca20100648395b244344706

closed 2024-07-31 by jason-mao