espressif/esp-adf #1138: 宏ADF_VER不是生成自ADF的git版本号

yyjdelete, 2024-01-23:
不确定我的理解是否存在问题, 但ADF的SDK在编译过程中生成的宏ADF_VER, 并非和IDF_VER(IDF_PATH)一样， 生成自ADF_PATH所对应的git版本号, 而是当前正在编译的主项目的git版本号

https://github.com/espressif/esp-adf/blob/331d9126c8a1620c4e7684063e5d15b5063b183a/CMakeLists.txt#L1-L4
上面CMake的代码是否遗漏了对路径的切换?

https://github.com/espressif/esp-adf/blob/331d9126c8a1620c4e7684063e5d15b5063b183a/project.mk#L1

TempoTian, 2024-02-02:
execute_process 命令的执行环境是由它本身所在的目录决定的，由于这个CMakeLists.txt在ADF的根目录这样写是OK的

yyjdelete, 2024-02-02:
@TempoTian
你可以实际试一下, 它实际上跑出来就是这样的, ADF_VER是当前编译的project的版本号, 如果project没做git版本控制就是空字符串

测试环境: IDF v5.1.2, ADF(master), win_x64, IDF的vscode插件v1.7.0(应该和这个无关, 直接在cmd里面跑也是一样的结果)
测试的项目就是ADF的example, 复制了一个到版本库以外的地方

手动改了下ADF里面的参数, 加了个`--test`方便区分, 拿procmon抓的进程信息
可以看到当前文件夹是项目文件夹下的build目录
[image]

TempoTian, 2024-02-02:
Thanks, 我测试了下确实是，可以按照下面的写法改下, 我来提交一个MR
```
execute_process(COMMAND git describe --always --tags --dirty
                OUTPUT_VARIABLE ADF_VER
                OUTPUT_STRIP_TRAILING_WHITESPACE
                WORKING_DIRECTORY ${CMAKE_CURRENT_LIST_DIR})
```

jason-mao, 2024-02-20:
It's fixed on 698deecaccc2635e3316c75b19a884177d6fcf7c

closed 2024-02-20 by jason-mao