espressif/esp-adf #1144: TS_DECODER: No Valid Audio or StreamType not supported! line:546

espradio, 2024-02-05:
Following stream shows issues latest esp-adf [1]
url=https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/playlist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/playlist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/chunklist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/media_1085.ts
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/media_1086.ts
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/media_1087.ts
E (141856) TS_DECODER: No Valid Audio or StreamType not supported! line:546
E (141857) TS_DECODER: not support, still under-developing, line:758
E (141860) AAC_DECODER: Prefill ts data error, line:405
E (141866) AUDIO_ELEMENT: [aac] AEL_STATUS_ERROR_OPEN,-1
W (141871) AUDIO_ELEMENT: [aac] audio_element_on_cmd_error,7

[1]
commit 48dea7711391e0abf00c65aca18557572fc5189a (HEAD -> master, origin/master, origin/HEAD)
Merge: 1b827c94 74c88720
Author: Jason-mao <maojianxin@espressif.com>
Date:   Fri Feb 2 15:59:12 2024 +0800

    Merge branch 'bugfix/fix_wwe_example_i2s_error' into 'master'

    example: Fix exceptions caused by the impact of the new i2s API on the www example

    See merge request adf/esp-adf-internal!1269

TempoTian, 2024-02-06:
The audio format in mentioned TS file is MP3 which is not supported currently.
Now only support AAC audio format in TS, needs a feature request to support it

TempoTian, 2024-02-07:
For a quick solution you can use attached code to test, put library into `components/esp-adf-libs/esp_codec/lib/esp32s3/`
You may need upgrade to latest ADF if show link error.
[HLS_mp3_support.zip](https://github.com/espressif/esp-adf/files/14190708/HLS_mp3_support.zip)

espradio, 2024-02-09:
Hi it is esp32 not s3. Could you provide lib for components/esp-adf-libs/esp_codec/lib/esp32

TempoTian, 2024-02-18:
Use following lib for esp32 and have a try
[libesp_processing.zip](https://github.com/espressif/esp-adf/files/14320881/libesp_processing.zip)

espradio, 2024-02-18:
Issue log changes, i.e new issue. Looks it is not the audio is not AAC frame

name=ADF_DEBUG url=https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/playlist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/playlist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/chunklist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/media_2683.ts
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/media_2684.ts
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/media_2685.ts
E (32823) MPEG_READER: Not find an effective frame, line:758.
E (32823) MPEG_READER: Resync error (line 411)
E (32823) AAC_DECODER: Encountered error reading stream
E (32830) AUDIO_ELEMENT: [aac] AEL_STATUS_ERROR_OPEN,-1
W (32834) AUDIO_ELEMENT: [aac] audio_element_on_cmd_error,7

TempoTian, 2024-02-18:
Have you use the wrapper decode in [libesp_processing.zip](https://github.com/espressif/esp-adf/files/14320881/libesp_processing.zip)
```
   aac_decoder_cfg_t aac_cfg = DEFAULT_AAC_DECODER_CONFIG();
    aac_decoder = wrapper_dec_init(&aac_cfg);
    //aac_decoder = aac_decoder_init(&aac_cfg);
```

espradio, 2024-02-18:
It is **working** now. Thanks a lot. How do you suggest to integrate this? i.e
a)Will ADF updated with new codec changes ?
b)Will application need wrapper audio element? 

~~~~~ Playing->0 name=ADF_DEBUG url=https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/playlist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/playlist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/chunklist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio214/media_3032.ts
    47 40 00 17 00 00 b0 0d 00 01 c1 00 00 00 01 ef ff 36 90 e2 3d ff ff ff ff ff ff ff ff ff ff ff
    ff fb 70 c0 c1 80 10 ad 35 5d 2c 3d 0d b2 1c 9f 6b 64 f7 bd 76 59 97 1c f2 e6 5b ef 57 72 55 7a

espradio, 2024-02-18:
Also wrapper logic still needs improvement as following streams is still has issue(no error, but not playing)

~~~~~ Playing->4 name=AIR Mahanadi Sambalpur url=https://air.pc.cdn.bitgravity.com/air/live/pbaudio182/playlist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio182/playlist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio182/chunklist.m3u8
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio182/media_5384.ts
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio182/media_5385.ts
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio182/media_5386.ts
URL: https://air.pc.cdn.bitgravity.com/air/live/pbaudio182/media_5387.ts

~~~~~ Playing->5 name=IRIB_Ava url=http://s1.cdn1.iranseda.ir:1935/liveedge/radio-nama-ava/playlist.m3u8
URL: http://s1.cdn1.iranseda.ir:1935/liveedge/radio-nama-ava/playlist.m3u8
URL: http://s1.cdn1.iranseda.ir:1935/liveedge/radio-nama-ava/chunklist_w1326855632.m3u8
URL: http://s1.cdn1.iranseda.ir:1935/liveedge/radio-nama-ava/media_w1326855632_5552.ts

TempoTian, 2024-02-18:
I have tested both the 2 manifest urls, although network is pool, it can play. If you see log like below it means decode OK:
```
Receive music info from aac decoder, sample_rates=44100, bits=16, ch=1
```
Old TS parse only support filter out AAC audio, so need enhance TS parser in `esp_codec` to add MP3 support, .
Also to support more than 2 decoder automatically for HLS, wrapper decoder is necessary,  it is used to parse data and select the decoder actually used.
 We will research and study how to implement it, current patch version is simple but enough, if there are other issues, I think it is easy to modify the wrapper code to fix it.

espradio, 2024-02-18:
>I have tested both the 2 manifest urls, although network is pool, it can play. If you see log like below it means decode OK:

Got the reason why it not playing. In my application, it is playlist driven app. if I move the stream as first then it working. Looks like some clean up missing in wrapper codec. 

>We will research and study how to implement it, current patch version is simple but enough

OK. Let me know the path after the research. In the mean time, it will be good if esp-adf-lib updated for "Old TS parse only support filter out AAC audio, so need enhance TS parser in esp_codec to add MP3 support"

Thanks

espradio, 2024-02-23:
Any update on making esp-adf-lib changes to mainstream?

TempoTian, 2024-03-06:
Merged on commit: https://github.com/espressif/esp-adf-libs/commit/38952f00560d8999ca34fdc364cf13e00679ec5b

espradio, 2024-03-07:
Thanks @TempoTian

espradio, 2024-03-09:
@TempoTian Could you expose following API is esp-adf-lib header file with proper documentation so that my application can use this properly.

int ts_parser_search_stream(unsigned char *buf, int len);
int ts_parser_dec_init(void *user_data);
int ts_parser_dewrapper_prefilled_data(void *codec_handle, int datanum, void *ts_buf, unsigned char *data);
int ts_parser_read_data(void *codec_handle, int bytes, void *ts_buf, unsigned char *buf);
void ts_parser_dec_deinit(void *user_data);

espradio, 2024-03-09:
Also who is allocating memory for  mpeg_buffer_t::ts_buf in above wrapper codec

espradio, 2024-03-09:
Also, These APIs internally using audio_element_handle_t which is threaded in nature? Is there version like standalone(example alc_volume_setup_process() where is does not need a thread).

TempoTian, 2024-03-12:
We will add extractor module in near feature to support container parse(TS, MP4 etc), so we not decide to export these API to be public. I add simple comment of the code as below.  Just think TS parser as a filter, input TS data and output audio data with internal PES cache.
```c
/* Search from buffer to check whether it is a TS stream, return TS stream matched position */
int ts_parser_search_stream(unsigned char *buf, int len);

/* Init TS parser, TS parser only use its member mpeg_buffer_t::ts_buf, ts_buf is deinit in `ts_parser_dec_deinit` */
int ts_parser_dec_init(void *user_data);

/*Prefill `datanum` data through `data`, it will try to read from input `codec_handle` element input ringfifo until parsed data size reach `datanum` and return data size actually parsed */
int ts_parser_dewrapper_prefilled_data(void *codec_handle, int datanum, void *ts_buf, unsigned char *data);

/* Read` bytes` data and save into `buf` from parser, it will try to read from `codec_handle` element and parse to return  the actually parsed data size*/
int ts_parser_read_data(void *codec_handle, int bytes, void *ts_buf, unsigned char *buf);

/* Deinitialize for TS parser*/
void ts_parser_dec_deinit(void *user_data);
```

espradio, 2024-03-16:
Thanks @TempoTian 

>We will add extractor module in near feature to support container parse(TS, MP4 etc),

Looking forward to it.

closed 2025-02-17 by jason-mao