espressif/esp-adf #1146: player/pipeline_hfp_stream example I2S configuration on LyraT 4.3 partially fails

smishra123, 2024-02-06:
In the player/pipeline_hfp_stream the default I2S configuration partially fails on LyraT 4.3.  I am able to get the speaker output on my LyraT 4.3 board but not the microphone input.

The message I get in the UART log is 

 (1453) AUDIO_HAL: Codec mode is 3, Ctrl:1
I (1463) HFP_EXAMPLE: [ 3 ] Create audio pipeline for playback
I (1463) HFP_EXAMPLE: [3.1] Create i2s stream to write data to codec chip and read data from codec chip
**E (1483) i2s(legacy): i2s_check_cfg_validity(926): this i2s port is in use
E (1483) i2s(legacy): i2s_driver_install(1582): I2S configuration is invalid**

I am using IDF version
I (561) cpu_start: ESP-IDF:          **v5.1.2-dirty**

and ADF version 
**esp-adf-v2.6**

smishra123, 2024-02-06:
Using it v4.4 of IDF referenced in the ADF gives a different error

**E (1492) I2S: register I2S object to platform failed**

no sound or microphone output comes out

---------------------------------
Log

I (77) boot:  2 factory          factory app      00 00 00010000 002dc6c0
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2a728h (173864) map
I (160) esp_image: segment 1: paddr=0003a750 vaddr=3ffbdb60 size=0404ch ( 16460) load
I (166) esp_image: segment 2: paddr=0003e7a4 vaddr=40080000 size=01874h (  6260) load
I (169) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a941ch (693276) map
I (425) esp_image: segment 4: paddr=000e9444 vaddr=40081874 size=1825ch ( 98908) load
I (478) boot: Loaded app from partition at offset 0x10000
I (478) boot: Disabling RNG early entropy source...
I (490) cpu_start: Pro cpu up.
I (490) cpu_start: Starting app cpu, entry point is 0x40081394
I (0) cpu_start: App cpu up.
I (506) cpu_start: Pro cpu start user code
I (506) cpu_start: cpu freq: 160000000
I (506) cpu_start: Application information:
I (511) cpu_start: Project name:     pipeline_hfp_stream_example
I (518) cpu_start: App version:      v2.6-76-gf1cd5b6a-dirty
I (524) cpu_start: Compile time:     Feb  6 2024 15:28:49
I (530) cpu_start: ELF file SHA256:  76f06e4fd5756d5f...
**I (536) cpu_start: ESP-IDF:          v4.4.4-278-g3c8bc2213c**
I (542) cpu_start: Min chip rev:     v0.0
I (547) cpu_start: Max chip rev:     v3.99
I (552) cpu_start: Chip rev:         v3.1
I (557) heap_init: Initializing. RAM available for dynamic allocation:
I (564) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (570) heap_init: At 3FFB7468 len 00000B98 (2 KiB): DRAM
I (576) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (582) heap_init: At 3FFCA358 len 00015CA8 (87 KiB): DRAM
I (588) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (594) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (601) heap_init: At 40099AD0 len 00006530 (25 KiB): IRAM
I (609) spi_flash: detected chip: gd
I (611) spi_flash: flash io: dio
W (615) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (630) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (662) HFP_EXAMPLE: [ 1 ] init Bluetooth
I (662) BTDM_INIT: BT controller compile version [2c56073]
I (672) system_api: Base MAC address is not set
I (672) system_api: read default base MAC address from EFUSE
I (682) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1402) HFP_EXAMPLE: [ 2 ] Start codec chip
I (1402) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (1432) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1432) ES8388_DRIVER: init,out:02, in:00
I (1442) AUDIO_HAL: Codec mode is 3, Ctrl:1
I (1442) HFP_EXAMPLE: [ 3 ] Create audio pipeline for playback
I (1442) HFP_EXAMPLE: [3.1] Create i2s stream to write data to codec chip and read data from codec chip
I (1452) HFP_EXAMPLE: [3.1.1] Reader
I (1462) I2S: APLL expected frequency is 8192000 Hz, real frequency is 8191999 Hz
I (1462) I2S: DMA Malloc info, datalen=blocksize=600, dma_buf_count=3
I (1472) I2S: DMA Malloc info, datalen=blocksize=600, dma_buf_count=3
I (1482) I2S: I2S0, MCLK output by GPIO0
I (1482) HFP_EXAMPLE: [3.1.2] Writer
**E (1492) I2S: register I2S object to platform failed**
I (1492) I2S: I2S0, MCLK output by GPIO0
I (1502) HFP_EXAMPLE: [3.2] Get hfp stream
I (1502) HFP_STREAM: outgoing stream init
I (1512) HFP_STREAM: incoming stream init
I (1512) HFP_EXAMPLE: [3.2] Register all elements to audio pipeline
I (1522) HFP_EXAMPLE: [3.3] Link it together [Bluetooth]-->hfp_in_stream-->i2s_stream_writer-->[codec_chip]
I (1532) AUDIO_PIPELINE: link el->rb, el:0x3ffd5ee4, tag:incoming, rb:0x3ffd6288
I (1542) AUDIO_PIPELINE: link el->rb, el:0x3ffd54c4, tag:i2s_reader, rb:0x3ffd83d8
I (1542) HFP_EXAMPLE: [ 4 ] Initialize peripherals
E (1552) gpio: gpio_install_isr_service(449): GPIO isr service already installed
I (1562) HFP_EXAMPLE: [ 5 ] Set up  event listener
I (1562) HFP_EXAMPLE: [5.1] Listening event from all elements of pipeline
I (1572) HFP_EXAMPLE: [5.2] Listening event from peripherals
I (1582) HFP_EXAMPLE: [ 6 ] Listen for all pipeline events

jason-mao, 2024-04-07:
This is an old I2S driver compatible issue. We have fixed on https://github.com/espressif/esp-adf/blob/master/components/audio_stream/i2s_stream_idf5.c.

closed 2024-04-07 by jason-mao