espressif/esp-adf #1149: google tts playback fails on LyraT-Mini (interrupt watchdog triggered)

PrathamG, 2024-02-08:
I have been trying to fix this audio out issue for days now but it won't work. Any help and suggestions will be appreciated. Thanks!

## Environment

- Audio development kit: ESP32-LyraT-Mini
- Module or chip used:  ESP32-WROVER-E
- IDF version: v5.0.0
- ADF version: v2.6
- Build system: idf.py
- Operating system: Linux
- Using an IDE?: VSCode
- Power supply: USB | external 5V

## Problem Description

I'm trying to run a modified version of the [google_translate_device](https://github.com/espressif/esp-adf/tree/master/examples/cloud_services/google_translate_device) example on LyraT-Mini-1.2 . When I receive the response from google_sr, I parse the response text and then send it to google_tts. When I send the API request to google tts, there is no audio played and I get an error saying the interrupt watchdog got triggered. 

This is the the modified task that I'm running:
```cpp
void event_process_Task(void *pv)
{       
    audio_event_iface_msg_t msg;

    while (1) {
        if(audio_event_iface_listen(evt_listener, &msg, portMAX_DELAY) != ESP_OK) {
            ESP_LOGE(TAG, "[ * ] Event process failed: src_type:%d, source:%p cmd:%d, data:%p, data_len:%d",
                     msg.source_type, msg.source, msg.cmd, msg.data, msg.data_len);
            continue;
        }

        if(google_tts_check_event_finish(tts, &msg)) {
            ESP_LOGI(TAG, "[ * ] TTS Finish");
            continue;
        }
        
        ESP_LOGI(TAG, "[ * ] Event received: src_type:%d, source:%p cmd:%d, data:%p, data_len:%d", msg.source_type, msg.source, msg.cmd, msg.data, msg.data_len);

        if ((msg.source_type == PERIPH_ID_TOUCH || msg.source_type == PERIPH_ID_BUTTON || msg.source_type == PERIPH_ID_ADC_BTN)) {
            if((int)msg.data == get_input_rec_id()) {
                if(msg.cmd == PERIPH_BUTTON_PRESSED) {
                    google_tts_stop(tts);
                    ESP_LOGI(TAG, "[ * ] Resuming SR pipeline");
                    google_sr_start(sr);
                } 
                else if(msg.cmd == PERIPH_BUTTON_RELEASE || msg.cmd == PERIPH_BUTTON_LONG_RELEASE){
                    ESP_LOGI(TAG, "[ * ] Stop SR pipeline");

                    char* response_text = google_sr_stop(sr);
                    if (response_text == NULL) {
                        continue;
                    }
                    ESP_LOGI(TAG, "response text = %s", response_text);
                    vTaskDelay(2000 / portTICK_PERIOD_MS);
                    ESP_LOGI(TAG, "TTS Start");
                    google_tts_start(tts, response_text, GOOGLE_TTS_LANG);   
                } 
                else if ((int)msg.data == get_input_mode_id()) {
                    ESP_LOGI(TAG, "Mode button was pressed, exit now");
                    break;
                }
            }
        }
    }

    ESP_LOGI(TAG, "[ 6 ] Stop audio_pipeline");
    google_sr_destroy(sr);
    google_tts_destroy(tts);
    // Stop all periph before removing the listener 
    esp_periph_set_stop_all(periph_set);
    audio_event_iface_remove_listener(esp_periph_set_get_event_iface(periph_set), evt_listener);
[... 269 lines cut ...]

0x4009307d: vPortTaskWrapper at /home/hansonrobotics/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

Core  1 register dump:
PC      : 0x4008cdf6  PS      : 0x00060b34  A0      : 0x800d3baa  A1      : 0x3ffbbd70  
0x4008cdf6: esp_cpu_wait_for_intr at /home/hansonrobotics/esp/esp-idf/components/esp_hw_support/cpu.c:110

A2      : 0x00000000  A3      : 0x05839baa  A4      : 0x00000001  A5      : 0x003fffff  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800f384e  A9      : 0x3ffbbd30  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffb8f30  A13     : 0x3ffb8f10  
A14     : 0x40088354  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
0x40088354: ipc_task at /home/hansonrobotics/esp/esp-idf/components/esp_system/esp_ipc.c:54

EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4008cdf3:0x3ffbbd70 0x400d3ba7:0x3ffbbd90 0x40090e4d:0x3ffbbdb0 0x4009307d:0x3ffbbdd0
0x4008cdf3: xt_utils_wait_for_intr at /home/hansonrobotics/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /home/hansonrobotics/esp/esp-idf/components/esp_hw_support/cpu.c:101

0x400d3ba7: esp_vApplicationIdleHook at /home/hansonrobotics/esp/esp-idf/components/esp_system/freertos_hooks.c:59

0x40090e4d: prvIdleTask at /home/hansonrobotics/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4197 (discriminator 1)

0x4009307d: vPortTaskWrapper at /home/hansonrobotics/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```

PrathamG, 2024-02-20:
Hi! Any updates or suggestions?

PrathamG, 2024-02-29:
Hi, any updates?

jason-mao, 2024-02-29:
Hi @PrathamG, I have not reproduced your case. But I saw `HTTP_STREAM: total_bytes=0`, which should not be there. Would you please check why the URL data length is 0?

jason-mao, 2024-08-01:
@PrathamG Did have any progress of this issue?

closed 2025-02-17 by jason-mao