espressif/esp-adf #1152: No hookup(answer) event from VoIP module when the peer receives/answers a phone call.

EdwardRyu, 2024-02-15:
**SW/HW versions**
Audio development kit: ESP32-LyraT
Audio kit version : v4.3
Module or chip used: ESP32-WROVER-E
IDF version: v4.4.4
ADF version: v2.5

**Issue**
No hookup event from VoIP module when the peer receives/answers a phone call.

**Detail**
I don't see any hook up event defined and supported by VoIP module.
Could you help us get the event when the called party answers the call?

Thanks,
Edward

EdwardRyu, 2024-02-22:
Hi Michael,

I have changed the configuration from server side(Twilio) and but media library still sends an event at a wrong time.
Here is a detail.
If you look at the snippet of the log, you can find that ESP_RTC_EVENT_AUDIO_SESSION_BEGIN event just comes up when  (SIP/2.0 183 Session progress) is received.

20 seconds later, the peer users answered the phone and modem received SIP/2.0 200 OK, but no event occurred.

I think ESP_RTC_EVENT_AUDIO_SESSION_BEGIN event should come up after receiving SIP/2.0 200 OK.

Attached log, please have a look into it and then get back to me.

Thanks,
Edward

[2024-02-22 09:30:03.277] D (59721) SIP_SERVICE: _esp_sip_event_handler event 1

[2024-02-22 09:30:15.965] Call-ID: 6F49555B95525C5A2EE1B3D181B8D3B452D899350AD0
[2024-02-22 09:30:15.965] CSeq: 3 INVITE

[2024-02-22 09:30:16.103] Call-ID: 6F49555B95525C5A2EE1B3D181B8D3B452D899350AD0
[2024-02-22 09:30:16.117] CSeq: 3 INVITE

[2024-02-22 09:30:16.272] SIP/2.0 180 Ringing
[2024-02-22 09:30:16.272] CSeq: 3 INVITE

[2024-02-22 09:30:16.272] D (72687) SIP_SERVICE: _esp_sip_event_handler event 3
[2024-02-22 09:30:16.272] I (72688) SIP_SERVICE: ESP_RTC_EVENT_CALLING Remote Ring...

[2024-02-22 09:30:18.665] SIP/2.0 183 Session progress
[2024-02-22 09:30:18.665] CSeq: 3 INVITE

[2024-02-22 09:30:18.679] D (75144) SIP_SERVICE: _esp_sip_event_handler event 7
[2024-02-22 09:30:18.679] I (75144) SIP_SERVICE: ESP_RTC_EVENT_AUDIO_SESSION_BEGIN

[2024-02-22 09:30:39.216] SIP/2.0 200 OK
[2024-02-22 09:30:39.216] CSeq: 3 INVITE

[2024-02-22 09:30:39.229] Call-ID: 6F49555B95525C5A2EE1B3D181B8D3B452D899350AD0
[2024-02-22 09:30:39.229] CSeq: 3 ACK

ahhfzhang, 2024-02-28:
Hi Edward,

Got it, I will build a library for you and add callback this 200 event to user.

Thanks.

ahhfzhang, 2024-02-28:
Hi Edward,

This [esp_media_protocols.zip](https://github.com/espressif/esp-adf/files/14429966/esp_media_protocols.zip)
library you can replace into adf-libs, and you can receive the `ESP_RTC_EVENT_ON_CALL` event in the user callback.

Thanks.

EdwardRyu, 2024-03-01:
Hi Michael,

Now we can receive pickup event at the right time but we have a huge drawback.
I have observed two critical problems as follows.

1) Not able to end the call.
- Make an outgoing call from ESP32 to other smartphone.
- Call is connected.
- ESP32 ends the call.
- Call doesn't get terminated.

2) Crash when ending the call in a below procedure.
- Receive and answer the incoming call.
- Call is connected.
- ESP32 ends the call.
- ESP32 gets crashed.

When I look at the size of library, I found that new one is much smaller than I am using as attached screenshot.
I wonder if you build the library on top of base line (ESP32, IDFv4.4.4, ADFv2.5) as mentioned at the beginning.
We also applied some patches for the issues ([issues#991](https://github.com/espressif/esp-adf/issues/991) &  [issues#1011](https://github.com/espressif/esp-adf/issues/1011)), but I wonder if they are also applied in the new library.

[image]

I will provide the logs later tomorrow, but in the meantime please have a look into this build related issues.

Thanks,
Edward

ahhfzhang, 2024-03-01:
Hi Edward,

I built a new [libesp_media_protocols.zip](https://github.com/espressif/esp-adf/files/14458441/libesp_media_protocols.zip)
 for (ESP32, IDFv4.4.4, ADFv2.5), but it is still this size. 
I have tested the two cases you mentioned and it is still ok.

Thanks.

EdwardRyu, 2024-03-07:
Hi Michael,

Now it works fine.

Thanks for your support.

closed 2024-03-07 by EdwardRyu