espressif/esp-adf #1155: How to build my project using esp-idf 5.1?

FlyyingPiggy2020, 2024-02-21:
i have a project use esp-idf 5.1.1.it works great.

now i want to add some audio functionality.

is that esp-adf(v2.6) seems to contain a entire copy of esp-idf (v4.4) within it?

How can I use IDF5.5 to compile my project?

TempoTian, 2024-02-22:
For easy usage you can use follow method:
```
1: cd your IDF5.1 PATH
   . ./export.sh
2: export ADF_PATH env
    export ADF_PATH=****/esp-adf-internal
3：cd your project folder and build, to use IDF 5.1 make sure enable option and patch for IDF use 
      https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v5.1_freertos.patch
     CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY
```

FlyyingPiggy2020, 2024-02-22:
Thank you, I compiled `/esp-adf/examples/get-started/play_mp3_control` successfully.