espressif/esp-adf #1164: dueros例程跑不了求解决

liuliuu-112, 2024-03-06:
PS C:\Users\15854\Desktop\dueros> set IDF_PATH=C:/ESP/espressif/.espressif/frameworks/esp-idf-v4.4.4/
PS C:\Users\15854\Desktop\dueros> C:/ESP/espressif/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe C:\ESP\espressif\.espressif\frameworks\esp-idf-v4.4.4\tools\idf_monitor.py -p COM6 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 c:\Users\15854\Desktop\dueros\build\esp_dueros.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- idf_monitor on \\.\COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7016
load:0x40078000,len:15676
ho 0 tail 12 room 4
load:0x40080400,len:4516
0x40080400: _init at ??:?

entry 0x400806b8
I (29) boot: ESP-IDF v4.4.4-dirty 2nd stage bootloader
I (29) boot: compile time 15:25:19
I (29) boot: chip revision: v3.0
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (50) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 8MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00310000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 3, min. application chip revision: 0
I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=96508h (615688) map
I (287) esp_image: segment 1: paddr=000a6530 vaddr=3ffb0000 size=04258h ( 16984) load
I (292) esp_image: segment 2: paddr=000aa790 vaddr=40080000 size=05888h ( 22664) load
I (300) esp_image: segment 3: paddr=000b0020 vaddr=400d0020 size=153790h (1390480) map
I (684) esp_image: segment 4: paddr=002037b8 vaddr=40085888 size=12f50h ( 77648) load
I (722) boot: Loaded app from partition at offset 0x10000
I (722) boot: Disabling RNG early entropy source...
I (733) psram: This chip is ESP32-D0WD
I (734) spiram: Found 64MBit SPI RAM device
I (734) spiram: SPI RAM mode: flash 80m sram 80m
I (739) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (746) cpu_start: Pro cpu up.
I (750) cpu_start: Starting app cpu, entry point is 0x40081758
0x40081758: call_start_cpu1 at C:/ESP/espressif/.espressif/frameworks/esp-idf-v4.4.4/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (1232) spiram: SPI SRAM memory test OK
I (1240) cpu_start: Pro cpu start user code
I (1240) cpu_start: cpu freq: 240000000
I (1240) cpu_start: Application information:
I (1243) cpu_start: Project name:     esp_dueros
I (1248) cpu_start: App version:      1
I (1253) cpu_start: Compile time:     Mar  6 2024 15:26:31
I (1259) cpu_start: ELF file SHA256:  b09e5a184f53e11d...
I (1265) cpu_start: ESP-IDF:          v4.4.4-dirty
I (1271) heap_init: Initializing. RAM available for dynamic allocation:
I (1278) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1284) heap_init: At 3FFB7478 len 00028B88 (162 KiB): DRAM
I (1290) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1297) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1303) heap_init: At 400987D8 len 00007828 (30 KiB): IRAM
I (1310) spiram: Adding pool of 4084K of external SPI memory to heap allocator
I (1318) spi_flash: detected chip: gd
I (1322) spi_flash: flash io: qio
I (1327) cpu_start: Starting scheduler on PRO CPU.
[... 212 lines cut ...]
Please switch to ESP32-S3 and wakenet9 !
I (12327) AFE_SR: wake num: 1, mode: 1, (Nov  1 2022 17:16:15)

W (12327) RECORDER_SR: Multinet is not enabled in SDKCONFIG
I (12327) AUDIO_THREAD: The feed_task task allocate stack on external memory
E (12337) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch" first

E (12357) AUDIO_THREAD: Error creating RestrictedPinnedToCore feed_task
I (12357) AUDIO_THREAD: The fetch_task task allocate stack on external memory
E (12367) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch" first

E (12387) AUDIO_THREAD: Error creating RestrictedPinnedToCore fetch_task
I (12397) AUDIO_THREAD: The recorder_task task allocate stack on external memory
E (12407) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch" first

E (12427) AUDIO_THREAD: Error creating RestrictedPinnedToCore recorder_task
E (12427) AUDIO_RECORDER: Fail to create recorder_task
I (12437) AUDIO_THREAD: The voice_read_task task allocate stack on external memory
E (12447) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch" first

E (12467) AUDIO_THREAD: Error creating RestrictedPinnedToCore voice_read_task

TempoTian, 2024-03-14:
可以按照log的提示先打一下patch
git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch
如果不会打的话可以把那个函数直接copy到freertos/task.c里面去

liuliuu-112, 2024-03-16:
> 可以按照log的提示先打一下patch git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch 如果不会打的话可以把那个函数直接copy到freertos/task.c里面去 路径: https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v4.4_freertos.patch
[image]
[image]
已经打包了，还是提醒

liuliuu-112, 2024-03-19:
> 可以按照log的提示先打一下patch git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch 如果不会打的话可以把那个函数直接copy到freertos/task.c里面去 路径: https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v4.4_freertos.patch

是只复制这个函数，是吧，大神

BaseType_t xTaskCreateRestrictedPinnedToCore( const TaskParameters_t * const pxTaskDefinition,
                                                  TaskHandle_t *pxCreatedTask,
                                                  const BaseType_t xCoreID)
	{
	TCB_t *pxNewTCB;
	BaseType_t xReturn = errCOULD_NOT_ALLOCATE_REQUIRED_MEMORY;

		configASSERT( pxTaskDefinition->puxStackBuffer );

		if( pxTaskDefinition->puxStackBuffer != NULL )
		{
			/* Allocate space for the TCB.  Where the memory comes from depends
+			on the implementation of the port malloc function and whether or
+			not static allocation is being used. */
			pxNewTCB = ( TCB_t * ) pvPortMallocTcbMem( sizeof( TCB_t ) );

			if( pxNewTCB != NULL )
			{
				/* Store the stack location in the TCB. */
				pxNewTCB->pxStack = pxTaskDefinition->puxStackBuffer;

				/* Tasks can be created statically or dynamically, so note
+				this task had a statically allocated stack in case it is
+				later deleted.  The TCB was allocated dynamically. */
				pxNewTCB->ucStaticallyAllocated = tskDYNAMICALLY_ALLOCATED_STACK_AND_TCB;

				prvInitialiseNewTask(	pxTaskDefinition->pvTaskCode,
										pxTaskDefinition->pcName,
										pxTaskDefinition->usStackDepth,
										pxTaskDefinition->pvParameters,
										pxTaskDefinition->uxPriority,
										pxCreatedTask, pxNewTCB,
										pxTaskDefinition->xRegions,
										xCoreID );

				prvAddNewTaskToReadyList( pxNewTCB, pxTaskDefinition->pvTaskCode, xCoreID );
				xReturn = pdPASS;
			}
		}

		return xReturn;
	}

jason-mao, 2024-04-07:
@liuliuu-112 你的问题怎么样了？