espressif/esp-adf #1167: AEL_MSG_CMD_REPORT_MUSIC_INFO not updating correct MUSIC info

espradio, 2024-03-10:
When AEL_MSG_CMD_REPORT_MUSIC_INFO generates, it generate wrong music info(it is 32bit and Ch = 1) sometimes.
Looks like MP3 decoder able to detect later (See MP3_DECODER: Music information changed message), What is AEL_MSG to detects this so that downstream component can be properly reconfigured at runtime.

log

```
name=98.8 Buddu Fm url=https://dc4.serverse.com/proxy/ccmxrgub/stream
URL: https://dc4.serverse.com/proxy/ccmxrgub/stream
mp3 info, sample_rates=48000, bits=16, ch=2
W (3165176) MP3_DECODER: Music information changed. sample_rate = 44100 : channels = 1
```
code

```
	case AEL_MSG_CMD_REPORT_MUSIC_INFO: {
		audio_element_info_t info = {0};
		audio_element_getinfo(msg->source, &info);
		printf("%s info, sample_rates=%d, bits=%d, ch=%d", "mp3", info.sample_rates, info.bits, info.channels);
```

Code versions:

esp-adf
```
commit 33ba0c5c698b6719e0cd8e2bd97d8e4c5a13d9d2 (HEAD -> master, origin/master, origin/HEAD)
Merge: 1b9d7ef2 96c1a724
Author: Jason-mao <maojianxin@espressif.com>
Date:   Fri Feb 23 18:12:40 2024 +0800

    Merge branch 'feature/support_aw88298_for_codec_dev' into 'master'

    esp_codec_dev: Support aw88298 audio amplifier

    See merge request adf/esp-adf-internal!1277
```
esp-adf-libs

```
commit f76dbe6a4bfc9d581d6dea758555bda66abba1e0 (HEAD -> master, origin/master, origin/HEAD)
Merge: 557d100 9994dcf
Author: Jason-mao <maojianxin@espressif.com>
Date:   Thu Mar 7 15:20:54 2024 +0800

    Merge branch 'bugfix/fix_mp3_not_skip_info_tag_frame_and_pad_data' into 'master'

    Fix mp3 not skip info tag frame and pad data

    See merge request adf/esp-adf-libs!249
```

TempoTian, 2024-03-14:
Now all decoder only support 16bits output, I think better to provide some log to check where the 32bits report from. 
I test the given stream, it always reports 44.1k, 16bits, 1channel.

espradio, 2024-03-16:
The output from esp-adf-lib 
W (3165176) MP3_DECODER: Music information changed. sample_rate = 44100 : channels = 1
VLC codec stats https://dc4.serverse.com/proxy/ccmxrgub/stream show it is 32bit, 1 channel

jason-mao, 2024-04-07:
@espradio I have checked your file, which is 44.1k, 16bits, and 1 channel, too. Did you get any audio quality issues on this stream? You know, in principle, the MP3 encoder does not support encoding 32-bit data.

jason-mao, 2024-07-31:
@espradio Do you have any updates for sharing？