espressif/esp-adf #1168: Runtime error in VoIP example in an ESP32 -LyraT-Mini_V1.2 :    ***ERROR*** A stack overflow in task OUT_iis has been detected

migmelccs, 2024-03-12:
Hello, I have been trying to run the VoIP example and cannot success in make it to behave as it is suppose to. First I used ESP-IDF v5.2.1 combined with ESP-ADF master. There is a runtime problem because of mixing between deprecated i2s.h and the new headers  i2s_std.h, etc.,  that depend on the framework and there is nothing I can do, unless I modify the ESP-IDF files. Looks like ESP-IDF uses the new i2s headers and ESP-ADF (stream.h or so) still uses the deprecated one.
Then I had to moved to ESP-IDF v4.4.7 combined with esp-adf-master, followed all the recommended in the README file, but a new problem arose (as shown in the title) even when I applied the recommended patch **idf_v4.4_freertos.patch** as suggested.
I have been able to make calls using FreeSWITCH and Zoiper client for testing, however, the device keeps rebooting so frequently that is impossible to make it stable.
please help me to solve this issue. 
Here are the logs. I will provide any other information if needed:

E:\idf-v4.4.7\esp-adf-master\examples\protocols\voip0>idf.py -p com12 monitor
Executing action: monitor
Running idf_monitor in directory E:\idf-v4.4.7\esp-adf-master\examples\protocols\voip0
Executing "E:\tools-v4.4.7\python_env\idf4.4_py3.11_env\Scripts\python.exe E:\idf-v4.4.7\tools/idf_monitor.py -p com12 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 E:\idf-v4.4.7\esp-adf-master\examples\protocols\voip0\build\voip_app.elf -m 'E:\tools-v4.4.7\python_env\idf4.4_py3.11_env\Scripts\python.exe' 'E:\idf-v4.4.7\tools\idf.py' '-p' 'com12'"...
--- idf_monitor on com12 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
D:0,�ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7004
ho 0 tail 12 room 4
load:0x40078000,len:15784
ho 0 tail 12 room 4
load:0x40080400,len:4536
0x40080400: _init at ??:?

entry 0x400806bc
I (31) boot: ESP-IDF v4.4.7-dirty 2nd stage bootloader
I (31) boot: compile time 18:00:31
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 8MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00200000
I (96) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000
I (104) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=35dd0h (220624) map
I (177) esp_image: segment 1: paddr=00045df8 vaddr=3ffb0000 size=04b50h ( 19280) load
I (183) esp_image: segment 2: paddr=0004a950 vaddr=40080000 size=056c8h ( 22216) load
I (191) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=f0d80h (986496) map
I (463) esp_image: segment 4: paddr=00140da8 vaddr=400856c8 size=11578h ( 71032) load
I (498) boot: Loaded app from partition at offset 0x10000
I (498) boot: Disabling RNG early entropy source...
I (510) cpu_start: Multicore app
I (510) psram: This chip is ESP32-D0WD
I (510) spiram: Found 64MBit SPI RAM device
I (512) spiram: SPI RAM mode: flash 80m sram 80m
I (517) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (524) cpu_start: Pro cpu up.
I (528) cpu_start: Starting app cpu, entry point is 0x40081534
0x40081534: call_start_cpu1 at E:/idf-v4.4.7/components/esp_system/port/cpu_start.c:151

I (0) cpu_start: App cpu up.
I (1030) spiram: SPI SRAM memory test OK
I (1039) cpu_start: Pro cpu start user code
I (1039) cpu_start: cpu freq: 240000000
I (1039) cpu_start: Application information:
I (1042) cpu_start: Project name:     voip_app
I (1047) cpu_start: App version:      v2.6-82-g33ba0c5c-dirty
I (1053) cpu_start: Compile time:     Mar 11 2024 17:59:48
I (1059) cpu_start: ELF file SHA256:  359700a4c2370f47...
I (1065) cpu_start: ESP-IDF:          v4.4.7-dirty
I (1071) cpu_start: Min chip rev:     v0.0
I (1076) cpu_start: Max chip rev:     v3.99 
[... 240 lines cut ...]
I (6605) AUDIO_THREAD: The Audio_forge task allocate stack on external memory
I (6613) AUDIO_FORGE: audio_forge opened
I (6613) AUDIO_THREAD: The OUT_iis task allocate stack on external memory
I (6625) I2S_STREAM: AUDIO_STREAM_WRITER
I (6629) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f806a70] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING

***ERROR*** A stack overflow in task OUT_iis has been detected.

Backtrace: 0x400821fd:0x3f830d60 0x40090ac9:0x3f830d80 0x400911c5:0x3f830da0 0x400922f9:0x3f830e20 0x40090bc4:0x3f830e40 0x40090b76:0xbf5d3838 |<-CORRUPTED
0x400821fd: panic_abort at E:/idf-v4.4.7/components/esp_system/panic.c:408

0x40090ac9: esp_system_abort at E:/idf-v4.4.7/components/esp_system/esp_system.c:137

0x400911c5: vApplicationStackOverflowHook at E:/idf-v4.4.7/components/freertos/port/xtensa/port.c:407

0x400922f9: vTaskSwitchContext at E:/idf-v4.4.7/components/freertos/tasks.c:3664

0x40090bc4: _frxt_dispatch at E:/idf-v4.4.7/components/freertos/port/xtensa/portasm.S:436

0x40090b76: _frxt_int_exit at E:/idf-v4.4.7/components/freertos/port/xtensa/portasm.S:231

ELF file SHA256: 359700a4c2370f47

Rebooting...
ets Jul 29 2019 12:21:46

TempoTian, 2024-03-14:
Seems you have changed the source code, default example use `av_stream ` to do audio output.
 Log show that i2s_stream task stack over flow, you can adjust it according following codes:
```c
    i2s_stream_cfg_t i2s_writer = I2S_STREAM_CFG_DEFAULT();
    i2s_writer.i2s_config.sample_rate = 48000;
    i2s_writer.i2s_config.mode = I2S_MODE_MASTER | I2S_MODE_TX;
    i2s_writer.type = AUDIO_STREAM_WRITER;
    i2s_writer.task_stack = NEW_STACK_SIZE; (such as 4k or more)
    i2s_h = i2s_stream_init(&i2s_writer);
    esp_audio_output_stream_add(player, i2s_h);
```

closed 2024-03-14 by migmelccs

migmelccs, 2024-03-14:
Hello, thank you,
I'll look at it.
I have not modified this example, other than the SIP user, SIP server IP,
WiFi credentials but nothing related to I2S. Actually this is the example
packet with esp adf v2.6 and master

Miguel

TempoTian, 2024-03-15:
Sorry，forget that audio tone also use audio player, you can modify it in code:
https://github.com/espressif/esp-adf/blob/master/examples/protocols/components/audio_flash_tone/audio_player_int_tone.c#L91

migmelccs, 2024-03-15:
Yes, I follow the error to this file, however no matter how big I make I2S_STREAM_TASK_STACK, it still sends the message ***ERROR*** A stack overflow in task OUT_iis has been detected. 

And get stuck in this last lines:
I (3615) AUDIO_THREAD: The OUT_iis task allocate stack on external memory
I (3627) I2S_STREAM: AUDIO_STREAM_WRITER
I (3631) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f806a70] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING