espressif/esp-adf #1169: When running VoIP example in LyraT_Mini_v1.2 get this Runtime error:   ***ERROR*** A stack overflow in task OUT_iis has been detected.

migmelccs, 2024-03-14:
## Environment

- Audio development kit:      ESP32-LyraT-Mini
- Audio kit version: v1
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version: v4.4.7
- [Required] ADF version:  v2.6-84-g75c9aed9
- Build system:    idf.py
- [Required] Running log:     See the log below
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0
- Operating system:     Windows
- (Windows only) Environment type: ESP Command Prompt
- Using an IDE?: Yes, VS Code version 1.87.2 (usersetup)
- Power supply:         USB

## Problem Description

I have been trying to run the VoIP example and cannot success in make it to behave as it is suppose to. First I used ESP-IDF v5.2.1 combined with ESP-ADF master. There is a runtime problem because of mixing between deprecated i2s.h and the new headers i2s_std.h, etc., that depend on the framework and there is nothing I can do about. ESP-IDF framework uses the new headers, while ESP-ADF (stream.h or so) still uses the deprecated one.

But the report is about this: I have moved to ESP-IDF v4.4.7 combined with esp-adf-master, followed all the recommended in the README file, but a new problem arose (as shown in the title) even when I applied the recommended patch idf_v4.4_freertos.patch as suggested.
The device keeps rebooting after the stack overflow message and after several reboots, finally it hangs so  it is necessary to manually reset it to start over. 
By repeatedly pressing the keys randomly it sometimes stops rebooting and then is possible to make calls to a client such as Zoiper. However after some time it starts rebooting again. Also, calling from Zoiper to esp32 cause the device start rebooting again.

### Expected Behavior

After initialization it should be ready to make and receive calls 

### Actual Behavior

Application periodically resets  after "A stack overflow in task OUT_iis has been detected" message, and eventually hangs

### Steps to Reproduce

1. step1: setup SIP server previously (If SIP server(FreeSWTICH) is not set, neither stack overflow nor rebooting happens)
2. ... Upload the code and set monitor. It will immediately start rebooting

[image]

### Code to Reproduce This Issue

The code is the unmodified example loaded with ESP ADF at:  examples/protocols/voip
The only modification is relative to WiFi credentials and SIP ID/SIP server IP

## Debug Logs

I added 2 cycles because the first time looks different from the others
```
E:\esp-idf-v4.4.7\esp-adf-master\examples\protocols\voip>idf.py flash -p COM5 -b 921600 monitor
Executing action: flash
Running ninja in directory E:\esp-idf-v4.4.7\esp-adf-master\examples\protocols\voip\build
Executing "ninja flash"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32
NOTICE: Skipping optional dependency: espressif/jsmn
NOTICE: Skipping optional dependency: espressif/nghttp
NOTICE: Skipping optional dependency: espressif/usb_stream
Processing 3 dependencies:
[1/3] espressif/esp-dsp (1.4.12)
[2/3] espressif/esp32-camera (2.0.8)
[3/3] idf (4.4.7)
NOTICE: Skipping optional dependency: espressif/jsmn
NOTICE: Skipping optional dependency: espressif/nghttp
NOTICE: Skipping optional dependency: espressif/usb_stream
-- Project sdkconfig file E:/esp-idf-v4.4.7/esp-adf-master/examples/protocols/voip/sdkconfig
Loading defaults file E:/esp-idf-v4.4.7/esp-adf-master/examples/protocols/voip/sdkconfig.defaults...
E:/esp-idf-v4.4.7/esp-adf-master/examples/protocols/voip/sdkconfig.defaults:2 CONFIG_FLASHMODE_QIO was replaced with CONFIG_ESPTOOLPY_FLASHMODE_QIO
E:/esp-idf-v4.4.7/esp-adf-master/examples/protocols/voip/sdkconfig.defaults:27 CONFIG_SPIRAM_SUPPORT was replaced with CONFIG_ESP32_SPIRAM_SUPPORT
E:/esp-idf-v4.4.7/esp-adf-master/examples/protocols/voip/sdkconfig.defaults:52 CONFIG_UDP_RECVMBOX_SIZE was replaced with CONFIG_LWIP_UDP_RECVMBOX_SIZE
Loading defaults file E:/esp-idf-v4.4.7/esp-adf-master/examples/protocols/voip/sdkconfig.defaults.esp32...
[... 605 lines cut ...]
0x40090b76: _frxt_int_exit at E:/esp-idf-v4.4.7/components/freertos/port/xtensa/portasm.S:231

ELF file SHA256: 44c960785f1493af

Rebooting...

```
Eventually after several reboots it will stop here:

I (3378) ESP_AUDIO_TASK: Recv Element[IN_flash-0x3f806724] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (3379) CODEC_ELEMENT_HELPER: Th element is 0x3f8068bc. The reserve data 2 is 0x0.
I (3401) ESP_AUDIO_TASK: Recv Element[DEC_mp3-0x3f8068bc] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING
I (3410) ESP_AUDIO_TASK: Received muisc info then on play
I (3415) ESP_AUDIO_TASK: On event play, status:UNKNOWN, 0
I (3421) AUDIO_THREAD: The Audio_forge task allocate stack on external memory
I (3431) AUDIO_FORGE: audio_forge opened
I (3431) AUDIO_THREAD: The OUT_iis task allocate stack on external memory
I (3442) I2S_STREAM: AUDIO_STREAM_WRITER
I (3446) ESP_AUDIO_TASK: Recv Element[OUT_iis-0x3f806a70] MSG,type:20000,cmd:8,len:4,status:AEL_STATUS_STATE_RUNNING

## Other Items If Possible

[s
[Uploading voip_app.elf.txt…]()
dkconfig.txt](https://github.com/espressif/esp-adf/files/14599098/sdkconfig.txt)

chen-laodi, 2024-03-15:
俺也一样，rtc与voip相关文档严重不足，官方例程也是古老破旧，想要自己实现voip完全无从下手，可以说整个adf就好像没有一样

TempoTian, 2024-03-20:
I have checked this issue on IDFv4.4.5 and IDF5.2 both verison can not reproduce the stackoverflow issue.
    [voip.zip](https://github.com/espressif/esp-adf/files/14661562/voip.zip)
To debug overflow issue, you can manually enlarge stack size, for I2s stream, if you do not know where the config come from can manually change code:
        https://github.com/espressif/esp-adf/blob/master/components/audio_stream/i2s_stream.c#L410C8-L410C41

If it is caused by memory overwrote you can turn on the watch point `CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK` to check where the overwrote come from in tasks.c
```c
#ifdef ESP_PLATFORM
        xSwitchingContext[ xPortGetCoreID() ] = pdFALSE;
        #if CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK
        vPortSetStackWatchpoint(pxCurrentTCB[xPortGetCoreID()]->pxStack);
        #endif
#else
```

migmelccs, 2024-03-20:
> I have checked this issue on IDFv4.4.5 and IDF5.2 both verison can not reproduce the stackoverflow issue.

Thank you, it's kind of weird. I tried first using v5.2 and got a runtime incompatibility error between i2s.h(deprecated) and the new i2s headers. Apparently there are still some code using the old headers in the ADF framework.
Have you been able to establish a real call? I saw in the logs the device is calling itself.
Now, I have realized that the error someway comes from the embedded audio tones, because when the tones are not uploaded I just receive the errors about it not being able to read flash tone partition, but when uploading the tones i get the mentioned overflow errors and I get some sounds such as cracking or bubbles exploding. 
I don't know what i am doing so wrong, because this is the original partition file:

**# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 2M,
flash_tone,data, 0xff,  0x210000 , 500K,**

And this is the command to upload the tones to the flash:

**python E:/esp-idf-v4.4.7/esp-adf-master/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM12 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x210000 ../components/audio_flash_tone/bin/audio-esp.bin**

As I understand it, the command in accordance with the memory position of the partition

TempoTian, 2024-03-21:
I will try the latest ADF and the default IDF in it and share with you the test results

chen-laodi, 2024-03-21:
不要下载flash提示音就不会产生堆栈溢出，使用idf4.4.7可以正常运行

chen-laodi, 2024-03-21:
不要下载flash提示音就不会产生堆栈溢出，清空分区后重新编译下载，使用idf4.4.7可以正常运行

TempoTian, 2024-03-21:
We always use idf.py to flash binary, I think can try idf.py firstly to check whether it is all right or not.
可以先试下用 idf.py 来下载程序，看能否正常
idf.py -p *** flash monitor

TempoTian, 2024-03-21:
I have reproduced on latest version.
Bug is caused by change: https://github.com/espressif/esp-adf/commit/8cc2edc693c4faa4152ced42bf0ce34664723c7b#diff-e8182213449674e98ba2e7dc27c0ef07ce53499f3de4a59b238614825f6da771 which cause infinite recursive call.
You can rollback audio_player_init_tone.c or use follow code:
```c
/*
 * ESPRESSIF MIT License
 *
 * Copyright (c) 2021 <ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD>
 *
 * Permission is hereby granted for use on all ESPRESSIF SYSTEMS products, in which case,
 * it is free of charge, to any person obtaining a copy of this software and associated
 * documentation files (the "Software"), to deal in the Software without restriction, including
 * without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense,
 * and/or sell copies of the Software, and to permit persons to whom the Software is furnished
 * to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all copies or
 * substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 */
#include <string.h>

#include "esp_log.h"
#include "audio_error.h"
#include "esp_audio.h"
#include "tone_stream.h"
#include "mp3_decoder.h"
#include "i2s_stream.h"
#include "algorithm_stream.h"
#include "audio_player_int_tone.h"

static const char *TAG = "PLAYER_INT_TONE";

static esp_audio_handle_t player;
static stream_func origin_write_cb;

static int i2s_write_cb(audio_element_handle_t el, char *buf, int len, TickType_t wait_time, void *ctx)
{
    size_t bytes_write = 0, size = len;

#ifndef CONFIG_ESP32_S3_KORVO2L_V1_BOARD
    /* Drop right channel */
    int16_t *tmp = (int16_t *)buf;
    for (int i = 0; i < len / 4; i++) {
        tmp[i] = tmp[i << 1];
    }
    size = len / 2;
#endif

#if CONFIG_IDF_TARGET_ESP32
    algorithm_mono_fix((uint8_t *)buf, size);
#endif
    if (origin_write_cb) {
        bytes_write = origin_write_cb(el, buf, len, wait_time, ctx);
    }
    return bytes_write;
}

audio_err_t audio_player_int_tone_init(int sample_rate, int channel_format, int bits_per_sample)
{
    esp_audio_cfg_t cfg = DEFAULT_ESP_AUDIO_CONFIG();
    cfg.prefer_type = ESP_AUDIO_PREFER_MEM;
    cfg.resample_rate = sample_rate;
    player = esp_audio_create(&cfg);
    AUDIO_MEM_CHECK(TAG, player, return ESP_FAIL;);

    // Create readers and add to esp_audio
    tone_stream_cfg_t tn_reader = TONE_STREAM_CFG_DEFAULT();
    tn_reader.type = AUDIO_STREAM_READER;
    tn_reader.task_core = 1;
    esp_audio_input_stream_add(player, tone_stream_init(&tn_reader));

    // Add decoders and encoders to esp_audio
    mp3_decoder_cfg_t mp3_dec_cfg = DEFAULT_MP3_DECODER_CONFIG();
    mp3_dec_cfg.task_core = 1;
    mp3_dec_cfg.task_prio = 20;
    esp_audio_codec_lib_add(player, AUDIO_CODEC_TYPE_DECODER, mp3_decoder_init(&mp3_dec_cfg));

    i2s_stream_cfg_t i2s_writer_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_writer_cfg.type = AUDIO_STREAM_WRITER;
    i2s_writer_cfg.stack_in_ext = true;
    i2s_writer_cfg.i2s_config.sample_rate = sample_rate;
    i2s_writer_cfg.i2s_config.channel_format = channel_format;
    i2s_writer_cfg.i2s_config.bits_per_sample = bits_per_sample;
    i2s_writer_cfg.task_core = 1;
    audio_element_handle_t i2s_writer = i2s_stream_init(&i2s_writer_cfg);
    origin_write_cb = audio_element_get_write_cb(i2s_writer);
    esp_audio_output_stream_add(player, i2s_writer);
    audio_element_set_write_cb(i2s_writer, i2s_write_cb, NULL);
    audio_element_set_output_timeout(i2s_writer, portMAX_DELAY);

    return ESP_OK;
}

audio_err_t audio_player_int_tone_stop()
{
    esp_audio_stop(player, TERMINATION_TYPE_NOW);
    return ESP_OK;
}

audio_err_t audio_player_int_tone_play(const char *url)
{
    esp_audio_play(player, AUDIO_CODEC_TYPE_DECODER, url, 0);
    return ESP_OK;
}

```

migmelccs, 2024-03-21:
> 可以先试下用 idf.py 来下载程序，看能否正常

Yes, it works without uploading the tones( as also suggested by @chen-laodi). It does not work as it should, but I have managed to make it work fine without having to modify IDF files. I can make calls and even receive calls from a SIP client such as Zoiper, the only problem is that without the tones there is not ringing when incoming calls arrive.
I am currently customizing a ringing function, however it would have been better and cleaner if the example tones worked as expected.

migmelccs, 2024-03-21:
> You can rollback audio_player_init_tone.c or use follow code

Thank you, I will try it and let you know the results

migmelccs, 2024-03-21:
> Thank you, I will try it and let you know the results

Well, it worked! Thank you.
Although the alarm wouldn't be my first option for ringing but I guess I cam make my own set of tones. The most important is the failure was solved.
Thank you again

TempoTian, 2024-03-21:
看下你的PA的实际增益是放大多少，目前对不同的板子增益配置的不一样，要匹配下，对 layrtmini设置的是 20db
https://github.com/espressif/esp-adf/blob/master/components/audio_board/lyrat_mini_v1_1/board_def.h#L62
如果没有增益这么多可以把这个值调小，这样同样的音量映射到 codec 的音量就会变大 (避免由于PA导致的饱和现象)
之前加这个是为了解决不同平台同一音量下连接相同 codec 和喇叭，声音不一致的问题

TempoTian, 2024-03-21:
对于8311录制播放公用的，目前ADF没对外释放API调mic的音量可以手动到函数里面改，对于mini的板子可以用get到codec hal的handle然后通过hal的API来调节
```c
audio_hal_set_volume(audio_board_get_handle()->adc_hal, vol)
```

chen-laodi, 2024-03-21:
https://github.com/espressif/esp-adf/assets/40551848/0ac581ca-60d2-46e3-b86d-e4335e49dbec

对于mp3播放他听起来噪音很大，我很能确定我的网络无比的通常，这是音频地址https://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/Me.mp3

TempoTian, 2024-03-21:
建议你单独开一个issue来追踪这个问题，听起来像是decoder的问题

chen-laodi, 2024-03-22:
@TempoTian 老哥，audio_hal_set_volume我看只是设置了喇叭声音的大小，并没有直接改变麦克风拾音的大小呢，另外我提交了关于噪音的议题你那里有人方便帮看看呗

chen-laodi, 2024-03-22:
@TempoTian esp_err_t es7243_adc_set_voice_volume(int volume);我已经在内部修改成了100，但声音依旧不是很大，距离麦克风10厘米远，linPhone那里喇叭显示的最高也就30%那样，mini板子用的麦克风灵敏度信噪比多少跟麦克风有关系么

TempoTian, 2024-03-22:
i2s stream config有一个配置可以使能ALC 来用软件改变增益, 可以用软件来增大下音量
0 以上属于增加增益，对应也有API可以调节

```c
typedef struct {
    bool                  use_alc;         /*!< It is a flag for ALC. If use ALC, the value is true. Or the value is false */
    int                   volume;          /*!< The volume of audio input data will be set. */
must be a multiple of 3. The recommended value is 3600 */
} i2s_stream_cfg_t;
```

jason-mao, 2024-04-07:
@chen-laodi 由于你的问题和这个问题的标题不符，我将关闭这个问题。你的问题我会在https://github.com/espressif/esp-adf/issues/1174里面回复。

closed 2024-04-07 by jason-mao