espressif/esp-adf #1170: 目前补丁已经打了，还是提醒$ADF_PATH/idf_patches/idf_v4.4_freertos.patch怎么解决啊

liuliuu-112, 2024-03-16:
[image]
[image]
目前补丁已经打了，现在程序还是报请打补丁，这个怎么解决啊

TempoTian, 2024-03-18:
这个路径有文件只说明patch在，要打的话需要切到IDF的根目录然后执行`git apply ****.patch`, 打完后确认下freertos/tasks.c里面有对应的函数在就可以了，手动将函数copy过去也可以

liuliuu-112, 2024-03-19:
> 这个路径有文件只说明patch在，要打的话需要切到IDF的根目录然后执行`git apply ****.patch`, 打完后确认下freertos/tasks.c里面有对应的函数在就可以了，手动将函数copy过去也可以

是只复制这个函数，对吧，求教

BaseType_t xTaskCreateRestrictedPinnedToCore( const TaskParameters_t * const pxTaskDefinition,
                                                  TaskHandle_t *pxCreatedTask,
                                                  const BaseType_t xCoreID)
	{
	TCB_t *pxNewTCB;
	BaseType_t xReturn = errCOULD_NOT_ALLOCATE_REQUIRED_MEMORY;

		configASSERT( pxTaskDefinition->puxStackBuffer );

		if( pxTaskDefinition->puxStackBuffer != NULL )
		{
			/* Allocate space for the TCB.  Where the memory comes from depends
+			on the implementation of the port malloc function and whether or
+			not static allocation is being used. */
			pxNewTCB = ( TCB_t * ) pvPortMallocTcbMem( sizeof( TCB_t ) );

			if( pxNewTCB != NULL )
			{
				/* Store the stack location in the TCB. */
				pxNewTCB->pxStack = pxTaskDefinition->puxStackBuffer;

				/* Tasks can be created statically or dynamically, so note
+				this task had a statically allocated stack in case it is
+				later deleted.  The TCB was allocated dynamically. */
				pxNewTCB->ucStaticallyAllocated = tskDYNAMICALLY_ALLOCATED_STACK_AND_TCB;

				prvInitialiseNewTask(	pxTaskDefinition->pvTaskCode,
										pxTaskDefinition->pcName,
										pxTaskDefinition->usStackDepth,
										pxTaskDefinition->pvParameters,
										pxTaskDefinition->uxPriority,
										pxCreatedTask, pxNewTCB,
										pxTaskDefinition->xRegions,
										xCoreID );

				prvAddNewTaskToReadyList( pxNewTCB, pxTaskDefinition->pvTaskCode, xCoreID );
				xReturn = pdPASS;
			}
		}

		return xReturn;
	}

TempoTian, 2024-03-19:
是的

closed 2024-04-07 by jason-mao