espressif/esp-adf #1172: Useing adf master build project,wifi_service error :can not find STAILQ_ENTRY

itsweet, 2024-03-20:
os:windows vscode

idf v5.2.1 
adf git version:75c9aed9b8e2f5e2be9c46671fec9abd40e92689
the  error log like that:
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:61:5: error: expected specifier-qualifier-list before 'STAILQ_ENTRY'
   61 |     STAILQ_ENTRY(wifi_setting_item)     next;
      |     ^~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:67:9: error: function definition declared 'typedef'
   67 | typedef STAILQ_HEAD(wifi_setting_list, wifi_setting_item) wifi_setting_list_t;
      |         ^~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:67:9: error: return type defaults to 'int' [-Werror=implicit-int] 
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c: In function 'STAILQ_HEAD':
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:67:59: error: expected declaration specifiers before 'wifi_setting_list_t'
   67 | typedef STAILQ_HEAD(wifi_setting_list, wifi_setting_item) wifi_setting_list_t;
      |                                                           ^~~~~~~~~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:73:5: error: unknown type name 'wifi_setting_list_t'
   73 |     wifi_setting_list_t                 setting_list;
      |     ^~~~~~~~~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:87:3: error: storage class specified for parameter 'wifi_service_ 
'
   87 | } wifi_service_t;
      |   ^~~~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:92:3: error: storage class specified for parameter 'wifi_serv_event_type_t'
   92 | } wifi_serv_event_type_t;
      |   ^~~~~~~~~~~~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:95:5: error: expected specifier-qualifier-list before 'wifi_serv_event_type_t'
   95 |     wifi_serv_event_type_t  msg_type;
      |     ^~~~~~~~~~~~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:99:3: error: storage class specified for parameter 'wifi_task_msg_t'
   99 | } wifi_task_msg_t;
      |   ^~~~~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:102:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  102 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:116:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  116 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:132:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  132 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:238:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  238 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:245:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  245 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:267:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  267 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:277:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  277 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:284:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  284 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:290:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  290 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:502:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  502 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:508:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  508 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:514:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  514 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:529:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  529 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:537:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  537 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:548:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  548 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:556:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  556 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:564:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  564 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:572:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  572 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:580:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  580 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:587:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  587 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:594:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  594 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:612:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  612 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:626:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  626 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:632:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
  632 | {
      | ^
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:67:9: error: type of 'wifi_setting_list' defaults to 'int' [-Werror=implicit-int]
   67 | typedef STAILQ_HEAD(wifi_setting_list, wifi_setting_item) wifi_setting_list_t;
      |         ^~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:67:9: error: type of 'wifi_setting_item' defaults to 'int' [-Werror=implicit-int]
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:99:3: error: declaration for parameter 'wifi_task_msg_t' but no such parameter
   99 | } wifi_task_msg_t;
      |   ^~~~~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:92:3: error: declaration for parameter 'wifi_serv_event_type_t' but no such parameter
   92 | } wifi_serv_event_type_t;
      |   ^~~~~~~~~~~~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:87:3: error: declaration for parameter 'wifi_service_t' but no such parameter
   87 | } wifi_service_t;
      |   ^~~~~~~~~~~~~~
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:690: error: expected '{' at end of input
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:690: error: control reaches end of non-void function [-Werror=return-type]
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c: At top level:
D:/tom/esp/frameworks/esp-adf/components/wifi_service/src/wifi_service.c:48:18: error: 'WIFI_TASK_DESTROY_BIT' defined but not used [-Werror=unused-const-variable=]
   48 | const static int WIFI_TASK_DESTROY_BIT  = BIT0;

jason-mao, 2024-04-07:
@itsweet It's weird, please help to check your IDF_PATH  and ADF_PATH in the right place.

ChuckFoo, 2024-04-22:
I had this same problem (under Linux) when building against esp-idf master (on 20240421), got around it by switching to v5.2.1 (via `git checkout v5.2.1` prior to building) instead (as the matrix in the adf README hints at currently)