espressif/esp-adf #1173: pipeline_raw_http example doesn't work correctly

bardakt, 2024-03-20:
## Environment

- Audio development kit:      [ESP32-LyraT]
- Audio kit version (for ESP32-LyraT): [v4.3]
- [Required] Module or chip used:  [ESP32-WROVER-E]
- [Required] IDF version
    // v5.1
- [Required] ADF version
    // v2.6-84-g75c9aed9
- Build system:         [idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-12.2.0_20230208) 12.2.0
- Operating system:     [Windows]
- (Windows only) Environment type: [ESP Command Prompt]
- Using an IDE?: [No]
- Power supply:         [USB]

## Problem Description

// I just run this example(WIFI SSID, Password and IP is correct) and got a WAV file on server, but file is empty(silence). I've tried to run another examples where a mic is used (examples/recorder/pipeline_wav_amr_sdcard, examples/speech_recognition/vad) and it works fine, i get a files with sound. i've got in monitor some warngings that looks not normal behavior. And a line:
W (4836) AUDIO_THREAD: Make sure selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` by `make menuconfig`
makes me confused, because in menuconfig i can't do required. According to the specification, SPIRAM_BOOT_INIT should be in: Component config > ESP32-specific > SPI RAM config, but i don't have one. I found a PSRAM in menuconfig, but there is no ESP32-specific. I tried to change the ESP-IDF version, but it doesn't give a result. I also tried to use python 2.7 for the server instead of 3.12, but still has no effect. I use a default code example, nothing has been changed.
I found same problems in the internet, but they are for LyraT-Mini and there is no clear solution to the problem.
Can someone help to fix this example, or maybe i just can use another format of files, for example, AMR or MP3(but as far as I know mp3 is not supported). I've tried to make a code for transfering a AMR file to http server, but it doesn't work, thanks to my excellent coding knowledge :)

### Expected Behavior
WAV is recorded on the server, sound is present

### Actual Behavior
WAV is being recorded to the server, there is no sound

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7084
ho 0 tail 12 room 4
load:0x40078000,len:15592
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3876
entry 0x4008064c
I (33) boot: ESP-IDF v5.1-dirty 2nd stage bootloader
I (33) boot: compile time Mar 20 2024 22:39:02
I (33) boot: Multicore bootloader
I (38) boot: chip revision: v3.1
I (41) boot.esp32: SPI Speed      : 80MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00200000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2a88ch (174220) map
I (159) esp_image: segment 1: paddr=0003a8b4 vaddr=3ffb0000 size=03694h ( 13972) load
I (164) esp_image: segment 2: paddr=0003df50 vaddr=40080000 size=020c8h (  8392) load
I (168) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a28f8h (665848) map
I (374) esp_image: segment 4: paddr=000e2920 vaddr=400820c8 size=139b4h ( 80308) load
I (413) boot: Loaded app from partition at offset 0x10000
I (413) boot: Disabling RNG early entropy source...
I (425) cpu_start: Multicore app
I (425) cpu_start: Pro cpu up.
I (425) cpu_start: Starting app cpu, entry point is 0x40081410
0x40081410: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.1/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (443) cpu_start: Pro cpu start user code
I (443) cpu_start: cpu freq: 160000000 Hz
I (443) cpu_start: Application information:
I (448) cpu_start: Project name:     record_raw_http
I (454) cpu_start: App version:      v5.1-dirty
I (459) cpu_start: Compile time:     Mar 20 2024 22:38:47
I (465) cpu_start: ELF file SHA256:  e9e6a5eae546441c...
I (471) cpu_start: ESP-IDF:          v5.1-dirty
I (476) cpu_start: Min chip rev:     v0.0
I (481) cpu_start: Max chip rev:     v3.99
I (486) cpu_start: Chip rev:         v3.1
I (491) heap_init: Initializing. RAM available for dynamic allocation:
I (498) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (504) heap_init: At 3FFB8070 len 00027F90 (159 KiB): DRAM
I (510) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (516) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (523) heap_init: At 40095A7C len 0000A584 (41 KiB): IRAM
I (530) spi_flash: detected chip: generic
I (534) spi_flash: flash io: dio
W (537) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (551) app_start: Starting scheduler on CPU0
I (556) app_start: Starting scheduler on CPU1
I (556) main_task: Started on CPU0
I (566) main_task: Calling app_main()
I (586) REC_RAW_HTTP: [ 1 ] Initialize Button Peripheral & Connect to wifi network
W (1596) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (2596) REC_RAW_HTTP: [ 2 ] Start codec chip
E (2596) gpio: gpio_install_isr_service(498): GPIO isr service already installed
I (2616) REC_RAW_HTTP: [3.0] Create audio pipeline for recording
I (2616) REC_RAW_HTTP: [3.1] Create http stream to post data to server
I (2616) REC_RAW_HTTP: [3.2] Create i2s stream to read audio data from codec chip
I (2626) REC_RAW_HTTP: [3.3] Register all elements to audio pipeline
I (2636) REC_RAW_HTTP: [3.4] Link it together [codec_chip]-->i2s_stream->http_stream-->[http_server]
I (2646) REC_RAW_HTTP: [ 4 ] Press [Rec] button to record, Press [Mode] to exit
W (2666) PERIPH_TOUCH: _touch_init
E (4816) REC_RAW_HTTP: [ * ] [Rec] input key event, resuming pipeline ...
W (4816) AUDIO_PIPELINE: Without stop, st:1
W (4816) AUDIO_PIPELINE: Without wait stop, st:1
W (4816) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (4826) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (4836) AUDIO_THREAD: Make sure selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` by `make menuconfig`
Total bytes written: 53248
Total bytes written: 57600
I (6676) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker
I (6826) REC_RAW_HTTP: [ + ] HTTP client HTTP_STREAM_FINISH_REQUEST
W (6826) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128
I (6826) REC_RAW_HTTP: Got HTTP Response = File 20240320T194117Z_16000_16_1.wav was written, size 57600
```

## Other Items If Possible

In the next message

bardakt, 2024-03-20:
Unfortunately, i can't drop elf file here.

jason-mao, 2024-04-07:
@bardakt Could you upload your empty file at here? The `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY`,in theory, it doesn't affect your function.

bardakt, 2024-04-10:
Actually, i find a solution but not for HTTP example. HTTP example is still broken, but here is all good:

/* Example of Voice Activity Detection (VAD)

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "esp_log.h"
#include "board.h"
#include "audio_common.h"
#include "audio_pipeline.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "i2s_stream.h"
#include "raw_stream.h"
#include "http_stream.h"
#include "periph_wifi.h"
#include "esp_http_client.h"
#include "filter_resample.h"

#include "audio_idf_version.h"

static const char *TAG = "HTTP-Audio";

#define VAD_SAMPLE_RATE_HZ 16000
#define RECORDING_TIME_SECONDS 5

audio_pipeline_handle_t pipeline_http;
audio_element_handle_t i2s_stream_reader, http_stream_writer;

esp_err_t _http_stream_event_handle(http_stream_event_msg_t *msg)
{
    esp_http_client_handle_t http = (esp_http_client_handle_t)msg->http_client;
    char len_buf[16];
    static int total_write = 0;

    if (msg->event_id == HTTP_STREAM_PRE_REQUEST) {
        // set header
        ESP_LOGI(TAG, "[ + ] HTTP client HTTP_STREAM_PRE_REQUEST, lenght=%d", msg->buffer_len);
        esp_http_client_set_method(http, HTTP_METHOD_POST);
        char dat[10] = {0};
        snprintf(dat, sizeof(dat), "%d", VAD_SAMPLE_RATE_HZ);
        esp_http_client_set_header(http, "x-audio-sample-rates", dat);
        memset(dat, 0, sizeof(dat));
        snprintf(dat, sizeof(dat), "%d", 32);
        esp_http_client_set_header(http, "x-audio-bits", dat);
        memset(dat, 0, sizeof(dat));
        snprintf(dat, sizeof(dat), "%d", 1);
        esp_http_client_set_header(http, "x-audio-channel", dat);
        total_write = 0;
        return ESP_OK;
    }

    if (msg->event_id == HTTP_STREAM_ON_REQUEST) {
        // write data
        int wlen = sprintf(len_buf, "%x\r\n", msg->buffer_len);
        if (esp_http_client_write(http, len_buf, wlen) <= 0) {
            return ESP_FAIL;
        }
        if (esp_http_client_write(http, msg->buffer, msg->buffer_len) <= 0) {
            return ESP_FAIL;
        }
[... 104 lines cut ...]
    vTaskDelay(RECORDING_TIME_SECONDS * 1000 / portTICK_PERIOD_MS); // Ожидание 5 секунд
    

    ESP_LOGI(TAG, "[ 8 ] Stop audio_pipeline and release all resources");
    audio_pipeline_stop(pipeline_http);
    audio_pipeline_wait_for_stop(pipeline_http);
    audio_pipeline_terminate(pipeline_http);

    

    audio_pipeline_unregister(pipeline_http, http_stream_writer);
    audio_pipeline_unregister(pipeline_http, i2s_stream_reader);

    /* Terminate the pipeline before removing the listener */
    audio_pipeline_remove_listener(pipeline_http);

    /* Release all resources */
    audio_pipeline_deinit(pipeline_http);
    audio_element_deinit(http_stream_writer);
    audio_element_deinit(i2s_stream_reader);

}

Something in this code from VAD example, something from HTTP, something my own code, but on the IDF v5.1 it works correctly.
If you think that in pipeline_raw_http is all good and it was only mine problem, then u can close this issue because i don't need any help. And i'm sorry but i cant upload an empty file because i delete them all.

closed 2024-07-31 by jason-mao