espressif/esp-adf #1174: pipeline_http_mp3 播放声音时噪声很大且不连续 There is a lot of noise when playing sound

chen-laodi, 2024-03-22:
## Environment

- Audio development kit: ESP32-LyraT-Mini
- Audio kit version: v1
- Module or chip used: ESP32-WROVER-E
- IDF version: v4.4.7
- ADF version: v2.6-82-g33ba0c5c
- Build system: idf.py
- Running log:
- Operating system: Windows
- Environment type: ESP Command Prompt
- Using an IDE?: Yes(Visual Studio Code1.87.2)
- Power supply: USB

## Problem Description

使用pipeline_http_mp3例程播放在线音乐时噪音很大，而且声音不连贯
There is a lot of noise and incoherent sound when playing online music using the pipeline_http_mp3 routine

我的网络非常通畅
My network is very smooth

使用的是4Ω3w 36mm直径喇叭，喇叭采用钕磁铁，磁铁直径16mm
It uses a 4Ω3w 36mm diameter speaker. The speaker uses neodymium magnets and the magnet diameter is 16mm.

声音播放时断断续续，还有噪音
Sound playback is intermittent and noisy

MP3 URI：https://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/gnypmfys.mp3

### Expected Behavior

应该平顺的播放音频

### Actual Behavior

播放时断断续续有噪音

### Steps to Reproduce

直接运行例程没有经过改动就会产生此问题
在日志里我收到了几个警告

### Code to Reproduce This Issue

没有经过代码改动的pipeline_http_mp3例程
pipeline_http_mp3 routine, no changes

## Debug Logs

```
PS D:\vscodeProject\pipeline_http_mp3> set IDF_PATH=D:/esp/idf4.4.7/esp-idf-v4.4.7
PS D:\vscodeProject\pipeline_http_mp3> D:\esp\idf4.4.7\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\esp\idf4.4.7\esp-idf-v4.4.7\tools\idf_monitor.py -p COM21 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 d:\vscodeProject\pipeline_http_mp3\build\play_http_mp3.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM21 instead...
--- idf_monitor on \\.\COM21 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6656
load:0x40078000,len:15052
load:0x40080400,len:3836
0x40080400: _init at ??:?

entry 0x4008069c
I (27) boot: ESP-IDF v4.4.7-dirty 2nd stage bootloader
I (27) boot: compile time 11:12:10
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v3.1
I (35) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 8MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 phy_init         RF data          01 01 0000d000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00300000
I (87) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2b484h (177284) map
I (164) esp_image: segment 1: paddr=0003b4ac vaddr=3ffb0000 size=035d4h ( 13780) load
I (170) esp_image: segment 2: paddr=0003ea88 vaddr=40080000 size=01590h (  5520) load
I (172) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=c17f8h (792568) map
I (465) esp_image: segment 4: paddr=00101820 vaddr=40081590 size=15268h ( 86632) load
I (512) boot: Loaded app from partition at offset 0x10000
I (512) boot: Disabling RNG early entropy source...
I (523) cpu_start: Multicore app
I (524) psram: This chip is ESP32-D0WD
I (525) spiram: Found 64MBit SPI RAM device
I (526) spiram: SPI RAM mode: flash 40m sram 40m
I (531) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (538) cpu_start: Pro cpu up.
I (542) cpu_start: Starting app cpu, entry point is 0x400815c4
0x400815c4: call_start_cpu1 at D:/esp/idf4.4.7/esp-idf-v4.4.7/components/esp_system/port/cpu_start.c:151

I (0) cpu_start: App cpu up.
I (1435) spiram: SPI SRAM memory test OK
I (1443) cpu_start: Pro cpu start user code
I (1443) cpu_start: cpu freq: 160000000
I (1443) cpu_start: Application information:
I (1446) cpu_start: Project name:     play_http_mp3
I (1452) cpu_start: App version:      1
I (1456) cpu_start: Compile time:     Mar 21 2024 18:36:24
I (1463) cpu_start: ELF file SHA256:  b100f83b7786ea59...
I (1469) cpu_start: ESP-IDF:          v4.4.7-dirty
I (1474) cpu_start: Min chip rev:     v0.0
I (1479) cpu_start: Max chip rev:     v3.99
I (1484) cpu_start: Chip rev:         v3.1
I (1489) heap_init: Initializing. RAM available for dynamic allocation:
I (1496) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1502) heap_init: At 3FFB7A80 len 00028580 (161 KiB): DRAM
I (1508) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1515) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1521) heap_init: At 400967F8 len 00009808 (38 KiB): IRAM
I (1528) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1537) spi_flash: detected chip: gd
I (1540) spi_flash: flash io: dio
I (1545) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1555) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1605) HTTP_MP3_EXAMPLE: [ 1 ] Start audio codec chip
W (1635) I2C_BUS: I2C bus has been already created, [port:0]
I (1645) HTTP_MP3_EXAMPLE: [2.0] Create audio pipeline for playback
I (1645) HTTP_MP3_EXAMPLE: [2.1] Create http stream to read data
I (1645) HTTP_MP3_EXAMPLE: [2.2] Create i2s stream to write data to codec chip
I (1655) HTTP_MP3_EXAMPLE: [2.3] Create mp3 decoder to decode mp3 file
I (1665) HTTP_MP3_EXAMPLE: [2.4] Register all elements to audio pipeline
I (1665) HTTP_MP3_EXAMPLE: [2.5] Link it together http_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (1685) HTTP_MP3_EXAMPLE: [2.6] Set up  uri (http as http_stream, mp3 as mp3 decoder, and default output is i2s)
I (1695) HTTP_MP3_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
W (2145) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (3145) HTTP_MP3_EXAMPLE: [ 4 ] Set up  event listener
I (3145) HTTP_MP3_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (3145) HTTP_MP3_EXAMPLE: [4.2] Listening event from peripherals
I (3155) HTTP_MP3_EXAMPLE: [ 5 ] Start audio_pipeline
I (7325) HTTP_MP3_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
W (303365) HTTP_STREAM: No more data,errno:0, total_bytes:10414628, rlen = 0
W (303985) HTTP_MP3_EXAMPLE: [ * ] Stop event received
I (303985) HTTP_MP3_EXAMPLE: [ 6 ] Stop audio_pipeline
W (303985) AUDIO_ELEMENT: [http] Element already stopped
W (303985) AUDIO_ELEMENT: [mp3] Element already stopped
W (303995) AUDIO_ELEMENT: [i2s] Element already stopped
W (304005) AUDIO_PIPELINE: There are no listener registered
W (304005) AUDIO_ELEMENT: [http] Element has not create when AUDIO_ELEMENT_TERMINATE
W (304015) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (304025) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (304045) PERIPH_WIFI: Wi-Fi disconnected from SSID hlkj, auto-reconnect disabled, reconnect after 1000 ms
W (304045) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:3
```

## Other Items If Possible
[sdkconfig.zip](https://github.com/espressif/esp-adf/files/14716765/sdkconfig.zip)
[play_http_mp3.elf.zip](https://github.com/espressif/esp-adf/files/14716911/play_http_mp3.elf.zip)
[recording.zip](https://github.com/espressif/esp-adf/files/14717024/recording.zip)

jason-mao, 2024-04-07:
@chen-laodi https://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/gnypmfys.mp3, 这个音乐失效了，请帮再提供一个。

chen-laodi, 2024-04-07:
@jason-mao 这是重新上传的三个音乐，http和https访问都可以
https://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/mp3/Relax.mp3
https://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/mp3/gnypmfys.mp3
https://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/mp3/Me.mp3

jason-mao, 2024-04-07:
你上面三首音乐是你录制，还是原始的音乐？音乐的幅度都太满了，容易消顶产生噪音。从你前面提供的recording.zip 文件来看，应该是丢数据了，有卡顿的现象。你先把HTTP output ringbuffer 增大看是否有好转，如下图，
[image]

jason-mao, 2024-07-31:
我将关闭这个不活跃的问题，如果有进一步的更新请随时重新打开或新建 Issue。

closed 2024-07-31 by jason-mao