espressif/esp-adf #1175: pipeline_flash_tone 例程会发出鸟叫？只播放了提示音的大概前0.1s

chen-laodi, 2024-03-23:
## Environment

- Audio development kit: ESP32-LyraT-Mini
- Audio kit version: v1
- Module or chip used: ESP32-WROVER-E
- IDF version: v4.4.7
- ADF version: v2.6-82-g33ba0c5c
- Build system: idf.py
- Running log:
- Operating system: Windows
- Environment type: ESP Command Prompt
- Using an IDE?: Yes(Visual Studio Code1.87.2)
- Power supply: USB

## Problem Description

例程是从adf中复制出来的
原本的例程没有经过任何修改，只是编译运行这样
我也尝试过自己定义提示音烧录，但是效果一样，每次运行都只播放了提示音的大概前0.1s
喇叭没有坏掉，耳朵也没有坏掉
我把录音放到下面了

## Debug Logs

```
PS D:\VisualStudionProjects\pipeline_flash_tone> set IDF_PATH=D:/sofe/Espressif-4.4.7/esp-idf-v4.4.7
PS D:\VisualStudionProjects\pipeline_flash_tone> D:\sofe\Espressif-4.4.7\python_env\idf4.4_py3.8_env\Scripts\python.exe D:\sofe\Espressif-4.4.7\esp-idf-v4.4.7\tools\idf_monitor.py -p COM14 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 d:\VisualStudionProjects\pipeline_flash_tone\build\play_tone_mp3.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM14 instead...
--- idf_monitor on \\.\COM14 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6656
load:0x40078000,len:15052
load:0x40080400,len:3836
0x40080400: _init at ??:?

entry 0x4008069c
I (27) boot: ESP-IDF v4.4.7-dirty 2nd stage bootloader
I (27) boot: compile time 17:48:46
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v3.1
I (35) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 8MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot:  3 flash_tone       Unknown data     01 ff 00110000 0007d000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=17b28h ( 97064) map
I (143) esp_image: segment 1: paddr=00027b50 vaddr=3ffb0000 size=01d2ch (  7468) load
I (146) esp_image: segment 2: paddr=00029884 vaddr=40080000 size=06794h ( 26516) load
I (159) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=2ff58h (196440) map
I (231) esp_image: segment 4: paddr=0005ff80 vaddr=40086794 size=0c038h ( 49208) load
I (260) boot: Loaded app from partition at offset 0x10000
I (260) boot: Disabling RNG early entropy source...
I (271) cpu_start: Multicore app
I (271) psram: This chip is ESP32-D0WD
I (273) spiram: Found 64MBit SPI RAM device
I (274) spiram: SPI RAM mode: flash 40m sram 40m
I (279) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (286) cpu_start: Pro cpu up.
I (290) cpu_start: Starting app cpu, entry point is 0x400814fc
0x400814fc: call_start_cpu1 at D:/sofe/Espressif-4.4.7/esp-idf-v4.4.7/components/esp_system/port/cpu_start.c:151

I (0) cpu_start: App cpu up.
I (1183) spiram: SPI SRAM memory test OK
I (1191) cpu_start: Pro cpu start user code
I (1192) cpu_start: cpu freq: 160000000
I (1192) cpu_start: Application information:
I (1195) cpu_start: Project name:     play_tone_mp3
I (1200) cpu_start: App version:      1
I (1205) cpu_start: Compile time:     Mar 23 2024 17:48:25
I (1211) cpu_start: ELF file SHA256:  b5e551e676726e03...
I (1217) cpu_start: ESP-IDF:          v4.4.7-dirty
I (1222) cpu_start: Min chip rev:     v0.0
I (1227) cpu_start: Max chip rev:     v3.99
I (1232) cpu_start: Chip rev:         v3.1
I (1237) heap_init: Initializing. RAM available for dynamic allocation:
I (1244) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1250) heap_init: At 3FFB27D0 len 0002D830 (182 KiB): DRAM
I (1257) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1263) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1270) heap_init: At 400927CC len 0000D834 (54 KiB): IRAM
I (1276) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1285) spi_flash: detected chip: gd
I (1288) spi_flash: flash io: dio
I (1293) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1303) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1313) TONE_MP3_EXAMPLE: [ 1 ] Start codec chip
W (1343) I2C_BUS: I2C bus has been already created, [port:0]
I (1353) TONE_MP3_EXAMPLE: [2.0] Create audio pipeline for playback
I (1353) TONE_MP3_EXAMPLE: [2.1] Create tone stream to read data from flash
I (1363) TONE_MP3_EXAMPLE: [2.2] Create i2s stream to write data to codec chip
I (1373) TONE_MP3_EXAMPLE: [2.3] Create mp3 decoder to decode mp3 file
I (1373) TONE_MP3_EXAMPLE: [2.4] Register all elements to audio pipeline
I (1383) TONE_MP3_EXAMPLE: [2.5] Link it together [flash]-->tone_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (1393) TONE_MP3_EXAMPLE: [2.6] Set up  uri (file as tone_stream, mp3 as mp3 decoder, and default output is i2s)
I (1403) TONE_MP3_EXAMPLE: [ 3 ] Set up event listener
I (1413) TONE_MP3_EXAMPLE: [3.1] Listening event from all elements of pipeline
I (1423) TONE_MP3_EXAMPLE: [ 4 ] Start audio_pipeline
I (1493) TONE_MP3_EXAMPLE: [ 4 ] Listen for all pipeline events
I (1493) TONE_MP3_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=16000, bits=16, ch=1
W (1513) TONE_STREAM: No more data,ret:0 ,info.byte_pos:5400
W (3133) TONE_MP3_EXAMPLE: [ * ] Stop event received
I (3133) TONE_MP3_EXAMPLE: [ 5 ] Stop audio_pipeline
W (3133) AUDIO_ELEMENT: [tone] Element already stopped
W (3143) AUDIO_ELEMENT: [mp3] Element already stopped
W (3143) AUDIO_ELEMENT: [i2s] Element already stopped
W (3153) AUDIO_PIPELINE: There are no listener registered
W (3153) AUDIO_ELEMENT: [tone] Element has not create when AUDIO_ELEMENT_TERMINATE
W (3163) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (3173) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
```

## 提示音烧录日志

```
PS D:\VisualStudionProjects\pipeline_flash_tone> python D:/sofe/Espressif/frameworks/esp-idf-v5.1.2/components/esptool_py/esptool/esptool.py --chip esp32 --port COM14 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x110000 D:/VisualStudionProjects/pipeline_flash_tone/tools/audio-esp.bin     
esptool.py v4.7.0
Serial port COM14
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:d1:f9:ec:57:84
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00110000 to 0x0017afff...
Compressed 434756 bytes to 399938...
Wrote 434756 bytes (399938 compressed) at 0x00110000 in 35.9 seconds (effective 96.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

https://github.com/espressif/esp-adf/assets/40551848/7a5afd1b-24fb-4623-b332-2ab3efcb857e

[bin+elf.zip](https://github.com/espressif/esp-adf/files/14731151/bin%2Belf.zip)

chen-laodi, 2024-03-27:
@TempoTian 哪个好心人能帮咱看看这个问题

TempoTian, 2024-03-28:
我本地用Mini板子跑了下是没有问题的，我看你的log也没有什么问题，对比正常的log
可以看下喇叭有没有插错口，靠近耳机的才是喇叭口，也可以插入耳机听听看
```
I (1486) TONE_MP3_EXAMPLE: [ 4 ] Listen for all pipeline events
I (1486) TONE_MP3_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=16000, bits=16, ch=1
W (1596) TONE_STREAM: No more data,ret:0 ,info.byte_pos:5400
W (3296) TONE_MP3_EXAMPLE: [ * ] Stop event received
I (3296) TONE_MP3_EXAMPLE: [ 5 ] Stop audio_pipeline
W (3296) AUDIO_ELEMENT: [tone] Element already stopped
W (3306) AUDIO_ELEMENT: [mp3] Element already stopped
W (3306) AUDIO_ELEMENT: [i2s] Element already stopped
W (3316) AUDIO_PIPELINE: There are no listener registered
W (3316) AUDIO_ELEMENT: [tone] Element has not create when AUDIO_ELEMENT_TERMINATE
W (3326) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (3336) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
```

chen-laodi, 2024-03-28:
@TempoTian 老哥真的不行，喇叭绝对没插错，耳机也试过了都是响一声就没了，我把工程文件打包了要不你看看，这块板子voip和http MP3都可以，唯独这个不行，我也换过电脑编译下载也一样，idf4.4.7 adf-master
https://static-web-file-001.obs.cn-north-4.myhuaweicloud.com/bin/pipeline_flash_tone.zip

[image]

TempoTian, 2024-03-29:
这个是I2S driver的Regression，可以尝试rollback一下下面的代码：
esp-adf/components/hal/i2s_hal.c
```c
void i2s_hal_tx_set_channel_style(i2s_hal_context_t *hal, const i2s_hal_config_t *hal_cfg)
{
    uint32_t chan_num = 2;
    uint32_t chan_bits = hal_cfg->chan_bits;
    uint32_t data_bits = hal_cfg->sample_bits;
#if SOC_I2S_SUPPORTS_TDM
    bool is_mono = (hal_cfg->chan_fmt == I2S_CHANNEL_FMT_ALL_RIGHT) ||
                   (hal_cfg->chan_fmt == I2S_CHANNEL_FMT_ALL_LEFT);
#else
    bool is_mono = (hal_cfg->chan_fmt == I2S_CHANNEL_FMT_ONLY_RIGHT) ||
                   (hal_cfg->chan_fmt == I2S_CHANNEL_FMT_ONLY_LEFT);
#endif

    /* Set channel number and valid data bits */
#if SOC_I2S_SUPPORTS_TDM
    chan_num = hal_cfg->total_chan;
    i2s_ll_tx_set_active_chan_mask(hal->dev, hal_cfg->chan_mask >> 16);
    i2s_ll_tx_set_chan_num(hal->dev, chan_num);
#else
    // 主要是这一行
     i2s_ll_tx_set_chan_mod(hal->dev, hal_cfg->chan_fmt < I2S_CHANNEL_FMT_ONLY_RIGHT ? hal_cfg->chan_fmt : (hal_cfg->chan_fmt >> 1)); // 0-two channel;1-right;2-left;3-righ;4-left
#endif  // SOC_I2S_SUPPORTS_TDM
#if SOC_I2S_SUPPORTS_PDM_CODEC
    if (hal_cfg->mode & I2S_MODE_PDM) {
        // Fixed to 16 while using mono mode and 32 while using stereo mode
        data_bits = hal_cfg->chan_fmt == I2S_CHANNEL_FMT_RIGHT_LEFT ? 32 : 16;
        chan_bits = data_bits;
    }
#endif
    i2s_ll_tx_set_sample_bit(hal->dev, chan_bits, data_bits);
    i2s_ll_tx_enable_mono_mode(hal->dev, is_mono);

    /* Set communication format */
    bool shift_en = hal_cfg->comm_fmt == I2S_COMM_FORMAT_STAND_I2S ? true : false;
    uint32_t ws_width = i2s_hal_get_ws_bit(hal_cfg->comm_fmt, chan_num, chan_bits);
    i2s_ll_tx_enable_msb_shift(hal->dev, shift_en);
    i2s_ll_tx_set_ws_width(hal->dev, ws_width);
#if SOC_I2S_SUPPORTS_TDM
    uint32_t half_sample_bits = chan_num * chan_bits / 2;
#if SOC_I2S_SUPPORTS_PDM_CODEC
    if (hal_cfg->mode & I2S_MODE_PDM) {
        half_sample_bits = 16; // Fixed to 16 in PDM mode
    }
#endif
    i2s_ll_tx_set_half_sample_bit(hal->dev, half_sample_bits);
#endif
}
```

chen-laodi, 2024-03-29:
@TempoTian 能用了，妙手回春啊大夫

closed 2024-04-07 by jason-mao