espressif/esp-adf #1177: PWM Stream produces error code 81002.

mike-2020, 2024-03-24:
ESP-IDF 5.1.2
ESP-ADF 2.6
ESP2-S3

Here is the code to init pwm stream:

`    pwm_stream_cfg_t pwm_cfg = PWM_STREAM_CFG_DEFAULT();
    pwm_cfg.pwm_config.tg_num = TIMER_GROUP_1;
    pwm_cfg.pwm_config.timer_num = TIMER_1;
    pwm_cfg.pwm_config.gpio_num_left = GPIO_NUM_47;
    pwm_cfg.pwm_config.gpio_num_right = GPIO_NUM_48;
    pwm_stream_writer = pwm_stream_init(&pwm_cfg);
    if(pwm_stream_writer==NULL) {
        ESP_LOGE(TAG, "Failed to init pwm stream.");
    }else{
        esp_audio_output_stream_add(handle, pwm_stream_writer);
    }`

but it reports errors below: (means no output stream). my app use both timer and ledc pwm, but I have tried to avoid conflict. please help me debug this issue.

[image]

TempoTian, 2024-03-27:
add `#pwm` to your play uri and have a try `file://mp3//power_on.mp3#pwm`

mike-2020, 2024-03-29:
Tried, but got similar error:

**I (542) SOUND_PLAYER: http_audio_play: play file://mp3//power_on.mp3#pwm.**
I (552) ESP_AUDIO_CTRL: Enter play procedure, src:0
I (552) ESP_AUDIO_CTRL: Play procedure, URL is ok, src:0
I (562) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3fca8448
I (562) ESP_AUDIO_TASK: It's a decoder
I (572) ESP_AUDIO_TASK: 1.CUR IN:[IN_file],CODEC:[DEC_mp3],RESAMPLE:[48000],OUT:[OUT_iis],rate:0,ch:0,pos:0
W (582) ESP_AUDIO_TASK: No more out stream handle
I (582) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:830, MEM Total:306116 Bytes

I (592) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:776, MEM Total:306116 Bytes

**I (602) ESP_AUDIO_CTRL: Exit play procedure, ret:81002
W (612) SOUND_PLAYER: esp_audio_play return 528386.**

TempoTian, 2024-03-29:
I see somecode in library only allow "raw" and "file" as output in hardcode.
To workaround to fix this issue you can force set stream name as iis. We will fix the lib issue soon.
```c
    esp_audio_output_stream_add(player, pwm_stream);
    audio_element_set_tag(pwm_stream, "OUT_iis");
```

mike-2020, 2024-03-29:
> I see somecode in library only allow "raw" and "file" as output in hardcode. To workaround to fix this issue you can force set stream name as iis. We will fix the lib issue soon.
> 
> ```c
>     esp_audio_output_stream_add(player, pwm_stream);
>     audio_element_set_tag(pwm_stream, "OUT_iis");
> ```

Thanks. This makes the error message disapeared, but it does not output any sound at all. And it seems the pipeline for the first mp3 will never stop. So, all following mp3 play will generate error message like below:

W (54602) ESP_AUDIO_CTRL: Pipeline is already running. Please stop it first.