espressif/esp-adf #1182: `i2s_driver_cleanup` in `i2s_stream_idf5.c->i2s_stream_init()` is not safe

yyjdelete, 2024-03-28:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
- [x] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

for [examples/speech_recognition/wwe](https://github.com/espressif/esp-adf/blob/master/examples/speech_recognition/wwe/main/main.c#L453-L454), if you switch the order of `setup_player()` and `start_recorder()`, 
`i2s_stream_init` will delete the old handle and create an new one, but not mutex for this, and it doesn't check `config->uninstall_drv`, so another i2s_stream maybe still using it.

https://github.com/espressif/esp-adf/blob/74c88720a0ecdf39a2acdffc55f3705078be1682/components/audio_stream/i2s_stream_idf5.c#L641
https://github.com/espressif/esp-adf/blob/74c88720a0ecdf39a2acdffc55f3705078be1682/components/audio_stream/i2s_stream_idf5.c#L165-L166
https://github.com/espressif/esp-adf/blob/74c88720a0ecdf39a2acdffc55f3705078be1682/components/audio_stream/i2s_stream_idf5.c#L471-L478

It may lead to random crash(i2s reader block forever on deleted mutex or random broken memory).
And require an safe way to do it as it does in the old `i2s_stream.c`

jason-mao, 2024-04-07:
@yyjdelete Thanks for your report, let's fix it.

shootao, 2024-07-12:
Hi @yyjdelete 
Please update [i2s_stream_idf5.zip](https://github.com/user-attachments/files/16191213/i2s_stream_idf5.zip)  and have a try