espressif/esp-adf #1183: ESP-IDF upgrade from v5.1.2 to v5.2.1 crashes on esp_periph_start

tank104, 2024-04-04:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, start similar to "How do I..". If it is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
- [ ] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps above, please follow these instructions so we can get the needed information to help you quickly and effectively.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your issues.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other Items If Possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the `Preview` tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: Please follow the above instructions and provide as many details as possible. It will save a lot of communication time and improve the efficiency of problem solving. The more details you provide, the faster we may be able to reproduce and resolve the issue. Thanks!**

----------------------------- Delete above -----------------------------

## Environment

- Audio development kit:      [ESP32-Dev]
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version: v5.2.1
- [Required] ADF version: v2.6-86-g1a03fe6acd esp-id
- Build system:        CMake
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version : that command doesn't work
- Operating system:   Windows
- (Windows only) Environment type: ESP Command Prompt
- Using an IDE?: PlatformIO & VSCode
- Power supply:     USB

## Problem Description

I have just upgraded IDF from 5.1.2 to 5.2.1 (PlatformIO 6.5 to 6.6), and the following code that works fine in 5.1.2 crashes in 5.2.1.
This is just example code that is already in your examples.

```
  ESP_LOGI(TAG, "[ 1 ] Mount spiffs");
  // Initialize Spiffs peripheral
  periph_spiffs_cfg_t spiffs_cfg = {
      .root = "/spiffs",
      .partition_label = NULL,
      .max_files = 5,
      .format_if_mount_failed = true};
  esp_periph_handle_t spiffs_handle = periph_spiffs_init(&spiffs_cfg);

  // Start spiffs
  esp_periph_start(set, spiffs_handle);
```

### Expected Behavior

It should run and load Spiffs

### Actual Behavior

It errors with this:
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x400ffd04  PS      : 0x00060e34  A0      : 0x80081f06  A1      : 0x3ffb7a10
A2      : 0x3ffb00e0  A3      : 0xffffffff  A4      : 0x0000cdcd  A5      : 0x00060e23  
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x8008a8d0  A9      : 0x3ffb7a10
A10     : 0x3ffb00e0  A11     : 0xb33fffff  A12     : 0x0000cdcd  A13     : 0x3ffb7930  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000010  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

### Steps to Reproduce

Just use this example:
https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_spiffs_mp3/main/play_spiffs_mp3_example.c

### Code to Reproduce This Issue
https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_spiffs_mp3/main/play_spiffs_mp3_example.c

## Debug Logs

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15736
load:0x40080400,len:4
load:0x40080404,len:3944
entry 0x40080654
I (28) boot: ESP-IDF 5.2.1 2nd stage bootloader
I (29) boot: compile time Apr  4 2024 11:55:36
;32mI (29) boot: Multicore bootloader
I (33) boot: chip revision: v3.0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (74) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (82) boot:  2 app0             OTA app          00 10 00010000 000e0000
I (89) boot:  3 storage          Unknown data     01 82 000f0000 00300000
I (97) boot: End of partition table
mI (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=131f0h ( 78320) map
[0;32mI (138) esp_image: segment 1: paddr=00023218 vaddr=3ffb0000 size=02560h (  9568) load
I (142) esp_image: segment 2: paddr=00025780 vaddr=40080000 size=0a898h ( 43160) load0m
[0;32mI (161) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=31100h (200960) map
[0;32mI (234) esp_image: segment 4: paddr=00061128 vaddr=4008a898 size=034d4h ( 13524) load
[0;32mI (247) boot: Loaded app from partition at offset 0x10000[0m
I (247) boot: Disabling RNG early entropy source...[0m
I (259) cpu_start: Multicore app
[0;32mI (268) cpu_start: Pro cpu start user code
I (268) cpu_start: cpu freq: 160000000 Hz
I (268) cpu_start: Application information:
I (271) cpu_start: Project name:     EspIdfAndAdf
I (276) cpu_start: App version:      7bdc175-dirty
I (282) cpu_start: Compile time:     Apr  4 2024 12:56:53
I (288) cpu_start: ELF file SHA256:  6e53dc855ff33b12...
I (294) cpu_start: ESP-IDF:          5.2.1
I (298) cpu_start: Min chip rev:     v0.0
I (303) cpu_start: Max chip rev:     v3.99
I (308) cpu_start: Chip rev:         v3.0
I (313) heap_init: Initializing. RAM available for dynamic allocation:
I (320) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (326) heap_init: At 3FFB3080 len 0002CF80 (179 KiB): DRAM
I (332) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (339) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
32mI (345) heap_init: At 4008DD6C len 00012294 (72 KiB): IRAMm
I (353) spi_flash: detected chip: generic
I (356) spi_flash: flash io: dio
W (360) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
;32mI (374) main_task: Started on CPU0
I (384) main_task: Calling app_main()
I (384) SPIFFS_MP3_EXAMPLE: [ 1 ] Mount spiffs
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x400ffd04  PS      : 0x00060e34  A0      : 0x80081f06  A1      : 0x3ffb7a10
A2      : 0x3ffb00e0  A3      : 0xffffffff  A4      : 0x0000cdcd  A5      : 0x00060e23  
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x8008a8d0  A9      : 0x3ffb7a10
A10     : 0x3ffb00e0  A11     : 0xb33fffff  A12     : 0x0000cdcd  A13     : 0x3ffb7930  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000010  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x400ffd01:0x3ffb7a10 0x40081f03:0x3ffb7a40 0x4008528f:0x3ffb7a60 0x4008684a:0x3ffb7a80 0x40084fc3:0x3ffb7aa0 0x400851d9:0x3ffb7ac0 0x400fe7f9:0x3ffb7af0 0x400e0173:0x3ffb7b50 0x400fea21:0x3ffb7b90 0x400febbd:0x3ffb7c40 0x400fecb3:0x3ffb7c60 0x400fed13:0x3ffb7c80 0x400f7d7e:0x3ffb7ca0 0x400f8798:0x3ffb7ce0 0x400db002:0x3ffb7db0 0x400db0c5:0x3ffb7e00 0x400dadc1:0x3ffb7e20 0x4008a69e:0x3ffb7e40

ELF file SHA256: 6e53dc855ff33b12

Rebooting...

randolfo75, 2024-04-05:
Same problem here

tank104, 2024-04-05:
Glad to know its not just me - are you using PlatformIO too?  Assume so but worth checking

randolfo75, 2024-04-06:
I am not using PlatformIO, I am using esp idf plugin on vscode. In my case I solved the problem, but I have mistaken the origin of the problem. After further analysis my problem was the behavior change in sdkconfig flag CONFIG_FREERTOS_UNICORE, it was configured to just one core, but my code was setting some tasks to another core. In the new version 5.2.1 it triggers assert errors. One way to solve it was just disable the CONFIG_FREERTOS_UNICORE flag. Maybe you can read the log change and find the solution for you too: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.2/peripherals.html .

tank104, 2024-04-07:
@randolfo75 that worked - I set CONFIG_FREERTOS_UNICORE=y and it sorts the problem.

It looks like this means FreeRTOS will run on first core only, not sure of the implications of this, and why it changed for 5.2.1 .  Is it desired behaviour?

The link you sent didn't help me - as it looked related to i2c, and I am using i2s

randolfo75, 2024-04-12:
Happy I could help in some way, at least you have a running code. Quite good questions you say, I am looking for the answers too. Now we have more clues though.

> @randolfo75 that worked - I set CONFIG_FREERTOS_UNICORE=y and it sorts the problem.
>
> It looks like this means FreeRTOS will run on first core only, not sure of the implications of this, and why it [...]