espressif/esp-adf #1184: WWE正常烧录，运行后不能正常使用，设备是ESP32S3 BOX Lite

thechen130, 2024-04-04:
PS D:\MyProject\ESP\espIDF\tools\esp-adf\examples\speech_recognition\wwe> set IDF_PATH=d:\MyProject\ESP\espIDF\esp-idf\v5.2.1\esp-idf

PS D:\MyProject\ESP\espIDF\tools\esp-adf\examples\speech_recognition\wwe> d:\MyProject\ESP\espIDF\tools\python_env\idf5.2_py3.11_env\Scripts\python.exe d:\MyProject\ESP\espIDF\esp-idf\v5.2.1\esp-idf\tools\idf_monitor.py -p COM3 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 d:\MyProject\ESP\espIDF\tools\esp-adf\examples\speech_recognition\wwe\build\example_wwe.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (100) boot:  4 model            Unknown data     01 82 ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4207bdd5
0x4207bdd5: memcpy_int16_728 at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/asm_functions/memcpy_int16_728.S:22

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1918
load:0x403c9700,len:0x4
load:0x403c9704,len:0xe5c
load:0x403cc700,len:0x302c
entry 0x403c993c
I (26) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader
I (27) boot: compile time Apr  4 2024 17:57:56
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00010000 00290000
I (100) boot:  4 model            Unknown data     01 82 002a0000 00500000
I (107) boot:  5 flash_tone       Unknown data     01 27 007a0000 00032000
I (115) boot: End of partition table
I (119) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=32368h (205672) map
I (159) esp_image: segment 1: paddr=00042390 vaddr=3fc9a300 size=056f0h ( 22256) load
I (163) esp_image: segment 2: paddr=00047a88 vaddr=40378000 size=08590h ( 34192) load
I (171) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=8795ch (555356) map
I (257) esp_image: segment 4: paddr=000d7984 vaddr=40380590 size=09cd0h ( 40144) load
I (273) boot: Loaded app from partition at offset 0x10000
I (274) boot: Disabling RNG early entropy source...
I (285) cpu_start: Multicore app
I (285) octal_psram: vendor id    : 0x0d (AP)
I (285) octal_psram: dev id       : 0x02 (generation 3)
I (288) octal_psram: density      : 0x03 (64 Mbit)
I (294) octal_psram: good-die     : 0x01 (Pass)
I (299) octal_psram: Latency      : 0x01 (Fixed)
I (304) octal_psram: VCC          : 0x01 (3V)
I (309) octal_psram: SRF          : 0x01 (Fast Refresh)
I (315) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (321) octal_psram: BurstLen     : 0x01 (32 Byte)
I (327) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (333) octal_psram: DriveStrength: 0x00 (1/1)
I (339) MSPI Timing: PSRAM timing tuning index: 4
I (343) esp_psram: Found 8MB PSRAM device
I (348) esp_psram: Speed: 80MHz
I (634) esp_psram: SPI SRAM memory test OK
I (643) cpu_start: Pro cpu start user code
I (643) cpu_start: cpu freq: 240000000 Hz
I (643) cpu_start: Application information:
I (646) cpu_start: Project name:     example_wwe
I (651) cpu_start: App version:      v2.6-84-g75c9aed9-dirty
I (658) cpu_start: Compile time:     Apr  4 2024 17:57:12
I (664) cpu_start: ELF file SHA256:  f8bb92005...
I (669) cpu_start: ESP-IDF:          v5.2.1-dirty
I (674) cpu_start: Min chip rev:     v0.0
I (679) cpu_start: Max chip rev:     v0.99
I (684) cpu_start: Chip rev:         v0.1
I (689) heap_init: Initializing. RAM available for dynamic allocation:
I (696) heap_init: At 3FCA0B98 len 00048B78 (290 KiB): RAM
I (702) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (708) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (714) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (722) spi_flash: detected chip: gd
I (726) spi_flash: flash io: qio
W (730) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (741) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (749) sleep: Configure to isolate all GPIO pins in sleep state
I (756) sleep: Enable automatic switching of GPIO sleep configuration
I (763) main_task: Started on CPU0
I (767) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (776) main_task: Calling app_main()
I (787) gpio: GPIO[46]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (803) AUDIO_BOARD: The board has already been initialized!

----------------------------- ESP Audio Platform -----------------------------
|                                                                            |
|                 ESP_AUDIO-v1.7.3-22-ga3cf4e4-9f1d1e8-e2f9463               |
|                     Compile date: Feb  5 2024-12:25:36                     |
------------------------------------------------------------------------------
I (838) wwe_example: Func:setup_player, Line:143, MEM Total:8634336 Bytes, Inter:292779 Bytes, Dram:292779 Bytes

I (845) wwe_example: esp_audio instance is:0x3c0d1fa4

I (853) wwe_example: Recorder has been created
I (856) MODEL_LOADER: The storage free size is 23744 KB
I (862) MODEL_LOADER: The partition size is 5120 KB
I (868) MODEL_LOADER: Successfully map model partition
I (873) RECORDER_SR: The first wakenet model: wn9_hilexin

I (880) AFE_SR: afe interface for speech recognition

I (885) AFE_SR: AFE version: SR_V220727

I (890) AFE_SR: Initial auido front-end, total channel: 3, mic num: 2, ref num: 1

I (898) AFE_SR: aec_init: 0, se_init: 1, vad_init: 1

I (904) AFE_SR: wakenet_init: 1

MC Quantized wakenet9: wakenet9_v1h24_hilexin_3_0.608_0.615, tigger:v3, mode:2, p:0, (Nov 17 2023 15:07:49)
I (1021) AFE_SR: wake num: 3, mode: 0, (Nov 17 2023 15:07:49)

Build fst from commands.
Quantized MultiNet6:rnnt_ctc_1.0, name:mn6_cn, (Nov 17 2023 15:07:49)
Quantized MultiNet6 search method: 2, time out:5.8 s
I (1941) main_task: Returned from app_main()

jason-mao, 2024-04-07:
@thechen130 Thanks for your report. Let's check and reply.

shootao, 2024-04-07:
Hi @thechen130 
我这边使用同样的 IDF/ADF 测试 esp-box-lite example wwe， 没有问题，log 如下
在测试前请打上下面的patch

```c
diff --git a/components/audio_board/esp32_s3_box_lite/board_def.h b/components/audio_board/esp32_s3_box_lite/board_def.h
index 7b1dc584..c1bad039 100644
--- a/components/audio_board/esp32_s3_box_lite/board_def.h
+++ b/components/audio_board/esp32_s3_box_lite/board_def.h
@@ -74,7 +74,7 @@
 #define HEADPHONE_DETECT            -1
 #define PA_ENABLE_GPIO              GPIO_NUM_46
 #define CODEC_ADC_I2S_PORT          (0)
-#define CODEC_ADC_BITS_PER_SAMPLE   I2S_BITS_PER_SAMPLE_32BIT
+#define CODEC_ADC_BITS_PER_SAMPLE   (32) /* 32bit */
 #define CODEC_ADC_SAMPLE_RATE       (48000)
 #define RECORD_HARDWARE_AEC         (false)
 #define BOARD_PA_GAIN               (0) /* Power amplifier gain defined by board (dB) */
diff --git a/components/audio_board/esp32_s3_box_lite/board_pins_config.c b/components/audio_board/esp32_s3_box_lite/board_pins_config.c
index cffc6238..d0fd207b 100644
--- a/components/audio_board/esp32_s3_box_lite/board_pins_config.c
+++ b/components/audio_board/esp32_s3_box_lite/board_pins_config.c
@@ -63,7 +63,7 @@ esp_err_t get_i2s_pins(int port, board_i2s_pin_t *i2s_config)
         i2s_config->data_in_num = -1;
         i2s_config->mck_io_num = -1;
     } else {
-        memset(i2s_config, -1, sizeof(i2s_pin_config_t));
+        memset(i2s_config, -1, sizeof(board_i2s_pin_t));
         ESP_LOGE(TAG, "i2s port %d is not supported", port);
         return ESP_FAIL;
     }
```

```c
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x403ce739
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1918
load:0x403c9700,len:0x4
load:0x403c9704,len:0xe5c
load:0x403cc700,len:0x302c
entry 0x403c993c
I (26) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader
I (27) boot: compile time Apr  7 2024 16:22:45
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 8MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00010000 00290000
I (100) boot:  4 model            Unknown data     01 82 002a0000 00500000
I (107) boot:  5 flash_tone       Unknown data     01 27 007a0000 00032000
I (115) boot: End of partition table
I (119) boot: No factory image, trying OTA 0
I (124) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=32358h (205656) map
I (164) esp_image: segment 1: paddr=00042380 vaddr=3fc9a300 size=056f0h ( 22256) load
I (168) esp_image: segment 2: paddr=00047a78 vaddr=40378000 size=085a0h ( 34208) load
I (176) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=87988h (555400) map
I (262) esp_image: segment 4: paddr=000d79b0 vaddr=403805a0 size=09cd4h ( 40148) load
I (278) boot: Loaded app from partition at offset 0x10000
I (307) boot: Set actual ota_seq=1 in otadata[0]
[... 74 lines cut ...]
Build fst from commands.
Quantized MultiNet6:rnnt_ctc_1.0, name:mn6_cn, (Nov 17 2023 15:07:49)
Quantized MultiNet6 search method: 2, time out:5.8 s
I (2073) main_task: Returned from app_main()
Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.
I (33133) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_START
I (33133) wwe_example: wakeup: vol -20.665825, mod idx 1, word idx 1
I (33133) CODEC_ELEMENT_HELPER: The element is 0x3c0d33fc. The reserve data 2 is 0x0.
I (34613) wwe_example: rec_engine_cb - REC_EVENT_VAD_START
W (34613) wwe_example: voice read begin
I (35483) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_START
I (35483) wwe_example: wakeup: vol -9.329292, mod idx 1, word idx 1
W (35483) AUDIO_RECORDER: Not in speeching, return 0
W (35493) wwe_example: audio recorder read finished 0
I (35523) CODEC_ELEMENT_HELPER: The element is 0x3c0d33fc. The reserve data 2 is 0x0.
I (36753) wwe_example: rec_engine_cb - REC_EVENT_VAD_START
W (36753) wwe_example: voice read begin
I (38423) wwe_example: rec_engine_cb - AUDIO_REC_COMMAND_DECT
W (38423) wwe_example: command 25, phrase_id 25, prob 0.936057, str:  da kai kong tiao
I (38433) CODEC_ELEMENT_HELPER: The element is 0x3c0d33fc. The reserve data 2 is 0x0.
I (38873) wwe_example: rec_engine_cb - REC_EVENT_VAD_STOP
W (38873) wwe_example: voice read stopped
I (39773) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_END
I (39773) wwe_example: Func:rec_engine_cb, Line:266, MEM Total:3497352 Bytes, Inter:242019 Bytes, Dram:242019 Bytes
```

shootao, 2024-04-19:
Hi @thechen130 
请问一下现在还有问题嘛 ？

closed 2024-04-19 by thechen130