espressif/esp-adf #1186: AUDIO_ELEMENT: [audio_forge] Read IO type ringbuf but ringbuf not set when #define NUMBER_SOURCE_FILE 2 in Audio Forge Example.

payday1991, 2024-04-04:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ *] I have read the documentation [Espressif Audio Development Guide](https://docs.espressif.com/projects/esp-adf/en/latest/index.html) and the issue is not addressed there.
- [* ] I have updated my ADF and IDF branch (master or release) to the latest version and checked that the issue is present there.
- [*] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps above, please follow these instructions so we can get the needed information to help you quickly and effectively.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your issues.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other Items If Possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the `Preview` tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: Please follow the above instructions and provide as many details as possible. It will save a lot of communication time and improve the efficiency of problem solving. The more details you provide, the faster we may be able to reproduce and resolve the issue. Thanks!**

----------------------------- Delete above -----------------------------

## Environment

- Audio development kit:      [none]
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): [N/A]
- [Required] Module or chip used:  [ESP32-S3-WROOM-1]
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    v5.1.2
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    v2.6-86-g1a03fe6a
- Build system:         [idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    xtensa-esp32s3-elf-gcc.exe (crosstool-NG esp-12.2.0_20230208) 12.2.0  
- Operating system:     [Windows]
- (Windows only) Environment type: [ESP Command Prompt]
- Using an IDE?: [Yes (VS Code with ESP-IDF extension)]
- Power supply:         [USB]

## Problem Description

Hi. I'm out of options and desperately need some help. Maybe somebody had this problem before. Basically, I'm just trying to use the Espressif's Audio Forge example(https://github.com/espressif/esp-adf/blob/master/examples/audio_processing/pipeline_audio_forge/main/audio_forge_pipeline_main.c) from ESP-ADF repository and compile it for my custom ESP32-S3 board. Everything works, and I **CAN HEAR** sound from the speaker when I

`#define NUMBER_SOURCE_FILE 1`,

but it's showing me the

"AUDIO_ELEMENT: [audio_forge] Read IO type ringbuf but ringbuf not set"

when I `#define NUMBER_SOURCE_FILE 2`.
The modification I've done was to add ```
for (int i = 0; i < NUMBER_SOURCE_FILE; i++) {
esp_err_t ret = audio_pipeline_run(pipeline[i]);
}
audio_pipeline_run(pipeline_mix);
``` 
before `while (1)` to start the audio pipeline because I don't have a board with the buttons.
This is my console output:

### Expected Behavior
I should hear 2 sounds simultaneously.
### Actual Behavior

E (1546) AUDIO_ELEMENT: [audio_forge] Read IO type ringbuf but ringbuf not set

And silence...
### Steps to Reproduce

1. #define NUMBER_SOURCE_FILE 2
[... 84 lines cut ...]
W (1556) AUDIO_ELEMENT: OUT-[file] AEL_IO_ABORT
W (1556) AUDIO_ELEMENT: [audio_forge] Element already stopped
W (1556) AUDIO_PIPELINE: There are no listener registered
I (1556) AUDIO_PIPELINE: audio_pipeline_unlinked
W (1556) AUDIO_ELEMENT: [file] Element has not create when AUDIO_ELEMENT_TERMINATE
W (1556) AUDIO_ELEMENT: [wav] Element has not create when AUDIO_ELEMENT_TERMINATE
W (1556) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINATE
W (1556) AUDIO_PIPELINE: There are no listener registered
I (1556) AUDIO_PIPELINE: audio_pipeline_unlinked
W (1556) AUDIO_ELEMENT: [file] Element has not create when AUDIO_ELEMENT_TERMINATE
W (1556) AUDIO_ELEMENT: [wav] Element has not create when AUDIO_ELEMENT_TERMINATE
W (1556) AUDIO_ELEMENT: [raw] Element has not create when AUDIO_ELEMENT_TERMINATE
W (1556) AUDIO_PIPELINE: There are no listener registered
I (1556) AUDIO_PIPELINE: audio_pipeline_unlinked
W (1556) AUDIO_ELEMENT: [audio_forge] Element has not create when AUDIO_ELEMENT_TERMINATE
I (1556) AUDIO_FORGE: audio_forge_destroy
I (1556) AUDIO_FORGE: Func:audio_forge_destroy, Line:406, MEM Total:8674748 Bytes, Inter:349959 Bytes, Dram:349959 Bytes

W (1556) AUDIO_ELEMENT: [iis] Element has not create when AUDIO_ELEMENT_TERMINATE
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
[sdkconfig.txt](https://github.com/espressif/esp-adf/files/14872947/sdkconfig.txt)

jason-mao, 2024-04-07:
@payday1991 Thanks for your report. Let's check and reply.

jason-mao, 2024-04-08:
It's fixed on b3add928e9ff82b7644502f9331eef97364772ff

closed 2024-04-08 by jason-mao

majingjing123, 2024-04-08:
Hi, this issue is already fixed.  If you only need mix, you can use example `audio_mixer_tone`.
If you must use audio forege, for your needs, you can use this test code 
`
/* Multiple pipeline playback with audio processing.

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_log.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "i2s_stream.h"
#include "mp3_decoder.h"
#include "fatfs_stream.h"
#include "audio_forge.h"
#include "raw_stream.h"
#include "board.h"
#include "periph_button.h"

static const char *TAG = "AUDIO_FORGE_PIPELINE";

#define DEFAULT_SAMPLERATE 44100
#define DEFAULT_CHANNEL 2
#define DEST_SAMPLERATE 11025
#define DEST_CHANNEL 1
#define TRANSMITTIME 0
#define MUSIC_GAIN_DB -10
#define NUMBER_SOURCE_FILE 2

int audio_forge_wr_cb(audio_element_handle_t el, char *buf, int len, TickType_t wait_time, void *ctx)
{
    audio_element_handle_t i2s_wr = (audio_element_handle_t)ctx;
    int ret = audio_element_output(i2s_wr, buf, len);
    return ret;
}

void app_main(void)
{
    audio_pipeline_handle_t pipeline[NUMBER_SOURCE_FILE] = {NULL};
    audio_element_handle_t fats_rd_el[NUMBER_SOURCE_FILE] = {NULL};
    audio_element_handle_t mp3_decoder[NUMBER_SOURCE_FILE] = {NULL};
    audio_element_handle_t el_raw_write[NUMBER_SOURCE_FILE] = {NULL};

    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set(TAG, ESP_LOG_INFO);

    ESP_LOGI(TAG, "[1.0] Start audio codec chip");
    audio_board_handle_t board_handle = audio_board_init();
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_DECODE, AUDIO_HAL_CTRL_START);

    ESP_LOGI(TAG, "[2.0] Start and wait for SDCARD to mount");
    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
    audio_board_sdcard_init(set, SD_MODE_1_LINE);
    audio_board_key_init(set);

    ESP_LOGI(TAG, "[3.0] Create pipeline_mix to mix");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    audio_pipeline_handle_t pipeline_mix = audio_pipeline_init(&pipeline_cfg);

    ESP_LOGI(TAG, "[3.1] Create audio_forge");
[... 145 lines cut ...]
    audio_pipeline_terminate(pipeline_mix);
    audio_pipeline_unregister_more(pipeline_mix, audio_forge, NULL);
    audio_pipeline_remove_listener(pipeline_mix);

    /* Stop all peripherals before removing the listener */
    esp_periph_set_stop_all(set);
    audio_event_iface_remove_listener(esp_periph_set_get_event_iface(set), evt);

    /* Make sure audio_pipeline_remove_listener & audio_event_iface_remove_listener are called before destroying event_iface */
    audio_event_iface_destroy(evt);

    for (int i = 0; i < NUMBER_SOURCE_FILE; i++) {
        /* Release resources */
        audio_pipeline_deinit(pipeline[i]);
        audio_element_deinit(fats_rd_el[i]);
        audio_element_deinit(mp3_decoder[i]);
        audio_element_deinit(el_raw_write[i]);
    }
    /* Release resources */
    audio_pipeline_deinit(pipeline_mix);
    audio_element_deinit(audio_forge);
    audio_element_deinit(i2s_writer);
    esp_periph_set_destroy(set);
}
`