espressif/esp-adf #1187: Add support for the ESP32-P4

ajsb85, 2024-04-04:
**Is your feature request related to a problem? Please describe.**

The lack of support for the ESP32-P4 microcontroller, specifically its ESP-RTC functionality, in the esp-adf repository poses limitations for developers aiming to leverage the full capabilities of this MCU in audio-related projects.

**Describe the solution you'd like**

I would like to see support for the ESP32-P4 MCU, including its ESP-RTC feature, integrated into the esp-adf repository. This integration would enable developers to seamlessly utilize the ESP32-P4's enhanced performance and real-time clock capabilities within the esp-adf framework, thereby expanding the range of supported hardware and enhancing the functionality of audio projects.

**Describe alternatives you've considered**

An alternative solution would be for developers to implement custom support for the ESP32-P4 and its ESP-RTC functionality outside of the esp-adf framework. However, this approach may be time-consuming and prone to compatibility issues, making it less ideal compared to having native support within the esp-adf repository.

**Additional context**

The ESP32-P4 is an increasingly popular microcontroller unit (MCU) offered by Espressif Systems, known for its enhanced performance and additional features compared to previous iterations in the ESP32 series. By adding support for the ESP32-P4 with its ESP-RTC feature to the esp-adf repository, developers would benefit from a more comprehensive and streamlined development experience for audio-related projects. This addition would contribute to the growth and versatility of the esp-adf framework, catering to the evolving needs of the developer community.

jason-mao, 2024-04-07:
@ajsb85 Thank you for your sincere advice. Please allow me to report on P4 support:

1. The esp-adf audio codec libraries are support ESP32P4
2. The ESP-RTC and other protocols are working; due to P4 having no wifi module, we have to develop external components such as esp_hoested and esp_extconn to make IDF network APIs work on P4.

shootao, 2025-02-14:
Hi @ajsb85 
ADF now supports ESP32p4 related development boards almost directly

closed 2025-02-14 by jason-mao