espressif/esp-adf #1194: Support for ESP32-S3-EYE v2.2

gamename, 2024-04-10:
Please add support for ESP32-S3-EYE for recording audio and for sound identification.

shootao, 2024-04-18:
Hi @gamename 
[0001-audio_board-support-esp32-s3-eye-board.patch](https://github.com/espressif/esp-adf/files/15019314/0001-audio_board-support-esp32-s3-eye-board.patch)
This patch support the ESP32-S3-EYE board witch is based on WWE example, please have a try

gamename, 2024-04-18:
> Hi @gamename [0001-audio_board-support-esp32-s3-eye-board.patch](https://github.com/espressif/esp-adf/files/15019314/0001-audio_board-support-esp32-s3-eye-board.patch) This patch support the ESP32-S3-EYE board which is based on WWE example, please have a try

Thanks! That's great!

shootao, 2024-07-17:
Hi @projectgus 
If there are no more issues, please close this issue. Thank you