espressif/esp-adf #1197: Service sounds are played poorly in the voip project

Olejan, 2024-04-23:
## Environment

- Audio development kit:      [ESP32-LyraT-Mini_v1.2]
- Audio kit version (for ESP32-LyraT-Mini): [v1]
- [Required] Module or chip used:  [ESP32-WROVER-E]
- [Required] IDF version: [v4.4.4-278-g3c8bc2213c]
- [Required] ADF version: [v2.6-108-g1f275b50]
- Build system:         [idf.py]
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version: [xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0]
- Operating system:     [Windows]
- (Windows only) Environment type: [ESP Command Prompt (cmd)]
- Using an IDE?: [No]
- Power supply:         [USB]

## Problem Description

I'm using the [voip](https://github.com/espressif/esp-adf/tree/master/examples/protocols/voip) project. A very good project, but there is an incomprehensible point - service audio signals (```Alarm.mp3, WiFi_Success.mp3```, ...), when played by the device, have very poor quality. It seems that the task of playing an audio file is very often interrupted by other tasks, and the sound in a speaker is very intermittent, almost indistinguishable. At the same time, I can normally call from a device to a SIP phone, or from a phone to a device, and the sound in the speaker will be very good. I suspect I need to configure the amount of memory allocated for the task of playing service signals, but I don’t know where this can be done.

Please tell me how to make the service signals sound good?

[Video](https://youtu.be/GG2p1XW9OHo)

## Debug Logs

```

d:\voip>idf.py monitor
Executing action: monitor
Serial port COM49
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Running idf_monitor in directory D:\voip
Executing "C:\Users\User\.espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe C:\Users\User\esp\esp-adf\esp-idf\tools/idf_monitor.py -p COM49 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 D:\voip\build\voip_app.elf -m 'C:\Users\User\.espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe' 'C:\Users\User\esp\esp-adf\esp-idf\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM49 instead...
--- idf_monitor on \\.\COM49 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6976
load:0x40078000,len:15780
ho 0 tail 12 room 4
load:0x40080400,len:4516
0x40080400: _init at ??:?

entry 0x400806b8
I (29) boot: ESP-IDF v4.4.4-278-g3c8bc2213c-dirty 2nd stage bootloader
I (29) boot: compile time 16:51:05
I (30) boot: chip revision: v3.1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32: SPI Speed      : 80MHz
I (44) boot.esp32: SPI Mode       : QIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00200000
I (92) boot:  3 flash_tone       Unknown data     01 ff 00210000 0007d000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=35770h (218992) map
I (172) esp_image: segment 1: paddr=00045798 vaddr=3ffb0000 size=03390h ( 13200) load
[... 802 lines cut ...]
I (51565) AUDIO_SYS: | IDLE              | 1001893     |49%  | 0    | 912       | Ready   | 1        | Intr
I (51576) AUDIO_SYS: | IDLE              | 953866      |47%  | 0    | 924       | Ready   | 0        | Intr
I (51586) AUDIO_SYS: | esp_periph        | 3424        | 0%  | 5    | 2184      | Blocked | 0        | Intr
I (51596) AUDIO_SYS: | sip_task          | 3096        | 0%  | 20   | 8036      | Blocked | 0        | Extr
I (51607) AUDIO_SYS: | tiT               | 363         | 0%  | 18   | 2164      | Blocked | 7fffffff | Intr
I (51617) AUDIO_SYS: | button_task       | 25274       | 1%  | 10   | 1568      | Blocked | 0        | Intr
I (51628) AUDIO_SYS: | Tmr Svc           | 0           | 0%  | 1    | 1460      | Blocked | 0        | Intr
I (51638) AUDIO_SYS: | sys_evt           | 0           | 0%  | 20   | 1644      | Blocked | 0        | Intr
I (51649) AUDIO_SYS: | ipc1              | 0           | 0%  | 24   | 904       | Blocked | 1        | Intr
I (51659) AUDIO_SYS: | esp_dispatcher    | 0           | 0%  | 10   | 1592      | Blocked | 0        | Intr
I (51669) AUDIO_SYS: | ipc0              | 0           | 0%  | 24   | 948       | Blocked | 0        | Intr
I (51680) AUDIO_SYS: | media_task        | 0           | 0%  | 6    | 1796      | Blocked | 0        | Extr
I (51690) AUDIO_SYS: | esp_timer         | 1127        | 0%  | 22   | 3300      | Suspended | 0        | Intr
I (51701) AUDIO_SYS: | wifi              | 12870       | 0%  | 23   | 4036      | Blocked | 0        | Intr
I (51711) AUDIO_SYS: | wifi_serv         | 0           | 0%  | 6    | 748       | Blocked | 0        | Extr
I (51722) AUDIO_SYS: | input_key_servi   | 0           | 0%  | 5    | 2232      | Blocked | 1        | Extr

I (51732) VoIP_EXAMPLE: Func:app_main, Line:212, MEM Total:4249335 Bytes, Inter:242595 Bytes, Dram:203803 Bytes
```

## Other Items If Possible

- [ ] sdkconfig file (Attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**Note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

redhyyy, 2024-04-24:
兄弟你那个乐鑫自建的SIP服务器能注册得上嘛，我从前天开始就登不上了，能分享下授权信息嘛我qq：1768815743

ktoto2011, 2024-04-30:
I'm also having the same issue, the previous issue where the firmware won't play a tone on an incoming call has been found here. However, the sound played by this workaround is completely distorted.https://github.com/espressif/esp-adf/issues/1174

ktoto2011, 2024-04-30:
https://github.com/espressif/esp-adf/assets/44020882/554170b0-2ac4-4f57-9710-f66306c27f63

This is the same hardware running pipeline_flash_tone firmware for the beep effect

ktoto2011, 2024-04-30:
https://github.com/espressif/esp-adf/assets/44020882/a6b76ea3-2ffc-49d8-ae13-0b7f62c5c6bb

This is the same hardware running void firmware with a beep effect, and the sound is very distorted