espressif/esp-adf #1198: ESP-RTC Example 乐鑫SIP服务器 401 Unauthorized

redhyyy, 2024-04-24:
SIP/2.0 401 Unauthorized
Via: SIP/2.0/TCP 192.168.43.193:50751;branch=z9hG4bK--1468416954;rport=54613;received=36.21.20.76
From: <sip:1012@52.81.228.8:9670>;tag=2018339741
To: <sip:1012@52.81.228.8:9670>;tag=Sc03HvFaHtv1p
Call-ID: 7DBFBD659E95FBAD44C1EFD4DF84FEB6FA51D3177AED
CSeq: 9 REGISTER
User-Agent: FreeSWITCH-mod_sofia/1.10.6-release~64bit
Allow: INVITE, ACK, BYE, CANCEL, OPTIONS, MESSAGE, INFO, UPDATE, REGISTER, REFER, NOTIFY, PUBLISH, SUBSCRIBE
Supported: path, replaces
WWW-Authenticate: Digest realm="52.81.228.8", nonce="c450354e-bce9-43f5-bb7e-cf00fe316a53", stale=true, algorithm=MD5, qop="auth"
Content-Length: 0

这两天开始没有办法注册用户了，是为什么嗷，想问问看还有没有测试服务器可以使用

davidHuW, 2024-04-28:
可以用其他开源的服务器代替，比如FreeSWITCH，Asterisk等

shootao, 2024-07-17:
@redhyyy 前些天内部搭建的确实宕机了，现在你可以再试试

redhyyy, 2024-07-27:
蟹蟹，辛苦了，已经可以连上了，感谢

closed 2024-07-31 by jason-mao