espressif/esp-adf #1200: Unable to run the examples play_mp3_control on esp32-s3-korvo-2 v3.1 development board , getting A stack overflow in task ipc0 has been detected error

Root2Technologies, 2024-04-27:
I have purchased esp32-s3-korvo-2 v3.1 development board and followed the getting started guide , but unable to run the sample code , it reporting a --- A stack overflow in task ipc0 has been detected error  , can you please help me find the issue .Thanks  
the build and flasing are successful but getting error later  
below is the output log when i monitor 
WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2da0
entry 0x403c9914
I (27) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (27) boot: compile time Apr 27 2024 12:32:34
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=914d8h (595160) map
I (206) esp_image: segment 1: paddr=000a1500 vaddr=3fc95200 size=02d30h ( 11568) load
I (209) esp_image: segment 2: paddr=000a4238 vaddr=40374000 size=0bde0h ( 48608) load
I (222) esp_image: segment 3: paddr=000b0020 vaddr=42000020 size=372c4h (225988) map
I (263) esp_image: segment 4: paddr=000e72ec vaddr=4037fde0 size=0535ch ( 21340) load
I (275) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
I (287) cpu_start: Multicore app
I (296) cpu_start: Pro cpu start user code
I (297) cpu_start: cpu freq: 160000000 Hz
I (297) cpu_start: Application information:
I (300) cpu_start: Project name:     play_mp3_control
I (305) cpu_start: App version:      1
I (310) cpu_start: Compile time:     Apr 27 2024 12:31:31
I (316) cpu_start: ELF file SHA256:  3653e8b15...
I (321) cpu_start: ESP-IDF:          v5.2.1
I (326) cpu_start: Min chip rev:     v0.0
I (331) cpu_start: Max chip rev:     v0.99
I (336) cpu_start: Chip rev:         v0.2
I (340) heap_init: Initializing. RAM available for dynamic allocation:
I (348) heap_init: At 3FC98B48 len 00050BC8 (322 KiB): RAM
I (354) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (360) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (366) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (373) spi_flash: detected chip: gd
I (376) spi_flash: flash io: dio
W (380) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (391) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (399) sleep: Configure to isolate all GPIO pins in sleep state
I (406) sleep: Enable automatic switching of GPIO sleep configuration
I (414) main_task: Started on CPU0
I (424) main_task: Calling app_main()
I (424) PLAY_FLASH_MP3_CONTROL: [ 1 ] Start audio codec chip
W (454) I2C_BUS: I2C bus has been already created, [port:0]
W (474) ES7210: Enable TDM mode. ES7210_SDP_INTERFACE2_REG12: 2
I (494) PLAY_FLASH_MP3_CONTROL: [ 2 ] Create audio pipeline, add all elements to pipeline, and subscribe pipeline event
I (494) PLAY_FLASH_MP3_CONTROL: [2.1] Create mp3 decoder to decode mp3 file and set custom read callback
I (504) PLAY_FLASH_MP3_CONTROL: [2.2] Create i2s stream to write data to codec chip
I (514) PLAY_FLASH_MP3_CONTROL: [2.3] Register all elements to audio pipeline
[... 63 lines cut ...]
I (340) cpu_start: Chip rev:         v0.2
I (345) heap_init: Initializing. RAM available for dynamic allocation:
I (352) heap_init: At 3FC98B48 len 00050BC8 (322 KiB): RAM
I (358) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (364) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (370) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (378) spi_flash: detected chip: gd
I (381) spi_flash: flash io: dio
W (385) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (396) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (404) sleep: Configure to isolate all GPIO pins in sleep state
I (411) sleep: Enable automatic switching of GPIO sleep configuration
I (418) main_task: Started on CPU0
I (428) main_task: Calling app_main()
I (428) PLAY_FLASH_MP3_CONTROL: [ 1 ] Start audio codec chip
W (458) I2C_BUS: I2C bus has been already created, [port:0]
W (478) ES7210: Enable TDM mode. ES7210_SDP_INTERFACE2_REG12: 2
I (498) PLAY_FLASH_MP3_CONTROL: [ 2 ] Create audio pipeline, add all elements to pipeline, and subscribe pipeline event
I (498) PLAY_FLASH_MP3_CONTROL: [2.1] Create mp3 decoder to decode mp3 file and set custom read callback
I (508) PLAY_FLASH_MP3_CONTROL: [2.2] Create i2s stream to write data to codec chip
I (518) PLAY_FLASH_MP3_CONTROL: [2.3] Register all elements to audio pipeline
I (528) PLAY_FLASH_MP3_CONTROL: [2.4] Link it together [mp3_music_read_cb]-->mp3_decoder-->i2s_stream-->[codec_chip]
I (538) PLAY_FLASH_MP3_CONTROL: [ 3 ] Initialize peripherals
 | P  ?W   ?W ↔♦<♦◄]☻ p  ?   ?♥W#
♠ g♦<   ?`Q☻    ?   ?ܖ ?   ?W

hbler99, 2024-07-12:
Enlarging the IPC task stack size to 1280 in `idf.py menuconfig -> TOP -> Component config -> IPC` should resolve this issue.

jason-mao, 2024-07-24:
The stack overflow occurred because there was no free ISR level trigger to print the error message, fixed on 2c6b0603bffcedc8beae979ece6d09a80dfe59b5

closed 2024-07-24 by jason-mao