espressif/esp-adf #1201: No way to allocate a new PERIPH_ID_

cskilbeck, 2024-04-30:
**Is your feature request related to a problem? Please describe.**

When defining a new peripheral ID, there's no way to allocate a new ID.

**Describe the solution you'd like**

It would be useful if there was a block of IDs for the purpose of user defined peripheral IDs.

**Describe alternatives you've considered**

For now I hard code my peripheral ID as follows:
`#define PERIPH_ID_ENCODER (PERIPH_ID_LCD + 1)`

**Additional context**

[Gist](https://gist.github.com/cskilbeck/1284c97ad3051ae012170b781ca5b5e9)

shootao, 2024-07-17:
Hi @cskilbeck 
I thinks the follow patch can meet you need
```c
diff --git a/components/esp_peripherals/esp_peripherals.c b/components/esp_peripherals/esp_peripherals.c
index 5e141123..b6cb75c4 100644
--- a/components/esp_peripherals/esp_peripherals.c
+++ b/components/esp_peripherals/esp_peripherals.c
@@ -35,6 +35,7 @@

 static const char *TAG = "ESP_PERIPH";

+#define DEFAULT_CUS_PERIPH_ID_MAX_NUM      (5)
 #define DEFAULT_ESP_PERIPH_WAIT_TICK       (10/portTICK_RATE_MS)

 struct esp_periph {
@@ -62,6 +63,7 @@ typedef struct esp_periph_sets {
     bool                                            ext_stack;
     bool                                            run;
     esp_periph_event_t                              event_handle;
+    int                                             periph_id[DEFAULT_CUS_PERIPH_ID_MAX_NUM];
     STAILQ_HEAD(esp_periph_list_item, esp_periph)   periph_list;
 } esp_periph_set_t;

@@ -199,6 +201,24 @@ esp_err_t esp_periph_remove_from_set(esp_periph_set_handle_t periph_set_handle,
     return ESP_OK;
 }

+esp_err_t esp_periph_allocated_periph_id(esp_periph_set_handle_t periph_set_handle, int *periph_id)
+{
+    if (periph_set_handle == NULL) {
+        AUDIO_ERROR(TAG, "Peripherals have not been initialized");
+        return ESP_FAIL;
+    }
+    for (int i = 0; i < DEFAULT_CUS_PERIPH_ID_MAX_NUM; i++) {
+        if (periph_set_handle->periph_id[i] == 0) {
+            *periph_id = i + PERIPH_ID_MAX;
+            periph_set_handle->periph_id[i] = *periph_id;
+            return ESP_OK;
+        }
+    }
+    ESP_LOGE(TAG, "Have no enough periph id space");
+    *periph_id = -1;
+    return ESP_FAIL;
+}
+
 esp_err_t esp_periph_set_destroy(esp_periph_set_handle_t periph_set_handle)
 {
     if (periph_set_handle == NULL) {
diff --git a/components/esp_peripherals/include/esp_peripherals.h b/components/esp_peripherals/include/esp_peripherals.h
index 0ea9dbad..ed1fb3a5 100644
--- a/components/esp_peripherals/include/esp_peripherals.h
+++ b/components/esp_peripherals/include/esp_peripherals.h
@@ -54,7 +54,8 @@ typedef enum {
     PERIPH_ID_GPIO_ISR   = AUDIO_ELEMENT_TYPE_PERIPH + 14,
     PERIPH_ID_WS2812     = AUDIO_ELEMENT_TYPE_PERIPH + 15,
     PERIPH_ID_AW2013     = AUDIO_ELEMENT_TYPE_PERIPH + 16,
-    PERIPH_ID_LCD        = AUDIO_ELEMENT_TYPE_PERIPH + 17
+    PERIPH_ID_LCD        = AUDIO_ELEMENT_TYPE_PERIPH + 17,
+    PERIPH_ID_MAX        = AUDIO_ELEMENT_TYPE_PERIPH + 18
 } esp_periph_id_t;

 /**
@@ -239,6 +240,20 @@ esp_err_t esp_periph_set_list_destroy(esp_periph_set_handle_t periph_set_handle)
  */
 esp_err_t esp_periph_remove_from_set(esp_periph_set_handle_t periph_set_handle, esp_periph_handle_t periph);

+/**
+ * @brief      Allocated a peripheral id from the periph_set.
+ *
+ * Note: This function is used to get a peripheral id from the periph_set, and this periph id not include in the esp_periph_id_t
+ *
+ * @param      periph_set_handle    The esp_periph_set_handle_t instance
+ * @param      periph_id            The peripheral identifier
+ *
+ * @return
+ *     - ESP_OK
+ *     - ESP_FAIL
+ */
+esp_err_t esp_periph_allocated_periph_id(esp_periph_set_handle_t periph_set_handle, int *periph_id);
+
 /**
  * @brief      Call this function to change periph_set waiting time.
  *
```

shootao, 2024-07-17:
Hi @cskilbeck 
I think the follow patch can meet your need
```c
diff --git a/components/esp_peripherals/esp_peripherals.c b/components/esp_peripherals/esp_peripherals.c
index 5e141123..b6cb75c4 100644
--- a/components/esp_peripherals/esp_peripherals.c
+++ b/components/esp_peripherals/esp_peripherals.c
@@ -35,6 +35,7 @@

 static const char *TAG = "ESP_PERIPH";

+#define DEFAULT_CUS_PERIPH_ID_MAX_NUM      (5)
 #define DEFAULT_ESP_PERIPH_WAIT_TICK       (10/portTICK_RATE_MS)

 struct esp_periph {
@@ -62,6 +63,7 @@ typedef struct esp_periph_sets {
     bool                                            ext_stack;
     bool                                            run;
     esp_periph_event_t                              event_handle;
+    int                                             periph_id[DEFAULT_CUS_PERIPH_ID_MAX_NUM];
     STAILQ_HEAD(esp_periph_list_item, esp_periph)   periph_list;
 } esp_periph_set_t;

@@ -199,6 +201,24 @@ esp_err_t esp_periph_remove_from_set(esp_periph_set_handle_t periph_set_handle,
     return ESP_OK;
 }

+esp_err_t esp_periph_allocated_periph_id(esp_periph_set_handle_t periph_set_handle, int *periph_id)
+{
+    if (periph_set_handle == NULL) {
+        AUDIO_ERROR(TAG, "Peripherals have not been initialized");
+        return ESP_FAIL;
+    }
+    for (int i = 0; i < DEFAULT_CUS_PERIPH_ID_MAX_NUM; i++) {
+        if (periph_set_handle->periph_id[i] == 0) {
+            *periph_id = i + PERIPH_ID_MAX;
+            periph_set_handle->periph_id[i] = *periph_id;
+            return ESP_OK;
+        }
+    }
+    ESP_LOGE(TAG, "Have no enough periph id space");
+    *periph_id = -1;
+    return ESP_FAIL;
+}
+
 esp_err_t esp_periph_set_destroy(esp_periph_set_handle_t periph_set_handle)
 {
     if (periph_set_handle == NULL) {
diff --git a/components/esp_peripherals/include/esp_peripherals.h b/components/esp_peripherals/include/esp_peripherals.h
index 0ea9dbad..ed1fb3a5 100644
--- a/components/esp_peripherals/include/esp_peripherals.h
+++ b/components/esp_peripherals/include/esp_peripherals.h
@@ -54,7 +54,8 @@ typedef enum {
     PERIPH_ID_GPIO_ISR   = AUDIO_ELEMENT_TYPE_PERIPH + 14,
     PERIPH_ID_WS2812     = AUDIO_ELEMENT_TYPE_PERIPH + 15,
     PERIPH_ID_AW2013     = AUDIO_ELEMENT_TYPE_PERIPH + 16,
-    PERIPH_ID_LCD        = AUDIO_ELEMENT_TYPE_PERIPH + 17
+    PERIPH_ID_LCD        = AUDIO_ELEMENT_TYPE_PERIPH + 17,
+    PERIPH_ID_MAX        = AUDIO_ELEMENT_TYPE_PERIPH + 18
 } esp_periph_id_t;

 /**
@@ -239,6 +240,20 @@ esp_err_t esp_periph_set_list_destroy(esp_periph_set_handle_t periph_set_handle)
  */
 esp_err_t esp_periph_remove_from_set(esp_periph_set_handle_t periph_set_handle, esp_periph_handle_t periph);

+/**
+ * @brief      Allocated a peripheral id from the periph_set.
+ *
+ * Note: This function is used to get a peripheral id from the periph_set, and this periph id not include in the esp_periph_id_t
+ *
+ * @param      periph_set_handle    The esp_periph_set_handle_t instance
+ * @param      periph_id            The peripheral identifier
+ *
+ * @return
+ *     - ESP_OK
+ *     - ESP_FAIL
+ */
+esp_err_t esp_periph_allocated_periph_id(esp_periph_set_handle_t periph_set_handle, int *periph_id);
+
 /**
  * @brief      Call this function to change periph_set waiting time.
  *
```

cskilbeck, 2024-07-17:
Looks good, thanks!

closed 2024-07-17 by cskilbeck