espressif/esp-adf #1203: ESP32-LyraTD-MSC board can't run ESP-ADF wwe example.

hyuan-kamuda, 2024-05-08:
**## Environment**

- Audio development kit:      ESP32-LyraTD-MSC
- Audio kit version: v2.2
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version: v5.2.1
- [Required] ADF version: v2.6-108-g1f275b50
- Build system:        visual code/idf.py
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version : xtensa-esp-elf-gcc.exe (crosstool-NG esp-13.2.0_20230928) 13.2.0
- Operating system: Windows
- (Windows only) Environment type: PowerShell
- Using an IDE?:  Yes, Visual Code+ IDF extension
- Power supply:         USB

**## Problem Description**

ESP32-LyraTD-MSC board can't run ESP-ADF wwe example, and results in CPU halt/crash.
With ‘I (12132) main_task: Returned from app_main()“ log printed, any speech input the system has no response, after xx seconds, the crash will happen. 

**### Expected Behavior**
Should work normally. 

**### Actual Behavior**
Crash.

**### Steps to Reproduce**

1.  From https://github.com/espressif/esp-adf/tree/master/examples readme table, wwe example(examples/speech_recognition/wwe) should be supported by ESP32-LyraTD-MSC board. And tried another speech_recognition example(examples/speech_recognition/vad) with same board which can work normally. 
2.  In VSCode, create a project from example of wwe. 
3.  Adjusted some memory/system settings following https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/design-considerations.html:
     a. Enabled SPI PSRAM. The option CONFIG_SPIRAM_CACHE_WORKAROUND, set by default in the same menu, kept enabled.
     b. Not enable Bluetooth.
     c. CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST set as “yes” 
     d. Set CONFIG_WL_SECTOR_SIZE as 512 in Component config > Wear Levelling
     e. Flash SPI mode as QIO
     f.  Flash SPI speed as 80MHz
     g. CPU frequency as 240MHz
     h. Set Default receive window size as 5 times(7200) greater than Maximum Segment Size(1440) in Component config > LWIP > TCP
     i.  Set Flash Size as 4MB
4.  Build the application and flash/run it, system will hang, found there's warning: "W (10615) RECORDER_SR: Multinet is not enabled in SDKCONFIG" , then:
     . enable "use multinet" 
     . Chinese Speech Commands Model: Chinese Single Recognition(mn2_cn)
     . Select wake words: nihaoxiaozhi(wn5_nihaoxiaozi)
6.  Rebuild application, and flash, run it. 

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7524
load:0x40078000,len:16528
ho 0 tail 12 room 4      
load:0x40080400,len:4    
0x40080400: _init at ??:?

load:0x40080404,len:4312 
entry 0x4008065c
I (31) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader
I (31) boot: compile time May  7 2024 23:55:14  
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz       
I (49) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00004000
[... 129 lines cut ...]
EXCVADDR: 0x4015c205  LBEG    : 0x400f8841  LEND    : 0x400f8866  LCOUNT  : 0x00000000
0x400f8841: WindowingEnergy at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/noise_suppression/webrtc_ns.c:1645
 (inlined by) WebRtcNs_ProcessCore at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/noise_suppression/webrtc_ns.c:1916
0x400f8866: WebRtcNs_ProcessCore at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/noise_suppression/webrtc_ns.c:1917

Backtrace: 0x4011e70c:0x3f826950 0x400f69c2:0x3f8269a0 0x400f8a39:0x3f8269f0 0x400f95be:0x3f827b10 0x400f4b89:0x3f827b30 0x400f36da:0x3f827b60 0x400f3898:0x3f827b80 0x400dd015:0x3f827ba0
0x4011e70c: bitrv2 at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/noise_suppression/webrtc_ns.c:230
0x400f69c2: WebRtc_rdft at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/noise_suppression/webrtc_ns.c:155
0x400f8a39: FFT at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/noise_suppression/webrtc_ns.c:1564
 (inlined by) WebRtcNs_ProcessCore at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/noise_suppression/webrtc_ns.c:1944
0x400f95be: WebRtcNs_Process at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/noise_suppression/webrtc_ns.c:2126
0x400f4b89: ns_process at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/esp_ns.c:16
0x400f36da: afe_feed_aec_init_false at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr.c:546
0x400f3898: afe_feed at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr.c:623
0x400dd015: feed_task at C:/Users/hyuan/esp/esp-adf/components/audio_recorder/recorder_sr.c:248

ELF file SHA256: b05478608

CPU halted.

## Other Items If Possible

- [ X] sdkconfig file (Attach the sdkconfig file from your project folder), please see attache sdkconfig. 

[sdkconfig.txt](https://github.com/espressif/esp-adf/files/15243313/sdkconfig.txt)

xkkjiayou, 2024-05-30:
我也是这个问题，解决了吗
bitrv2 at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_processor/noise_suppression/webrtc_ns.c:234

hbler99, 2024-07-15:
Have you execute `tw_upload_dsp_firmware(-1) ===> tw_upload_dsp_firmware(0); ` after first flash? This change is to update the firmware for zl38063.c, and after updating, it needs to be reverted back.

jason-mao, 2024-07-31:
The ESP32-LyraTD-MSC is EOL, please test WWE with other boards.

closed 2024-07-31 by jason-mao