espressif/esp-adf #1204: ADF wants to support websocket streaming

lihuimin01, 2024-05-08:
**Is your feature request related to a problem? Please describe.**

A clear and concise description of what the problem is. For example, I'm always frustrated when [...]

**Describe the solution you'd like**

A clear and concise description of what you want to happen.

**Describe alternatives you've considered**

A clear and concise description of any alternative solutions or features you've considered.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

**Additional context**

Add any other context or screenshots about the feature request here.

lihuimin01, 2024-05-08:
If there is no support plan to recommend it, please recommend to do real-time speech recognition demo based on Websocket protocol

Ahypnis, 2024-08-31:
@jason-mao I need this. Do you have any plans to incorporate new features?