espressif/esp-adf #1209: 如何接管手机端的音量控制

robbietree8, 2024-05-31:
## Environment

- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    v4.4.3-210-g932b6699ac
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    v2.4.1-12-g81bac5f
- Build system:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0

## Problem Description

目前手机端的音量调节会直接反应到硬件端，我的需求是这个音量调节只能通过硬件端本身的按钮发起，忽略手机端的音量调节指令。

### Sample Code

目前是直接用了bluetooth_service，参考例程是`play_bt_music_example.c`

jason-mao, 2024-06-05:
你可以把https://github.com/espressif/esp-adf/blob/37a49af8ff785ba09492b8fa41b8da7dc4db1c30/examples/player/pipeline_a2dp_sink_stream/main/play_bt_music_example.c#L123，这个配置NULL， a2dp_stream.c 中的 bt_avrc_volume_set_by_local，bt_avrc_volume_set_by_controller就不会工作。

robbietree8, 2024-06-05:
@jason-mao 谢谢，依此调整代码后，能正常工作了。不过遇到了一个新的问题，本地的音频和蓝牙音频在同等音量大小下，感受到的声音大小不一样，蓝牙音频明显响于本地音频，我再研究一下。

closed 2024-06-05 by robbietree8