espressif/esp-adf #1211: English documentation is broken

deniz-frick, 2024-06-04:
## Environment
Not applicable

## Problem Description
English documentation does not exist anymore.
It is just a copy of the Chinese documentation.

Read the docs automatically uses any conf.py for Sphinx and it uses the chinese one by default as seen [here](https://readthedocs.com/projects/espressif-esp-adf/builds/2255926/).

### Expected Behavior
English documentation should exist.

### Actual Behavior
It does not.

### Steps to Reproduce

Go here [https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/dev-boards/get-started-esp32-lyrat.html](https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/dev-boards/get-started-esp32-lyrat.htm)

jason-mao, 2024-06-05:
@deniz-frick Thanks for your report, I am fixing it.

jason-mao, 2024-06-06:
It fixed on 0593164bf7e64c268dad56e332d6f07f0e9c37ab.

closed 2024-06-06 by jason-mao