espressif/esp-adf #1212: ESP32 LyraT keeps ringing after hanging up SIP call

carlosrodridel, 2024-06-10:
## Environment

- Audio development kit:      ESP32-LyraT
- Audio kit version: v4
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version: v4.4.5
- [Required] ADF version: v2.6-114-g0593164b
- Build system:   idf.py
- [Required] Running log:  
```
==========================================================================
I (10524) SIP_SERVICE: ESP_RTC_EVENT_REGISTERED
W (10529) SIP: CHANGE STATE FROM 1, TO 2, :func: sip_register:1863
I (12842) SIP: [1970-01-01/00:00:05]<<=====READ 0969 bytes==
I (12842) SIP: 

INVITE sip:X@192.168.2.2:14077;transport=udp SIP/2.0
Call-ID: 19832-RQ-db9531d1-45ee3cf06@sip3.ovh.fr
Contact: <sip:91.121.129.159:5060>
Content-Type: application/sdp
CSeq: 1740970801 INVITE
From: "+33XX" <sip:0XX@sip3.ovh.fr;user=phone>;tag=19832-SL-db9531d2-2ee7c6865
Max-Forwards: 29
Record-Route: <sip:91.121.129.159:5060;lr>;session=75155
To: <sip:0184250727@10.7.1.68;user=phone>
Via: SIP/2.0/UDP 91.121.129.159:5060;branch=z9hG4bK-FRRA-c9bed035-2f5362dc
Allow: REFER,INVITE,NOTIFY,ACK,UPDATE,OPTIONS,REGISTER,SUBSCRIBE,NOTIFY,CANCEL,BYE,PRACK
User-Agent: Cirpack/v4.76 (gw_sip)
Content-Length: 323

v=0
o=anonymous 171802877280 171802877280 IN IP4 91.121.129.159
s=SIP Call
c=IN IP4 91.121.129.146
t=0 0
m=audio 31586 RTP/AVP 8 0 18 101
b=AS:82
a=rtpmap:8 PCMA/8000/1
a=rtpmap:0 PCMU/8000/1
a=rtpmap:18 G729/8000/1
a=fmtp:18 annexb=no
a=rtpmap:101 telephone-event/8000
a=fmtp:101 0-15
a=ptime:20
a=sendrecv

I (12927) SIP: [1970-01-01/00:00:05]<<======================
I (12935) SIP: Remote ARTP port=31586
I (12938) SIP: Remote RTP addr=91.121.129.146
I (12943) SIP: call from 0XX
I (12947) SIP: [1970-01-01/00:00:05]=======WRITE 0543 bytes>>
I (12954) SIP: 

SIP/2.0 100 Trying
Via: SIP/2.0/UDP 91.121.129.159:5060;branch=z9hG4bK-FRRA-c9bed035-2f5362dc
Record-Route: <sip:91.121.129.159:5060;lr>;session=75155
Contact: <sip:X@192.168.2.2:14077;transport=UDP>
From: "+33XX" <sip:0XX@sip3.ovh.fr;user=phone>;tag=19832-SL-db9531d2-2ee7c6865
To: <sip:0184250727@10.7.1.68;user=phone>;tag=1804088857
Call-ID: 19832-RQ-db9531d1-45ee3cf06@sip3.ovh.fr
CSeq: 1740970801 INVITE
Server: ESP32 SIP/2.0
Allow: ACK, INVITE, BYE, UPDATE, CANCEL, OPTIONS, INFO
Content-Length: 0

I (13005) SIP: [1970-01-01/00:00:05]=======================>>
I (13014) SIP: [1970-01-01/00:00:05]=======WRITE 0543 bytes>>
I (13017) SIP: 

SIP/2.0 180 Ringing
[... 512 lines cut ...]

After hanging up the call from the ESP32 LyraT board, the board should not receive any further "ESP_RTC_EVENT_INCOMING" events unless a new call is made.

### Actual Behavior

After hanging up the call, the ESP32 LyraT board continues to receive "ESP_RTC_EVENT_INCOMING" events, causing it to ring as if it is receiving a new call.

### Steps to Reproduce

1. Set up the ESP32 LyraT with the ESP-ADF VOIP example.
2. Use the OVH SIP service to make a call to the ESP32 LyraT from any phone.
3. Hang up the call from the ESP32 LyraT board.
4. Observe that the board continues to ring due to receiving "ESP_RTC_EVENT_INCOMING".

### Additional Information:

- OVH SIP Service Behavior: It appears that OVH sends another INVITE SDP package that needs to be replied to with a 200 OK SDP. If not replied to, OVH keeps sending the same package even when the call has ended, which causes the board to ring, thinking it's a new call.

#### Wireshark traces when calling to the LyraT
[image]

- When making the same call using the "Telephone" application on MAC, this problem does not occur. This suggests that the issue is specific to how the ESP-ADF handles the SIP communication with OVH.

#### Wireshark traces when calling to "Telephone" Mac application
[image]

TempoTian, 2024-06-13:
I have checked the log seems the issue is caused by second invite response not contain sdp body, I have build one version with response with sdp also when reply after connection build, you can have a try. I just upload the version for ESP32, you can replace it   to folder https://github.com/espressif/esp-adf-libs/tree/master/esp_media_protocols/lib/esp32.
[libesp_media_protocols.zip](https://github.com/user-attachments/files/15816449/libesp_media_protocols.zip)

carlosrodridel, 2024-06-18:
> I have checked the log seems the issue is caused by second invite response not contain sdp body, I have build one version with response with sdp also when reply after connection build, you can have [...]

Hi,

Thank you for the quick response and for providing the updated version.

I have tested the solution and observed in Wireshark that the ESP32 is now responding to the INVITE SDP messages with the corresponding 200 OK SDP messages. However, I'm still receiving more INVITE messages from the server. This indicates that there might be something missing or incorrect in the response message from the ESP32.

[image]

I will continue to perform tests to identify the exact issue and will post a new comment with my findings as soon as possible.

Thanks again for your assistance.

carlosrodridel, 2024-06-26:
Hi @TempoTian ,

I have completed the necessary tests and have some findings to share.

Using a Python VoIP library, I reproduced the OK SDP replies that the ESP32 is sending to the server to debug the SIP packets. Here are my analysis and conclusions:

## Analysis of Frames Replicating ESP32 Behavior:
[image]

### INVITE SDP from Server:
```
INVITE sip:0033XXXXXXXX@192.168.2.101:5060;transport=udp SIP/2.0
Call-ID: 07739-VS-e9e0fd68-34308c521@sip3.ovh.fr
Contact: <sip:XX.XXX.XXX.XXX:5060>
Content-Type: application/sdp
CSeq: 1992890617 INVITE
From: "+33XXXXXXXXX" <sip:0XXXXXXXXX@sip3.ovh.fr;user=phone>;tag=07739-HP-e9e0fd69-5378e1814
Max-Forwards: 29
Record-Route: <sip:XX.XXX.XXX.XXX:5060;lr>;session=467403
To: <sip:0XXXXXXXX@10.7.1.68;user=phone>
Via: SIP/2.0/UDP XX.XXX.XXX.XXX:5060;branch=z9hG4bK-OHOR-13ec59b9-147ba413
Allow: REFER,INVITE,NOTIFY,ACK,UPDATE,OPTIONS,REGISTER,SUBSCRIBE,NOTIFY,CANCEL,BYE,PRACK
User-Agent: Cirpack/v4.76 (gw_sip)
Content-Length: 323

v=0
o=anonymous 171932821733 171932821733 IN IP4 XX.XXX.XXX.XXX
s=SIP Call
c=IN IP4 91.121.129.155
t=0 0
m=audio 36700 RTP/AVP 8 0 18 101
b=AS:82
a=rtpmap:8 PCMA/8000/1
a=rtpmap:0 PCMU/8000/1
a=rtpmap:18 G729/8000/1
a=fmtp:18 annexb=no
a=rtpmap:101 telephone-event/8000
a=fmtp:101 0-15
a=ptime:20
a=sendrecv
```

### First 200 OK SDP:
```
SIP/2.0 200 OK
Via: SIP/2.0/UDP XX.XXX.XXX.XXX:5060;branch=z9hG4bK-OHOR-13ec59b9-147ba413
Record-Route: <sip:XX.XXX.XXX.XXX:5060;lr>;session=467403
From: "+33XXXXXXXXX" <sip:0XXXXXXXXX@sip3.ovh.fr;user=phone>;tag=07739-HP-e9e0fd69-5378e1814
To: <sip:0XXXXXXXX@10.7.1.68;user=phone>;tag=-1239239399
Call-ID: 07739-VS-e9e0fd68-34308c521@sip3.ovh.fr
CSeq: 1992890617 INVITE
Server: ESP32 SIP/2.0
Contact: <sip:0033XXXXXXXX@192.168.2.101:5060;transport=UDP>
Allow: INVITE, ACK, BYE, CANCEL, UPDATE, OPTIONS, INFO
Content-Type: application/sdp
Content-Length: 153

v=0
o=Z 0 3 IN IP4 192.168.2.6
s=Z
t=0 0
c=IN IP4 192.168.2.6
m=audio 31952 RTP/AVP 8 101
a=rtpmap:8 PCMA/8000
a=rtpmap:101 telephone-event/8000
```

### Second INVITE SDP from Server:
```

[... 157 lines cut ...]
CSeq: 2003502912 INVITE
Server: ESP32 SIP/2.0
Contact: <sip:0033XXXXXXXX@192.168.2.101:5060;transport=UDP>
Allow: INVITE, ACK, BYE, CANCEL, UPDATE, OPTIONS, INFO
Content-Type: application/sdp
Content-Length: 153

v=0
o=Z 0 3 IN IP4 192.168.2.6
s=Z
t=0 0
c=IN IP4 192.168.2.6
m=audio 31952 RTP/AVP 8 101
a=rtpmap:8 PCMA/8000
a=rtpmap:101 telephone-event/8000
```

After making the correction, the server properly acknowledges the OK replies with an ACK and stops the continuous INVITE SDP requests.

**Note on CSeq Number:**
It was also observed that the ESP32 replies to the first INVITE SDP with the same CSeq number as the request, which is correct. However, for subsequent INVITE requests from the server, the CSeq increments while the ESP32 replies with the same CSeq as the first request. Currently, this is not causing any issues with the server's acceptance of the response, but it might be a potential issue with different servers in the future.

Finally, I would like to request this correction specifically for the ESP32-S3, as we are currently using a custom ESP32-S3 board to implement our solution. The Lyra Board was utilized solely for testing purposes.

Please let me know if there are any further steps or additional information required.

TempoTian, 2024-06-26:
Thanks for you investigation. I have fixed the issue you mentioned, please help to try if OK we will merge it
[esp_media_protocol.zip](https://github.com/user-attachments/files/15987984/esp_media_protocol.zip)

carlosrodridel, 2024-06-26:
Thank you for the quick response and fix. I have tested the updated library on both the ESP32 and ESP32S3, and I can confirm that both are now working as expected. Thanks again for your help!

carlosrodridel, 2024-07-08:
hi, @TempoTian Could you please remember to merge the fix once it's verified? Thank you!