espressif/esp-adf #1213: input key service causes esp_timer to consume 50% of CPU time on ESP-LyraTv4.3

bs-eng, 2024-06-13:
## Environment

- Audio development kit:      ESP32-LyraT
- Audio kit version: v4.3
- Module or chip used: ESP32-WROVER-E
- IDF version: v5.1.2 (also others before that)
- ADF version : v2.6
- Build system:       CMake
- Running log:         [audioapp_115k__custom_app_timer_load_log.txt](https://github.com/user-attachments/files/15821889/audioapp_115k__custom_app_timer_load_log.txt) + other logs, see below
- Compiler version: [esp_tools.txt](https://github.com/user-attachments/files/15821933/esp_tools.txt)
- Operating system:     Windows10
- (Windows only) Environment type: ?
- Using an IDE?: Eclipse with Espressif IDF plugin
- Power supply:         USB

## Problem Description

I ran into a speed issue with my code. The code runs so slow that audio playback is stuttering on all audio files I play. It seems that the *input key service* is consuming large amounts of CPU time (~50%).

### Expected Behavior

The CPU load of the timer should be much lower, to a degree that audios play without stuttering. Using ADFv2.4 and IDFv3.3 (which is what our code originally was compiled on) the system ran smooth without any stuttering audios.

### Actual Behavior

See also thread on Espressif forum for additional details: [esp_timer consumes 50% of CPU time, please help](https://esp32.com/viewtopic.php?f=13&t=39795&p=132740#p131927)
I gathered the info on all tasks and noticed that the task esp_timer is consuming about half of the available CPU time. See stats here:
```
W (587604) AudioApp:  #            task name  core free stack   run time      in %%
W (587614) AudioApp:  0            AudioTask     1      13856     373664         <1%
W (587614) AudioApp:  1                 main     0       2056     463896         <1%
W (587614) AudioApp:  2                 IDLE     1        936  585918953         99%
W (587634) AudioApp:  3                 IDLE     0        928  274725393         46%
W (587634) AudioApp:  4           esp_periph     0       2316    2387659         <1%
W (587634) AudioApp:  5              Tmr Svc     0        160     924679         <1%
W (587634) AudioApp:  6                 file     0       2176     268981         <1%
W (587634) AudioApp:  7                  mp3     0       3892    2513101         <1%
W (587634) AudioApp:  8               filter     0       2960    1114041         <1%
W (587634) AudioApp:  9                  i2s     0       2640     599156         <1%
W (587634) AudioApp: 10                 ipc0     0        952       8576         <1%
W (587644) AudioApp: 11                 hciT     0       1568     504208         <1%
W (587644) AudioApp: 12             BTU_TASK     0       2964     752456         <1%
W (587644) AudioApp: 13             BTC_TASK     0       1508     355212         <1%
W (587644) AudioApp: 14         btController     0       2016    5001571         <1%
W (587644) AudioApp: 15            esp_timer     0       3324  296709377         50%
W (587644) AudioApp: 16      input_key_servi     1       2472         68         <1%
```
From there I drilled down to find out what is going on. I ended up disabling all timers in my application. Still the problem persisted. So I got more output from the timer task using `esp_timer_dump()`. This showed that the "touch filter timer" is using massive amount of CPU time in its callback function.

### Steps to Reproduce
From there I drilled down more and used two plain examples from from the ESP-ADFv2.6 on a ESP-Lyratv4.3 called **pipeline_play_sdcard_music** and **pipeline_sdcard_mp3_control** and added support for:

- esp_timer_dump()
- uxTaskGetSystemState() (optional for reproducing)

in order to collect system stats.
Both projects were build in a standard way

1. import example project
2. add esp_timer_dump() and uxTaskGetSystemState() - see code at bottom
3. configure sdkconfig
4. build
5. flash to ESP-LyraTv4.3
6. run

In pipeline_play_sdcard_music the touch peripheral is not used, not even initialized as far as I can see. There is no timer running according to esp_timer_dump().
```
W (17969) PLAY_SDCARD_MUSIC:  #            task name  core free stack   run time      in %%
W (17979) PLAY_SDCARD_MUSIC:  0                 main     0       1588    1250767          7%
W (17979) PLAY_SDCARD_MUSIC:  1                 IDLE     1       1036   16569970         99%
[... 73 lines cut ...]
            {
                // What percentage of the total run time has the task used?
                // This will always be rounded down to the nearest integer.
                // ulTotalRunTimeDiv100 has already been divided by 100.
                ulStatsAsPercentage = pxTaskStatusArray[ x ].ulRunTimeCounter / ulTotalRunTime;

                if( ulStatsAsPercentage > 0UL )
                {
                    ESP_LOGW( TAG, "%2i %20s %5i %10lu %10lu %10lu%%", x, pxTaskStatusArray[ x ].pcTaskName, xTaskGetAffinity(pxTaskStatusArray[ x ].xHandle), pxTaskStatusArray[ x ].usStackHighWaterMark, pxTaskStatusArray[ x ].ulRunTimeCounter, ulStatsAsPercentage );
                }
                else
                {
                    // If the percentage is zero here then the task has
                    // consumed less than 1% of the total run time.
                	ESP_LOGW( TAG, "%2i %20s %5i %10lu %10lu         <1%%", x, pxTaskStatusArray[ x ].pcTaskName, xTaskGetAffinity(pxTaskStatusArray[ x ].xHandle), pxTaskStatusArray[ x ].usStackHighWaterMark, pxTaskStatusArray[ x ].ulRunTimeCounter );
                }
            }
        }

        // The array is no longer needed, free the memory it consumes.
        vPortFree( pxTaskStatusArray );
    }
    ESP_ERROR_CHECK(esp_timer_dump(stdout));
}
```

hbler99, 2024-07-18:
We are grateful for your input; however, this issue was not identified during our testing process.

bs-eng, 2025-07-17:
> same problem...
> IDF: 5.3.1
> devboard: LyraT 4.3
> Task Name Core Stack Left CPU %
>
> main 0 456 1.74%
> IDLE1 1 1024 99.82%
> IDLE0 0 916 43.87%
> Tmr Svc - 1132 0.08%
> esp_periph 0 2036 0.38%
> audio_a2dp_stre 0 1484 0.00%
> input_key_servi 1 2416 0.00%
> hciT 0 1528 0.25%
> BTU_TASK 0 2680 0.26%
> BTC_TASK 0 976 0.21%
> bt_audio_task 1 3524 0.00%
> btController 0 2080 4.21%
> esp_timer 0 2172 48.94%
> ipc1 1 504 0.13%
> ipc0 0 400 0.09%
>
> @bs-eng <https://github.com/bs-eng> did you find a solution?

The problem still exists. The solution for me was to downgrade to the IDF
that comes directly with the ADF.