espressif/esp-adf #1215: how to run (master)adf on the board esp32s3lcd evboard2(800x480)?

liujkek22, 2024-06-21:
1. Thanks for looking into this issue; your response might be greatly helpful to me.
2. Based on my current understanding, the steps I've taken include modifying the `adf/components/audio_board` directory to accommodate the `esp32s3lcdevboard2` directory and including four files within it: `board.h`, `board.c`, `board_defs.h`, and `board_pins_config.c`. These files were initially sourced from the `esp32_s3_korvo2_v3` board and subsequently tailored to match the actual configurations of the `esp32s3lcdevboard2`. Additionally, adjustments were made to the `./CMakeLists.txt` file and the `Kconfig.projbuild` to facilitate these changes.
3. Upon attempting to run a sample project titled "Play MP3 Control," an I2C error emerged. I'm aware that the differences between the two boards are significant, and it's crucial to properly configure the power supply for the PA (Power Amplifier) on the `esp32s3lcdevboard2` to ensure successful execution.
4. If you have any suggestions or guidance, please feel free to email me at liuhutao2021@163.com. Your insights would be much appreciated.

hbler99, 2024-07-12:
Some steps need to be carried:
1. Add [esp32_s3_lcd_ev_v1_5.zip](https://github.com/user-attachments/files/16192529/esp32_s3_lcd_ev_v1_5.zip) to $ADF_PATH/components/audio_board
2. Add esp32s3lcd evboard2 option in CMakeLists.txt, component.mk, and Kconfig.projbuild.
3. Modify line 35 in components/audio_hal/driver/es7210/es7210.c to:
`#define ES7210_ADDR                   ES7210_AD1_AD0_01`
5. Change the function in components/audio_hal/driver/es8311/es8311.c at line 291 to:
`esp_err_t es8311_pa_power(bool enable)
{
    // esp_err_t ret = ESP_OK;
    // if (enable) {
    //     ESP_LOGW(TAG, "PA Enable");
    //     ret = gpio_set_level(get_pa_enable_gpio(), 1);
    // } else {
    //     ret = gpio_set_level(get_pa_enable_gpio(), 0);
    // }
    // return ret;

    esp_tca9554_config_t pca_cfg = {
        .i2c_scl = GPIO_NUM_18,
        .i2c_sda = GPIO_NUM_8,
        .interrupt_output = -1,
    };
    tca9554_init(&pca_cfg);
    tca9554_set_io_config(BIT(0), TCA9554_IO_OUTPUT);
    esp_err_t ret = ESP_OK;
    if (enable) {
        ESP_LOGW(TAG, "PA Enable");
        ret = tca9554_set_output_state(BIT(0), TCA9554_IO_HIGH);    
    } else {
        ret = tca9554_set_output_state(BIT(0), TCA9554_IO_LOW);
    }
    return ESP_OK;
}`
If you need to use other boards, revert the changes in step 3 and step 4.
6.  Run idf.py menuconfig and select your user-defined audio hal -> Audio Board
7. Good luck

liujkek22, 2024-07-12:
Thank you for your guidance; despite my numerous attempts following your suggested approach, the monitor persistently displays an error: "E (455) I2C_BUS: D:/ESP52/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:115 (i2c_bus_write_bytes): I2C Bus Write Register Error." I appreciate your assistance and am grateful for any further advice.

hbler99, 2024-07-13:
@liujkek22  looks like there might be a misconfiguration of the I2C. Which version of the esp32s3lcd evboard are you using? v1.5 or earlier?

liujkek22, 2024-07-14:
Yes,the Verssion is truely is 1.5.

hbler99, 2024-07-15:
Please check the following:

1. Ensure the board is configured as the provided audio_board.
2. In the es8311.c file, make sure to include #include "tca9554.h".
3. Verify the I2C pins are correctly configured as I2C_SCL->GPIO_NUM_18 and I2C_SDA->GPIO_NUM_8.

closed 2025-02-17 by jason-mao