espressif/esp-adf #1216: esp32_s3_korvo2_v3开发板如何设置MIC麦克风的增益大小

zsjlovestudy, 2024-06-23:
在我的应用中我需要更改麦克风的增益大小，默认的增益太大了用不了。
我看到board_def.h，有一个增益大小定义。还有在es8311.c中有一个修改增益的函数。我修改成12DB，但是均不成功。
而且在使用es8311中的更改增益函数后，发现录音的文件更是直接没有声音，估计是增益过于的小。
[image]
[image]
我想要知道如何修改mic麦克风的增益大小。
我发现这个设定的增益的宏定义根本没有被调用。
[image]

TempoTian, 2024-06-26:
下面的函数是设置mic增益的
```c
esp_err_t es8311_set_mic_gain(es8311_mic_gain_t gain_db)
{
    esp_err_t res = ESP_OK;
    res = es8311_write_reg(ES8311_ADC_REG16, gain_db); // MIC gain scale
    return res;
}
```

Oreobird, 2024-07-30:
es8311不是接speaker的么？Mic的增益是用es7211_set_mic_gain()?