espressif/esp-adf #1218: Setting afe_mode=SR_MODE_HIGH_PERF caused watchdog

Oreobird, 2024-06-24:
I use esp-idf-v5.1.1 and esp-adf-master-1f275b5 to run speach_reconigization application on ESP32-S3-BOX-3. The recorder sr config as follows:
```
recorder_sr_cfg_t sr_cfg = DEFAULT_RECORDER_SR_CFG();
sr_cfg.afe_cfg.memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM;
sr_cfg.afe_cfg.afe_mode = SR_MODE_HIGH_PERF;  // SR_MODE_LOW_COST;
sr_cfg.afe_cfg.vad_mode = VAD_MODE_4;
sr_cfg.afe_cfg.wakenet_init = true;
sr_cfg.afe_cfg.agc_mode = AFE_MN_PEAK_AGC_MODE_3;
sr_cfg.multinet_init = s_ctx.multinet_enable;
sr_cfg.afe_cfg.aec_init = true;
sr_cfg.rb_size = 12 * 1024;
```
Setting sr_cfg.afe_cfg.afe_mode = SR_MODE_LOW_COST works well, but setting to SR_MODE_HIGH_PERF cause me watchdog. Error log as below:
```
E (192217) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (192217) task_wdt:  - IDLE (CPU 0)
E (192217) task_wdt: Tasks currently running:
E (192217) task_wdt: CPU 0: app_task
E (192217) task_wdt: CPU 1: IDLE
E (192217) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4200C563:0x3FCA96C0 0x40379735:0x3FCA96F0 0x4207C8EB:0x3FCE1D80 0x4207B016:0x3FCE1DA0 0x4207B740:0x3FCE1F60 0x4206074E:0x3FCE1F80 0x4205E753:0x3FCE1FB0 0x42056EBD:0x3FCE1FF0 0x4205788E:0x3FCE2330 0x4204AE17:0x3FCE23C0 0x420255E1:0x3FCE2450 0x4200435A:0x3FCE2470 0x420031B1:0x3FCE24F0 0x4205270B:0x3FCE25B0
0x4200c563: task_wdt_timeout_handling at /home/platforms/esp-idf_v5.1.1/components/esp_system/task_wdt/task_wdt.c:461
 (inlined by) task_wdt_isr at /home/platforms/esp-idf_v5.1.1/components/esp_system/task_wdt/task_wdt.c:585

0x40379735: _xt_lowint1 at /home/platforms/esp-idf_v5.1.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1236

0x4207c8eb: flash_model_info at /home/sunxiangyu/workspace/esp_sr_lib/components/hufzip/hufzip_model.c:473

0x4207b016: model_create at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9_quantized.c:285 (discriminator 1)

0x4207b740: model_init at /home/sunxiangyu/workspace/esp_sr_lib/components/wakenet/wakenet9_quantized.c:884

0x4206074e: afe_create_from_config at /home/sunxiangyu/workspace/esp_sr_lib/components/esp_audio_front_end/esp_afe_sr.c:328

0x4205e753: recorder_sr_create at /home/platforms/esp-adf_1f275b5/components/audio_recorder/recorder_sr.c:591
```

Related config in sdkconfig is as follows:
```

#
# ESP Speech Recognition
#
CONFIG_MODEL_IN_SPIFFS=y
# CONFIG_MODEL_IN_SDCARD is not set
CONFIG_USE_AFE=y
CONFIG_AFE_INTERFACE_V1=y
# CONFIG_USE_NSNET is not set
CONFIG_USE_WAKENET=y
# CONFIG_SR_WN_WN8_ALEXA is not set
# CONFIG_SR_WN_WN9_HILEXIN is not set
# CONFIG_SR_WN_WN9_XIAOAITONGXUE is not set
# CONFIG_SR_WN_WN9_ALEXA is not set
# CONFIG_SR_WN_WN9_HIESP=y
CONFIG_SR_WN_WN9_HIMFIVE=y
# CONFIG_SR_WN_WN9_NIHAOXIAOZHI is not set
# CONFIG_SR_WN_WN9_JARVIS_TTS=y
# CONFIG_SR_WN_WN9_COMPUTER_TTS is not set
# CONFIG_SR_WN_WN9_CUSTOMWORD is not set
# CONFIG_SR_WN_LOAD_MULIT_WORD is not set
# CONFIG_SR_WN_WN9_ALEXA_MULTI is not set
# CONFIG_SR_WN_WN9_HIESP_MULTI is not set
# CONFIG_SR_WN_WN9_JARVIS_TTS_MULTI is not set
CONFIG_USE_MULTINET=y
CONFIG_SR_MN_CN_NONE=y
# CONFIG_SR_MN_CN_MULTINET5_RECOGNITION_QUANT8 is not set
# CONFIG_SR_MN_CN_MULTINET6_QUANT=y
# CONFIG_SR_MN_CN_MULTINET6_AC_QUANT is not set
# CONFIG_SR_MN_CN_MULTINET7_QUANT is not set
# CONFIG_SR_MN_CN_MULTINET7_AC_QUANT is not set
# CONFIG_SR_MN_EN_MULTINET5_SINGLE_RECOGNITION_QUANT8 is not set
CONFIG_SR_MN_EN_MULTINET6_QUANT=y
# CONFIG_SR_MN_EN_MULTINET7_QUANT is not set
# end of ESP Speech Recognition

#
# ADF Features
#
CONFIG_ESP_DISPATCHER_DELEGATE_TASK_CORE=0
CONFIG_ESP_DISPATCHER_DELEGATE_TASK_PRIO=10
CONFIG_ESP_DISPATCHER_DELEGATE_STACK_SIZE=4096
# end of ADF Features
```

hbler99, 2024-07-12:
I encountered a similar issue while running the WWE example using ESP-IDF v5.1.1, but the problem was resolved when I upgraded to ESP-IDF v5.2. Maybe try upgrading your IDF version to v5.2 to solve this problem.

Oreobird, 2024-08-02:
> I encountered a similar issue while running the WWE example using ESP-IDF v5.1.1, but the problem was resolved when I upgraded to ESP-IDF v5.2. Maybe try upgrading your IDF version to v5.2 to solve this problem.

Change the wakenet can solve the problem.

closed 2024-08-02 by Oreobird