espressif/esp-adf #1226: A lot of "deprecated" in VoIP example

Jon811, 2024-07-08:
## Environment

- Audio development kit:      Waiting for ESP32-LyraT-Mini
- [Required] Module or chip used:  Choose ESP32-LyraT-Mini
- [Required] IDF version v4.4.4-278-g3c8bc2213c
- [Required] ADF version v2.6-122-g28736657
- Build system:        Visual Studio code
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version: not a command in that folder
- Operating system:     [Windows
- (Windows only) Environment type: VSCode
- Using an IDE?:  ESP-IDF addition in VSCode
- Power supply:         [USB

At compilation stage have a lot of "deprecated"

<details><summary>Details</summary>
<p>
[931/1326] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/sip_service.c.obj
In file included from C:/Users/eulitin/esp/v5.2/esp-idf/components/driver/deprecated/driver/i2s.h:23,
                 from C:/Users/eulitin/.espressif/esp-adf/examples/protocols/components/av_stream/av_stream_hal/av_stream_hal.h:28,
                 from C:/Users/eulitin/.espressif/esp-adf/examples/protocols/components/av_stream/av_stream.h:28,
                 from C:/Users/eulitin/YandexDisk/Firmware/VoIP_outside/voip/voip/main/sip_service.h:30,
                 from C:/Users/eulitin/YandexDisk/Firmware/VoIP_outside/voip/voip/main/sip_service.c:28:
C:/Users/eulitin/esp/v5.2/esp-idf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
C:/Users/eulitin/esp/v5.2/esp-idf/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
[932/1326] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/voip_app.c.obj
In file included from C:/Users/eulitin/esp/v5.2/esp-idf/components/driver/deprecated/driver/i2s.h:23,
                 from C:/Users/eulitin/.espressif/esp-adf/examples/protocols/components/av_stream/av_stream_hal/av_stream_hal.h:28,
                 from C:/Users/eulitin/.espressif/esp-adf/examples/protocols/components/av_stream/av_stream.h:28,
                 from C:/Users/eulitin/YandexDisk/Firmware/VoIP_outside/voip/voip/main/sip_service.h:30,
                 from C:/Users/eulitin/YandexDisk/Firmware/VoIP_outside/voip/voip/main/voip_app.c:20:
C:/Users/eulitin/esp/v5.2/esp-idf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
C:/Users/eulitin/esp/v5.2/esp-idf/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
[997/1326] Building C object esp-idf/esp-adf-libs/CMakeFiles/__idf_esp-adf-libs.dir/esp_codec/audio_forge.c.obj
C:/Users/eulitin/.espressif/esp-adf/components/esp-adf-libs/esp_codec/audio_forge.c: In function 'audio_forge_sonic_set_speed':
C:/Users/eulitin/.espressif/esp-adf/components/esp-adf-libs/esp_codec/audio_forge.c:993:15: warning: using integer absolute value function 'abs' when argument is of floating-point type 'float' [-Wabsolute-value]
  993 |     if ((int)(abs((sonic_speed - audio_forge->sonic_speed) * 100)) <= 5) {
      |               ^~~
C:/Users/eulitin/.espressif/esp-adf/components/esp-adf-libs/esp_codec/audio_forge.c: In function 'audio_forge_sonic_set_pitch':
C:/Users/eulitin/.espressif/esp-adf/components/esp-adf-libs/esp_codec/audio_forge.c:1018:15: warning: using integer absolute value function 'abs' when argument is of floating-point type 'float' [-Wabsolute-value]
 1018 |     if ((int)(abs((sonic_pitch - audio_forge->sonic_pitch) * 100)) <= 5) {
      |               ^~~
[1019/1326] Building C object esp-idf/esp_peripherals/CMakeFiles/__idf_esp_peripherals.dir/periph_adc_button.c.obj
In file included from C:/Users/eulitin/.espressif/esp-adf/components/esp_peripherals/include/periph_adc_button.h:28,
                 from C:/Users/eulitin/.espressif/esp-adf/components/esp_peripherals/periph_adc_button.c:28:
C:/Users/eulitin/esp/v5.2/esp-idf/components/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
[1028/1326] Building C object esp-idf/esp_peripherals/CMakeFiles/__idf_esp_peripherals.dir/periph_wifi.c.obj
In file included from C:/Users/eulitin/.espressif/esp-adf/components/esp_peripherals/periph_wifi.c:26:
C:/Users/eulitin/esp/v5.2/esp-idf/components/wpa_supplicant/esp_supplicant/include/esp_wpa2.h:10:9: note: '#pragma message: esp_wpa2.h is deprecated. Use esp_eap_client.h instead.'
   10 | #pragma message("esp_wpa2.h is deprecated. Use esp_eap_client.h instead.")
      |         ^~~~~~~
C:/Users/eulitin/.espressif/esp-adf/components/esp_peripherals/periph_wifi.c: In function '_wifi_init':
C:/Users/eulitin/.espressif/esp-adf/components/esp_peripherals/periph_wifi.c:435:9: warning: 'esp_wifi_sta_wpa2_ent_set_ca_cert' is deprecated: Use 'esp_eap_client_set_ca_cert' instead [-Wdeprecated-declarations]
  435 |         ESP_ERROR_CHECK(esp_wifi_sta_wpa2_ent_set_ca_cert((const unsigned char *)periph_wifi->wpa2_e_cfg->ca_pem_start, ca_pem_bytes));
      |         ^~~~~~~~~~~~~~~
C:/Users/eulitin/esp/v5.2/esp-idf/components/wpa_supplicant/esp_supplicant/include/esp_wpa2.h:171:11: note: declared here
  171 | esp_err_t esp_wifi_sta_wpa2_ent_set_ca_cert(const unsigned char *ca_cert, int ca_cert_len);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/eulitin/.espressif/esp-adf/components/esp_peripherals/periph_wifi.c:436:9: warning: 'esp_wifi_sta_wpa2_ent_set_cert_key' is deprecated: Use 'esp_eap_client_set_certificate_and_key' instead  
[... 152 lines cut ...]
</details> 

After uploading to ESP32-CAM (for now i have only it, waiting for Lyra) at start i have: 
<details><summary>Details</summary>
<p>
I (844) heap_init: Initializing. RAM available for dynamic allocation:
I (851) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (857) heap_init: At 3FFB5BE8 len 0002A418 (169 KiB): DRAM
I (864) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (870) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (876) heap_init: At 400970E0 len 00008F20 (35 KiB): IRAM
I (883) esp_psram: Adding pool of 2033K of PSRAM memory to heap allocator
I (891) spi_flash: detected chip: generic
I (895) spi_flash: flash io: qio
E (899) i2s(legacy): CONFLICT! The new i2s driver can't work along with the legacy i2s driver
W (908) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
W (920) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
W (931) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (940) main_task: Started on CPU0
I (944) esp_psram: Reserving pool of 18K of internal memory for DMA/internal allocations
I (952) main_task: Calling app_main()
</p>
</details> 

So even before start the main program we have a conflict. Hope all this can be fixed.

Jon811, 2024-07-10:
Changing ISP-IDF to 4.4.8 fixed problems, but why can't we use last versions of IDF? I don't understand.

closed 2024-07-10 by Jon811

mrmonteith, 2024-10-27:
> Changing ISP-IDF to 4.4.8 fixed problems, but why can't we use last versions of IDF? I don't understand.

I tried some of the examples on 4.4.8 today and tons are depricated and one final error.  It does make it hard to get very far.