espressif/esp-adf #1227: To configure RTP DTMF events duration and volume.

Zoyolin, 2024-07-09:
**Describe the cituation**
I use the [esp_rtc_send_dtmf](https://github.com/espressif/esp-adf-libs/blob/master/esp_media_protocols/include/esp_rtc.h#L200) to send dtmf events "out of band" as part of the VOIP (rtp/sip) stream. In order for the receiving device to understand them certain durations and volume are required. This arrises even more when the out of band dtmf packet reaches a gateway device that converts it in analog, in band, for a PSTN (non-packetted) telephone network.

The current values I saw are :
 - duration=80 in timestamp units. 1 is the length of a rtp sampling period, in my case (and most sound codec) the Fs is 8kHz so 80 corresponds to 10ms (which is already too short according to ITU-T Recommendation, see rfc). However to handle the compatibility with custom hardwares, I'll need to set my specific values in this field, not only 40ms (the ITU-T recommendation).
 - volume=13 which corresponds to -13dBm0. In comparison to pjsip default, -10dBm0, or another stack default -7dBm0 that does result is significantly lower signals once reintroduced inband. I'd like to be able to set louder signals.

**Describe the solution you'd like**
 - [ ] I'd like to be able to configure the dtmf event [duration](https://datatracker.ietf.org/doc/html/rfc4733#section-2.3.5). I need to be able to configure this field separately for each dtmf event. RTP timestamp rate are media dependant and in my case I need a range from 0 to 4s at 8000Hz aka highest value 32000, which is covered by the RFC 16bit field.
 - [ ] I'd like to be able to configure the dtmf event [volume](https://datatracker.ietf.org/doc/html/rfc4733#section-2.3.4). This configuration can a slower one, I need to be able to set it every few rtp packets, potentially only once at boot up (RFC is 6bits depth value).

**Additional context**
[image]
As per the above sniffed packets, the Dtmf event starts on packet N°2468 and ends after 3 end-event packets on N°2881. According to the RFC, the event duration should increase with each non-end-event rtp packet. We sniffed these 9 non-end-event packets a bunch of milliseconds apart (2-5ms) corresponding to increases of rougthly about 40units, but the duration field stays at 80units.

OS darwin arm64 23.5.0 
Visual Studio Code version 1.90.2 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.8.0 
Git version 2.45.0
ESP-IDF version v4.4.5
ADF version v2.6-122-g28736657
Python version 3.12.4
Python's pip version 24.0
Chip used:  ESP32-S3 custom board
Build system: idf.py

TempoTian, 2024-07-15:
I think mat be `int esp_rtc_send_dtmf(esp_rtc_handle_t esp_rtc, uint8_t dtmf_event, uint8_t vol, uint16_t duration (unit ms));` fit your needs. Also I see in code it send all packet at once, is it fit your needs, or do you need to handle the send packet by yoursellf, like provide API ``int esp_rtc_send_dtmf(esp_rtc_handle_t esp_rtc, uint8_t dtmf_event, uint8_t vol, uint16_t duration, bool end);`, repeat call it to construct a DTMF event.

Zoyolin, 2024-07-15:
Hello @TempoTian, Indeed the first prototype API would satisfy us, thanks!! We don't need (at present) to control the rtp packets individually. As you said, `int esp_rtc_send_dtmf(esp_rtc_handle_t esp_rtc, uint8_t dtmf_event, uint8_t vol, uint16_t duration (unit ms));` is what we'd need. You're welcome to keep the present behaviour of sending X times the same rtp packet + Y end rtp packets or the alike for resilience.

TempoTian, 2024-07-19:
I have implement the new API, you can use library to verify it also.
https://github.com/user-attachments/files/16308908/esp_media_protocols.zip

closed 2025-02-17 by jason-mao