espressif/esp-adf #1229: examples\advanced_examples\algorithm 这个例程中，请问下如何获取到 aec和ns后的音频数据呢

Trasills, 2024-07-10:
examples\advanced_examples\algorithm\main\algorithm_examples.c
1 这个例程是在播放音乐的同时将麦克风收录的声音先进行回声消除，然后存储到 microSD 卡中，我这边不需要编码成wav写入SD卡中，我尝试在代码中找在哪里将音频编码写入sd卡，但是没有找到。

2 i2s_read 函数读出来的数据已经是aec后的数据吗
[image]

hbler99, 2024-07-11:
可以用 `audio_element_input(audio_element_handle_t el, char *buffer, int wanted_size)` 提取音频数据，可以在下面链接看到函数构造https://github.com/espressif/esp-adf/blob/bed20d72b9bbcd48183ef3dddcccb0af3b264eb1/components/audio_pipeline/audio_element.c#L373，`ESP_LOG_BUFFER_HEX` 函数可以用来显示内容

Trasills, 2024-07-11:
> 可以用 提取音频数据，可以在下面链接看到函数构造 [https://github.com/espressif/esp-adf/blob/bed20d72b9bbcd48183ef3dddcccb0af3b264eb1/components/audio_pipeline/audio_element.c#L373。](https://github.com/espressif/esp-adf/blob/bed20d72b9bbcd48183ef3dddcccb0af3b264eb1/components/audio_pipeline/audio_element.c#L373%E3%80%82) 函数可以用来显示内容'`audio_element_input(audio_element_handle_t el, char *buffer, int wanted_size)``ESP_LOG_BUFFER_HEX`

非常感谢你的支持，我这边验证一下，如果有问题，再向你请教。

jason-mao, 2024-07-29:
关闭这个任务，您如有需要可以reopen 或新建issue。

closed 2024-07-29 by jason-mao