espressif/esp-adf #1230: Examples give warnings and not work

Jon811, 2024-07-13:
- Audio development kit:     ESP32-LyraT-Mini
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): v1.2
- [Required] Module or chip used:  ESP32-WROVER-E
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it):
    // v4.4.8
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it):
    // v2.6-122-g28736657
- Build system:       CMake (Visual Studio)
- [Required] Running log:         All logs from power-on to problem recurrence
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it):
    // not a program name
- Operating system:    Windows
- (Windows only) Environment type: Visual Studio Code 
- Using an IDE?: Visual Studio Code
- Power supply: external 5V

## Problem Description

When trying to use example "SPIFFS_AMR_RESAMPLE_EXAMPLE" from isp-idf i see error in Serial:
E (16565) I2S: register I2S object to platform failed.
That happen when program try to "Create audio elements for playback pipeline"
But in example log i see: I2S: I2S driver already installed
Searching of this phrase in ADF did not give anything.
Same phrase found in logs of different examples on github.
How to make it work?

### Expected Behavior
Working example

### Steps to Reproduce
Try to compile any example that use 2 channels at once

hbler99, 2024-07-15:
Could you please print out the complete log? This way, we might be able to find the issue more quickly.

Jon811, 2024-07-15:
> Could you please print out the complete log? This way, we might be able to find the issue more quickly.

Here is my log when i try to run it:
I (27) boot: ESP-IDF v4.4.8-dirty 2nd stage bootloader
I (27) boot: compile time 11:38:39
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v3.1
I (35) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 8MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=23900h (145664) map
I (160) esp_image: segment 1: paddr=00033928 vaddr=3ffb0000 size=01d08h (  7432) load
I (163) esp_image: segment 2: paddr=00035638 vaddr=40080000 size=0a9e0h ( 43488) load
I (184) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=4d6dch (317148) map
I (299) esp_image: segment 4: paddr=0008d704 vaddr=4008a9e0 size=08120h ( 33056) load
I (322) boot: Loaded app from partition at offset 0x10000
I (322) boot: Disabling RNG early entropy source...
I (333) cpu_start: Multicore app
I (333) psram: This chip is ESP32-D0WD
I (335) spiram: Found 64MBit SPI RAM device
I (335) spiram: SPI RAM mode: flash 40m sram 40m
I (340) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (348) cpu_start: Pro cpu up.
I (352) cpu_start: Starting app cpu, entry point is 0x40081530
0x40081530: call_start_cpu1 at C:/Users/eulitin/esp/v4.4.8/esp-idf/components/esp_system/port/cpu_start.c:151

I (0) cpu_start: App cpu up.
I (1245) spiram: SPI SRAM memory test OK
I (1253) cpu_start: Pro cpu start user code
I (1253) cpu_start: cpu freq: 160000000
I (1253) cpu_start: Application information:
I (1256) cpu_start: Project name:     spiffs_amr_resample_app
I (1263) cpu_start: App version:      1
I (1267) cpu_start: Compile time:     Jul 15 2024 11:37:31
I (1273) cpu_start: ELF file SHA256:  12b9076593bd8589...
I (1279) cpu_start: ESP-IDF:          v4.4.8-dirty
I (1285) cpu_start: Min chip rev:     v0.0
I (1290) cpu_start: Max chip rev:     v3.99 
I (1295) cpu_start: Chip rev:         v3.1
I (1300) heap_init: Initializing. RAM available for dynamic allocation:
I (1307) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1313) heap_init: At 3FFB27C8 len 0002D838 (182 KiB): DRAM
I (1319) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1326) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1332) heap_init: At 40092B00 len 0000D500 (53 KiB): IRAM
I (1339) spiram: Adding pool of 4095K of external SPI memory to heap allocator
I (1348) spi_flash: detected chip: gd
I (1351) spi_flash: flash io: dio
I (1356) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1365) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1505) PERIPH_SPIFFS: Partition size: total: 896321, used: 4518
W (1535) I2C_BUS: I2C bus has been already created, [port:0]
I (1545) SPIFFS_AMR_RESAMPLE_EXAMPLE: [1.1] Initialize recorder pipeline
I (1545) SPIFFS_AMR_RESAMPLE_EXAMPLE: [1.2] Create audio elements for recorder pipeline
I (1555) SPIFFS_AMR_RESAMPLE_EXAMPLE: [1.3] Register audio elements to recorder pipeline
I (1555) SPIFFS_AMR_RESAMPLE_EXAMPLE: [2.2] Create audio elements for playback pipeline
E (1565) I2S: register I2S object to platform failed
I (1575) SPIFFS_AMR_RESAMPLE_EXAMPLE: [2.3] Register audio elements to playback pipeline
I (1585) SPIFFS_AMR_RESAMPLE_EXAMPLE: [ 3 ] Set up  event listener
E (8295) SPIFFS_AMR_RESAMPLE_EXAMPLE: STOP playback and START recording
W (8295) AUDIO_PIPELINE: Without stop, st:1
W (8295) AUDIO_PIPELINE: Without wait stop, st:1
I (8295) SPIFFS_AMR_RESAMPLE_EXAMPLE: Link audio elements to make recorder pipeline ready
I (8305) SPIFFS_AMR_RESAMPLE_EXAMPLE: Setup file path to save recorded audio
I (12695) SPIFFS_AMR_RESAMPLE_EXAMPLE: STOP recording and START playback
W (12695) AUDIO_ELEMENT: OUT-[filter_downsample] AEL_IO_ABORT
W (12715) AUDIO_ELEMENT: OUT-[i2s_reader] AEL_IO_ABORT
W (12715) AUDIO_ELEMENT: OUT-[i2s_reader] AEL_IO_ABORT
W (12715) AUDIO_ELEMENT: OUT-[i2s_reader] AEL_IO_ABORT
W (12725) AUDIO_ELEMENT: OUT-[i2s_reader] AEL_IO_ABORT
W (12725) AUDIO_ELEMENT: OUT-[amrnb_encoder] AEL_IO_ABORT
W (12755) AUDIO_ELEMENT: IN-[file_writer] AEL_IO_ABORT
I (12765) SPIFFS_AMR_RESAMPLE_EXAMPLE: Link audio elements to make playback pipeline ready
I (12765) SPIFFS_AMR_RESAMPLE_EXAMPLE: Setup file path to read the amr audio to play
W (12785) SPIFFS_STREAM: No more data, ret:0

Jon811, 2024-07-15:
In another hand when i try VoIP example i have the same error message and it's not working, so i think it will happen in any example containing both speaker and mic.

hbler99, 2024-07-16:
No worries, this is normal for IDF v4.4 and does not affect your work.

Jon811, 2024-07-16:
If it doesn't affect, then it's just not working. How to make it work in VSCode? May be there some example menuconfig, that works from someone?

Jon811, 2024-07-16:
Here is recording:
I (8305) SPIFFS_AMR_RESAMPLE_EXAMPLE: Setup file path to save recorded audio
I (12695) SPIFFS_AMR_RESAMPLE_EXAMPLE: STOP recording and START playback
4 seconds, and here is playback:
I (12765) SPIFFS_AMR_RESAMPLE_EXAMPLE: Setup file path to read the amr audio to play
W (12785) SPIFFS_STREAM: No more data, ret:0
20 milliseconds. So it's not work.