espressif/esp-adf #1231: ESP-ADF does not get recognized in VSCode after install

UnforeseenOcean, 2024-07-14:
## Environment

- Audio development kit: N/A
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): [v1|v2|v3|v4]
- [Required] Module or chip used:  [ESP32-WROOM-32E]
- [Required] IDF version: v4.4.8
- [Required] ADF version: v2.6-122-g28736657
- Build system: [CMake]
- [Required] Running log: Not applicable -- Problem occurs before the code is even compiled
- Compiler version: xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2-patch2) 8.4.0
- Operating system: Windows
- (Windows only) Environment type: PowerShell / default VSCode terminal
- Using an IDE?: [Visual Studio Code]
- Power supply:         [USB]

## Problem Description
Using any part of ESP-ADF in the project will cause "cannot open source file" error (such as `audio_common.h`)

However, using idf.py to test its installation status works fine.

### Expected Behavior
The header files are included with no issues.

### Actual Behavior
Visual Studio Code refuses to accept the `components` folder and causes compile errors when any of the ESP-ADF components are used.

### Steps to Reproduce
1. Install ESP-IDF
2. Install ESP-IDF v4.4 (since it's a modified version, it must be installed after installing VSCode extension via "Existing Setup" apparently, refer to [this repository](https://github.com/cfint/esp-idf/tree/v4.4-a2dp-sink-codecs) for the exact copy of ESP-IDF that has been grafted onto the stock v4.4.8)
3. Run `install.bat` for ESP-IDF
4. Run `export.bat` for ESP-IDF
5. Install ESP-ADF through in-app menu
6. Run `install.bat` for ESP-**A**DF
7. Run `export.bat` for ESP-**A**DF
8. Turn off the editor
9. Turn on the editor
10. Reset IntelliSense DB
11. Try to compile, however even that is not required since the example code below will result in an error

### Code to Reproduce This Issue
```cpp
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "audio_common.h" // A part of ESP-ADF that *should* be included if it's installed correctly

void setup() {
}

extern "C" void app_main(void) {
      setup();
}
```

[sdkconfig](https://github.com/user-attachments/files/16216202/sdkconfig.txt)

hbler99, 2024-07-15:
I recommend using the IDF release/v5.2 version. It worked normally for me.

UnforeseenOcean, 2024-07-15:
I figured out the sequence of steps to take to undo the damage mixed versions cause.

1. Download v4.4.8 and install it through VSCode extension (**specifically** choose that version from the list)
2. We only need the tools, delete v4.4.8 folder
3. Clone the modded v4.4.x into the folder NEXT to where v4.4.8 folder used to be
4. Use these commands to set up the local repository for the modded ESP-IDF correctly:
```
git config --global core.symlinks true
git clone -b v4.4-a2dp-sink-codecs https://github.com/cfint/esp-idf.git --recurse-submodules
```
5. Run the following to get the project:
```
git clone https://github.com/cfint/esp32-a2dp-sink.git --recurse-submodules
```
6. Now go through the settings AND the following files to update the paths:
```
./.vscode/c_cpp_properties.json (check compilerPath, includePath and browse path to make sure both IDF and ADF paths are included)
./.vscode/settings.json (check idf.espIdfPath, idf.espIdfPathWin, idf.espAdfPath, idf.espAdfPathWin, idf.pythonBinPathWin, idf.customExtraPaths, OPENOCD_SCRIPTS, idf.gitPathWin)
```
Pay **VERY** close attention to `idf.customExtraPaths` as they contain paths to ALL required tools. You'll encounter all sorts of weird errors if you make a typo or leave something outdated here.
7. Add and modify these lines in ./Makefile:
```makefile
PROJECT_NAME := your_project_name
include $(IDF_PATH)/project.mk 
# Change to: include $(ADF_PATH)/project.mk
```
8. Change contents of  ./CMakeLists.txt to:
```cmake
cmake_minimum_required(VERSION 3.5)
include($ENV{ADF_PATH}/CMakeLists.txt)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(your_project_name)
```
While doing this, you might want to use a tool named *Voidtools Everything* as this makes finding those folders and executables extremely easy.

If the build failed (evidenced by the FreeRTOS include failures), you need to remove the `build` folder too.

9. Then run these tasks one after another by pressing F1 in VSCode and selecting it from the list:
```
>ESP-IDF: Run idf.py reconfigure task
>C/C++: Rescan workspace
>C/C++: Reset IntelliSense database
```
10. Press "ESP-IDF: Build Project" (the wrench/spanner icon)
11. If the build was successful, the errors are fixed.

UnforeseenOcean, 2024-07-15:
It seems like if you use a custom version or something installed outside of the extension for the ESP-IDF environment, the default version of the tools installed after it is fixed to v5.0 instead of v4.4 as it should be, even if you have `.git` folder. This causes another layer of path confusion.

hbler99, 2024-07-16:
You're right, very cool work!