espressif/esp-adf #1234: 加入自己写的i2c程序 结合官方的rstp报错 api冲突如何修改

kkang520, 2024-07-17:
E (1086) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver
void rtsp(void *arg)
{
    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set("AUDIO_ELEMENT", ESP_LOG_ERROR);
    esp_log_level_set("AFE_VC", ESP_LOG_ERROR);
    AUDIO_MEM_SHOW(TAG);

    /* nvs init */
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES)
    {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);

    ESP_LOGI(TAG, "[ 1 ] Initialize peripherals management");
    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);

    setup_wifi(set);

    ESP_LOGI(TAG, "[ 2 ] Initialize av stream");
    av_stream_config_t av_stream_config = {
        .algo_mask = ALGORITHM_STREAM_USE_AEC,
        .acodec_samplerate = AUDIO_CODEC_SAMPLE_RATE,
        .acodec_type = AV_ACODEC_G711A,
        .vcodec_type = AV_VCODEC_MJPEG,
        .hal = {
            .uac_en = false,
            .uvc_en = false,
            .video_soft_enc = false,
            .audio_samplerate = AUDIO_HAL_SAMPLE_RATE,
            .audio_framesize = PCM_FRAME_SIZE,
            .video_framesize = RTSP_FRAME_SIZE,
        },
    };
    av_stream = av_stream_init(&av_stream_config);
    AUDIO_NULL_CHECK(TAG, av_stream, return);
    ESP_LOGI(TAG, "[ 3 ] Initialize console command");
    // console_init();
    register_rtsp_cmd();
}
void get_iic_data(void *arg)
{
    uint8_t k = 0;
    uint8_t pos = 0;
    int16_t result;
    uint8_t high_nibble;
    uint8_t low_byte;
    // uint16_t recast;
    char rx_buffer[128];
    uint8_t t = 0;
    float converted;
    float f_c[64];
    int sock;
    char host_ip[] = HOST_IP_ADDR;
    int addr_family = 0;
    int ip_protocol = 0;

    vTaskDelay(100);
    // 初化server的地址结构体sockaddr_in
    // 建立socket连接：
    // socklen_t len = sizeof(struct sockaddr);
    i2c_master_bus_config_t i2c_bus_config = {
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .i2c_port = PORT_NUMBER,
        .scl_io_num = SCL_IO_PIN,
        .sda_io_num = SDA_IO_PIN,
        .glitch_ignore_cnt = 7,
    };
    i2c_master_bus_handle_t bus_handle;

    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_bus_config, &bus_handle));
    i2c_eeprom_config_t eeprom_config = {
        .eeprom_device.scl_speed_hz = MASTER_FREQUENCY,
        .eeprom_device.device_address = 0x69,
        .addr_wordlen = 1,
        .write_time_ms = 10,
    };}
如何修改啊

shootao, 2024-07-31:
Hi @kkang520 
关于 支持 i2c master 的最近会释放出来，你可以先用该patch 试一下
[0001-i2c_bus-support-i2c-master.patch](https://github.com/user-attachments/files/16440190/0001-i2c_bus-support-i2c-master.patch)

jason-mao, 2024-12-10:
It's fixed on e27cd5262986dbf93785d44fb71dfa854b8e9763

closed 2024-12-10 by jason-mao

hyansuper, 2025-04-21:
通过 esp-registry 安装的 esp_codec_dev 1.3.4 还是有这个问题